lvgl/lvgl #5906: LVGL9: lv_scale_set_line_needle_value supports only 1 object

s-hadinger, 2024-03-18:
### LVGL version

v9.0.0

### What happened?

`lv_scale_set_line_needle_value()` works only if there is only 1 scale object. As soon as you update a line, it changes all lines of all scale object.

The root cause is that the line points are defined as `static lv_point_precise_t needle_line_points[2];`. Since line object don't keep a local copy of the points array, the points coordinates are shared with all scale objects.

https://github.com/lvgl/lvgl/blob/1cb3e219a971c6f80a692928c00d91c61cfb5f84/src/widgets/scale/lv_scale.c#L183

Currently I do a not-so-clean workaround:
- call `lv_scale_set_line_needle_value()`
- read back the points from the `line` object
- copy the points to a structure allocated per line object
- set back the line points with `lv_line_set_points()`

liamHowatt, 2024-03-21:
Thanks @s-hadinger

I was able to reproduce with a modified version of `lv_example_scale_3`.

I think the fix could be as simple as this. A needle line array can live in each `lv_scale_t`.

```diff
diff --git a/src/widgets/scale/lv_scale.c b/src/widgets/scale/lv_scale.c
index 1e8df7ade..c21fc50ff 100644
--- a/src/widgets/scale/lv_scale.c
+++ b/src/widgets/scale/lv_scale.c
@@ -180,7 +180,6 @@ void lv_scale_set_line_needle_value(lv_obj_t * obj, lv_obj_t * needle_line, int3
     int32_t scale_width, scale_height;
     int32_t actual_needle_length;
     int32_t needle_length_x, needle_length_y;
-    static lv_point_precise_t needle_line_points[2];
 
     LV_ASSERT_OBJ(obj, MY_CLASS);
     lv_scale_t * scale = (lv_scale_t *)obj;
@@ -224,12 +223,12 @@ void lv_scale_set_line_needle_value(lv_obj_t * obj, lv_obj_t * needle_line, int3
     needle_length_x = (actual_needle_length * lv_trigo_cos(scale->rotation + angle)) >> LV_TRIGO_SHIFT;
     needle_length_y = (actual_needle_length * lv_trigo_sin(scale->rotation + angle)) >> LV_TRIGO_SHIFT;
 
-    needle_line_points[0].x = scale_width / 2;
-    needle_line_points[0].y = scale_height / 2;
-    needle_line_points[1].x = scale_width / 2 + needle_length_x;
-    needle_line_points[1].y = scale_height / 2 + needle_length_y;
+    scale->needle_line_points[0].x = scale_width / 2;
+    scale->needle_line_points[0].y = scale_height / 2;
+    scale->needle_line_points[1].x = scale_width / 2 + needle_length_x;
+    scale->needle_line_points[1].y = scale_height / 2 + needle_length_y;
 
-    lv_line_set_points(needle_line, needle_line_points, 2);
+    lv_line_set_points(needle_line, scale->needle_line_points, 2);
 }
 
 void lv_scale_set_image_needle_value(lv_obj_t * obj, lv_obj_t * needle_img, int32_t value)
diff --git a/src/widgets/scale/lv_scale.h b/src/widgets/scale/lv_scale.h
index 8ac377eb1..01c179c1b 100644
--- a/src/widgets/scale/lv_scale.h
+++ b/src/widgets/scale/lv_scale.h
@@ -89,6 +89,7 @@ typedef struct {
     int32_t custom_label_cnt;
     int32_t last_tick_width;
     int32_t first_tick_width;
+    lv_point_precise_t needle_line_points[2];
 } lv_scale_t;
 
 LV_ATTRIBUTE_EXTERN_DATA extern const lv_obj_class_t lv_scale_class;
```

s-hadinger, 2024-03-21:
Thanks. I'm afraid that won't be enough. It would still be great to have 2 line on the same scale like here 
[image]

liamHowatt, 2024-03-21:
> It would still be great to have 2 line on the same scale like here

I created a draft PR  #5937 that achieves this. Try it out for your use case if you want. I changed the API a little bit.

I'm glad you also have a workaround.

kisvegabor, 2024-04-05:
The PR is merged, closing this issue.

closed 2024-04-05 by kisvegabor

s-hadinger, 2025-01-08:
I don't know, but your question is completely off-topic