lvgl/lvgl #5928: Text Decoding

fatihozen3, 2024-03-20:
### Introduce the problem

I use LVGL v9, I will receive text datas in unicode format, but I have to convert it to utf-8 format. How can I do that ?

For example for Capital letter A I will receive 0x00,0x41 datas from buffer. How can I show it on my screen with lv_label_set_text function ?

I know if I write "\x41" I can see the characterlv_label_set_text(test2, buffer); but I don't know how to give this data to the function.

I have tried below piece of code. I assumed 0x41 received and write it into buffer with sprintf as below but I see '\x41' on the screen I don't see 'A'. Because there are two '\'. I can't write just one..

uint16_t received_array[6] = {0x41};
char buffer[100] = {0};
sprintf(buffer, "\\x%x", received_array[0]);
lv_label_set_text(test, buffer);

W-Mai, 2024-03-21:
You can find some reference code here to meet your needs, and you can copy it to your project for use.

https://github.com/lvgl/lvgl/blob/4b41590cf1593e33b8a1ae0e86461f3c0046c1cc/tests/src/test_libs/rnd_unicodes/rnd_unicodes/src/library.c#L59-L129

liamHowatt, 2024-03-21:
You might be interested in joining the discussion here: #5913

fatihozen3, 2024-03-22:
> You can find some reference code here to meet your needs, and you can copy it to your project for use.
> 
> https://github.com/lvgl/lvgl/blob/4b41590cf1593e33b8a1ae0e86461f3c0046c1cc/tests/src/test_libs/rnd_unicodes/rnd_unicodes/src/library.c#L59-L129

I have modified it as below: I give it an array and it converts all characters if needed.

uint16_t array2[7] = {0x218,0x219,0x226,0x227,0x22a,0x1ff,0x1ff}; // I give this one.
uint16_t newbuf[7] = {0}; // I give this to lv_label_set_text function.

`void unicode_to_uft16(uint8_t * buf, uint16_t size, uint16_t * received_buf)
{
	for(int i = 0;i<size;i++)
	{
		static int buf_index = 0;

		if(received_buf[i]<0x80)
				{buf[buf_index++] = (uint8_t) received_buf[i];}
		else
		{
				buf[buf_index++] = (uint8_t)(0xc0 | (received_buf[i] >> 6));
				buf[buf_index++] = (uint8_t)(0x80 | (received_buf[i] & 0x3f));
		}
	}
}`

it works if I stay in one page but if I try to change it, program stucks at the beginning in Mem Manage Handler

closed 2024-06-20 by kisvegabor