lvgl/lvgl #5934: Missing documentation on how to use menuconfig

pvginkel, 2024-03-21:
### LVGL version

v9.0.0

### What happened?

I've used LVGL with ESP-IDF. Menuconfig is integrated into the Visual Studio Code environment and the idf.py tool. I'd also like to be able to use menuconfig on projects that are not based on ESP-IDF, but I cannot find documentation on how to achieve this. I can get as far as to get a `.config` file from `kconfig-mconf`, but I don't know how to get an `lv_conf.h` file from this. Could you provide the steps to generate an `lv_conf.h` header using menuconfig?

C47D, 2024-03-21:
As far as I know if you use the menuconfig tool there's no need to define lv_config.h, because the menuconfig defines all of the configuration symbols.

What platform or framework are you using?

pvginkel, 2024-03-21:
I'm not using any framework. I'm trying to get this to work on a Ubuntu machine using the `lv_port_linux_frame_buffer` repo. Basically I'd like to know the steps you have to follow to use the Kconfig system to generate an `lv_conf.h` (or similar) for a new clone of that repository.

kisvegabor, 2024-03-25:
LVGL itself doesn't have actual Kconfig backend as it's normally provided by frameworks into which LVGL is integrated. (E.g. ESP-IDF, Zephyr, NuttX) 

However, I've just tried out how to use LVGL's Kconfig directly, and found that you can download [menuconfig.py](https://github.com/ulfalizer/Kconfiglib/blob/master/menuconfig.py) to the `lvgl` folder and just call it as `python3 menuconfig.py`. (See the comments at the top of the file for more info).

It will generate a `.config` file which is not processed by LVGL. So one more step is required to convert `.config` file to `.h` file with `#define`s. (I haven't explored this step)

pvginkel, 2024-03-28:
I'm OK with closing the ticket if you prefer. A lot of effort was put into support for Kconfig, so I assumed that I just didn't understand how to use it in a stand alone project. If Kconfig can only be used in tandem with a framework like ESP-IDF, that's enough of an answer for me.

kisvegabor, 2024-03-28:
> If Kconfig can only be used in tandem with a framework like ESP-IDF, that's enough of an answer for me.

Really this is the case now. 

I'm closing this issue, however if interested is shown in this feature form a wider audience, we can consider implementing it.

closed 2024-03-28 by kisvegabor