lvgl/lvgl #5935: Can littlevgl porting to android

MyHubTo, 2024-03-21:
### Introduce the problem

LVGL is a good project, and we have a good development experience under Linux (through the cache area /dev/fb0). But recently we tried to port it to the Android system, but it could not be displayed normally (through /dev/graphics/fb0); does the porting currently support porting to the Android platform? Or are there any instructions?

### Proposal

are there any instructions or plan?

antayirp, 2024-03-21:
There is an article in chinese:

https://blog.csdn.net/jayyuz/article/details/123842040

kisvegabor, 2024-06-20:
I close this issue because there was no activity here for a while.

If you have remarks about this topic feel free to comment here and reopen the issue.

programadorthi, 2025-01-06:
I'm studying native development and I did a project working Android 24+.

https://github.com/programadorthi/android-native/tree/lvgl