lvgl/lvgl #5938: Optimize layout calculations

kisvegabor, 2024-03-21:
### Problem to solve

Based on https://github.com/lvgl/lvgl/issues/5901

Currently layouts are calculated in an iterative way: repeat until there is no change. It's not optimal as 
1. not predictable
2. with nested layouts it can take very long 

### Success criteria

There is a predictable layout recalculation flow with a know number of required passes.

TODO: add a target layout calculation time for a reference UI

### Solution outline

**Group the units like this**
- absolute unit: px
   - these sizes can be resolved independently of anything
- relative units: 
  - parent dependent unit: %, flex-grow
  - children dependent unit: `LV_SIZE_CONTENT`, `LV_GRID_CONTENT`
     - the position and size of all their children need to be resolved to calculate these values 

The idea is to resolve the `LV_SIZE_CONTENT` widgets first, as their size is required for the siblings and/or the parent. When the size of all siblings are resolved, update their positions. Based on that the  `LV_SIZE_CONTENT` sizes of the parent can be set too.

**List example**
- A list item has 3 children: 
   1. an icon with  `LV_SIZE_CONTENT` sizes
   2. a label with `LV_SIZE_CONTENT` height and flex-grow width
   3. a button with 30% width and 30px height.
- The list item has `LV_SIZE_CONTENT` height, and 100% width
- The list has a flex column layout to place the items
- The list's width is 50%
- The list is on the screen
- There are two list below each other with a flex column. The first have  `LV_SIZE_CONTENT` and the second has flex-grow height

We can calculate the sizes and positions like this:
1. Resolve all px size units (the screen and and all list button heights)
2. Try to resolve the size of the lists, but as `list1` has  `LV_SIZE_CONTENT` height, so its children needs to be resolved first. It's width can be set to 50%
3. Check `list_item1`: it has  `LV_SIZE_CONTENT` so resolve its children first, but set its with to 100%
4. Check the icon: it has  `LV_SIZE_CONTENT` size, so resolve its children first, but it has no children, so just set its size
5. Now there are no children dependent sizes horizontally, so update its children's sizes with flex
6. The label's height is  `LV_SIZE_CONTENT` but as the width is resolved it can be calculated now too. (The text might wrap into multiple lines)
7. Update the position of the 3 children with flex
8. As the children are resolved we can set the height of the list item
9. Do the same with all list items on `list1`
10. As all children of `list1` are resolved, we can set the position of the children
11. Now the height of `list1` can be set too
12. `list1` is resolved, `list2` has flex-grow, so we can calculate its size
13. Resolve the list items of `list2`
14. As both lists are ready we set their positions on the screen

### Rabbit holes

- % size on flex-grow: as flex-grow is also parent dependent it should be resolved first, so the % children can be calculated 
- flex-grow on % size: same
- % on  `LV_SIZE_CONTENT`: not possible, circular dependency (% child depends on the  `LV_SIZE_CONTENT` which depends on the % child)
- ?

### Testing

Very deep testing is required. We need to collect as much use cases as possible.

### Teaching

There is o difference on the API level

kisvegabor, 2024-09-05:
It's not on the [roadmap of v9.3](https://github.com/lvgl/lvgl/issues/6763) so I close it for now. Let's warm it up for a later release if there is an interest. 

If anyone is interested in picking it up, please write a comment.

closed 2024-09-05 by kisvegabor

kisvegabor, 2025-05-15:
Thank you for showing interest! 

Just to better understand: why is it interesting for you? Do you see performance issues?

vwheeler63, 2025-05-15:
@kisvegabor I also have interest in this:  I helped to build a "windowing UI framework" that was designed be generic.  I built the layout part of it, and have an eye for CPU optimization....  Just thinking ahead.  :-)