lvgl/lvgl #5942: Build error when using v9.1.0 with SDL and 2 Buffers

Ka1mes, 2024-03-22:
### LVGL version

9.1.0

### What happened?

I am using LVGL with SDL on Windows and just ported to version 9.1.0. The project would initially not build due to a small error in the lv_sdl_window.c file. 
lvgl\src\drivers\sdl\lv_sdl_window.c:402:47: error: 'ver_res' undeclared (first use in this function)
  402 |         dsc->fb2 = realloc(dsc->fb2, stride * ver_res);

The fix is to change: 
#if LV_SDL_BUF_COUNT == 2
        dsc->fb2 = realloc(dsc->fb2, stride * ver_res);
        memset(dsc->fb2, 0x00, stride * ver_res);
#endif

 to:

#if LV_SDL_BUF_COUNT == 2
        dsc->fb2 = realloc(dsc->fb2, stride * disp->ver_res);
        memset(dsc->fb2, 0x00, stride * disp->ver_res);
#endif 

### How to reproduce?

Just try to build using SDL with 2 Buffers and Windows backend.

C47D, 2024-03-22:
Hi, could you send a pull request with the fix?

kisvegabor, 2024-04-08:
Not stale

Ka1mes, 2024-04-19:
Merge request was approved

closed 2024-04-19 by Ka1mes