lvgl/lvgl #5945: Long press in LV_INDEV, data->state

keerthansys22, 2024-03-23:
### LVGL version

v8.3

### What happened?

I have linked my external key button to lvgl library using index_drv.type = LV_INDEV_TYPE_BUTTON, I can read my button press but the states included in the library are data->state = LV_INDEV_STATE_RELEASED and data->state = LV_INDEV_STATE_PRESSED. I didn't find the status for long press.

liamHowatt, 2024-03-23:
Hey @keerthansys22, the `LV_EVENT_LONG_PRESSED` event is generated automatically after the button state has been `LV_INDEV_STATE_PRESSED` for 400ms.

keerthansys22, 2024-03-24:
Thanks for your response @liamHowatt, how do I differentiate the long and short press? I'm supposed to perform some unique tasks for short and long press individually.

liamHowatt, 2024-03-24:
You can create an event handler for the thing being pressed and handle short and long presses accordingly.

Here is an example that handles different types of events
https://docs.lvgl.io/master/overview/event.html#handle-multiple-events

And here is a slightly modified version of that example that shows a way you could differentiate short presses and long presses
```c
static bool text_is_equal(const char *a, const char *b)
{
    return lv_strcmp(a, b) == 0;
}

static void event_cb(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t * label = lv_event_get_user_data(e);

    switch(code) {
        case LV_EVENT_PRESSED:
            lv_label_set_text(label, "Pressing now...");
            break;
        case LV_EVENT_RELEASED:
            if (text_is_equal(lv_label_get_text(label), "Pressing now...")) {
                /* handle short press */
                lv_label_set_text(label, "It was a short press!");
            } else {
                /* handle long press released */
                lv_label_set_text(label, "The long press was released");
            }
            break;
        case LV_EVENT_LONG_PRESSED:
            /* handle long press */
            lv_label_set_text(label, "It was a long press!");
            break;
        default:
            break;
    }
}

/**
 * Handle multiple events
 */
void lv_example_event_2(void)
{
    lv_obj_t * btn = lv_button_create(lv_screen_active());
    lv_obj_set_size(btn, 100, 50);
    lv_obj_center(btn);

    lv_obj_t * btn_label = lv_label_create(btn);
    lv_label_set_text(btn_label, "Click me!");
    lv_obj_center(btn_label);

    lv_obj_t * info_label = lv_label_create(lv_screen_active());
    lv_label_set_text(info_label, "Click the button!");

    lv_obj_add_event_cb(btn, event_cb, LV_EVENT_ALL, info_label);
}
```

Here are all the events you can handle
https://docs.lvgl.io/master/overview/event.html#input-device-events

And here is a related question
#5821

keerthansys22, 2024-03-25:
Hi @liamHowatt, I have tried the method you said but I think I'm missing the linking part of button press with event_cb function. How will I link? I have created a button with the coordinates which I initialized in lv_point_t point_array[] also. But the event_cb is not considering the events.

Meanwhile, is there any way we can detect the long press in data->state which is similar to data->state =  LV_INDEV_STATE_PRESSED?

keerthansys22, 2024-03-27:
Thanks for your response @liamHowatt, I will include my code below:-
```
. . . // indev initialisation
lv_indev_drv_init(&index_drv);
index_drv.type = LV_INDEV_TYPE_BUTTON;
index_drv.read_cb = button_read;

my_indev = lv_indev_drv_register(&index_drv);
static const lv_point_t points_array[] = {{15,100}};
lv_indev_set_button_points(my_indev, points_array);
. . .

// indev button read
static void button_read(lv_indev_drv_t * drv, lv_indev_data_t * data)
{
static uint32_t last_btn = 0;
int btn_pr = my_btn_read();   // detect the button press from I2C & returns 0
if(btn_pr >= 0)
{
last_btn = btn_pr;
data->state = LV_INDEV_STATE_PRESSED;
}
else
{
if(btn_pr != 0){
data->state = LV_INDEV_STATE_RELEASED;
btn_pr = 0;}
}
data->btn_id = last_btn;
}

// create a button on point_array[] coordinates
. . . 
btn = lv_btn_create(screen);
lv_obj_set_pos(btn, 15, 100);
lv_obj_add_event_cb(btn, event_cb, LV_EVENT_ALL, NULL);    // create a button event
. . . 

// widget button event handler
static void event_cb(lv_event_t * e)
{
lv_event_code_t code = lv_event_get_code(e);
lv_obj_t* label = lv_event_get_user_data(e);
switch(code)
{
case LV_EVENT_LONG_PRESSED:
lv_label_set_text(label, "The long press \r\n");
default:
break;
}
}
```

liamHowatt, 2024-03-27:
A few notes.

`data->btn_id` should always be set to `0` since there's only one indev button. This tells LVGL which button an event occurred for. You will want to always set it to `0` in the callback.

<br>

Change
```c
lv_obj_add_event_cb(btn, event_cb, LV_EVENT_ALL, NULL);    // create a button event
```
to
```c
lv_obj_t* label = lv_label_create(btn);
lv_obj_add_event_cb(btn, event_cb, LV_EVENT_ALL, label);    // create a button event
```
so that the callback gets the label object.

<br>

Does this make the label show your text?

keerthansys22, 2024-03-27:
Hi @liamHowatt, I have set the data->btn_id to 0 as you suggested. Even added the label and created a button event.

I observed that the callback is going inside the event_cb() function but not entering any switch cases!!

liamHowatt, 2024-03-27:
There are a lot of different kinds of events and unfortunately none of them may be press related.

For good measure, add more cases in case the press is so short it's not a long press.

```c
case LV_EVENT_LONG_PRESSED:
lv_label_set_text(label, "The long press \r\n");
break;
case LV_EVENT_PRESSED:
lv_label_set_text(label, "pressed");
break;
case LV_EVENT_RELEASED:
lv_label_set_text(label, "released");
break;
```

If that doesn't work, add some logs in `button_read` if you can, to make sure `LV_INDEV_STATE_PRESSED` is being set.

keerthansys22, 2024-03-28:
Hi @liamHowatt, I have added logs and I observed that program can go inside **button_read** and it can detect the **LV_INDEV_STATE_PRESSED**.

But in the **event_cb**, its going inside the function and no detecting any cases - **LV_EVENT_PRESSED** or **LV_EVENT_RELEASED** or **LV_EVENT_LONG_PRESSED** !!

liamHowatt, 2024-03-28:
I tested your code on v8.3 and v9.1 and I am seeing the long press message being shown on the label. The indev button coordinates are on the top-left corner of the widget button and it seems to be working as expected for me. I am getting the `LV_EVENT_LONG_PRESSED` event in the callback.

How do you know the switch cases are not being entered? Are you logging, debugging, etc.? Could you please send the event codes being logged? They can be cast to `int` to be logged and compared against [the enum values](https://github.com/lvgl/lvgl/blob/b4bdc9e482ec919d74bf5a8bb3a236a2f8b214de/src/core/lv_event.h#L29-L91).

Also, are you getting any compiler warnings?

This kind of troubleshooting can be tricky but I'm sure we're close to solving it 🙂️

keerthansys22, 2024-03-29:
I'm glad to see your response @liamHowatt,

Now I can see the changes in the label when button pressed and it's detecting the different cases. But it's not consistent!!  The **event_cb** function is called multiple times!! I'm missing something.

If you go through the logs, The last part of this thread gives you a better idea :)

I have logged the program as below - 

```
static void button_read(lv_indev_drv_t * drv, lv_indev_data_t * data)  
{
    int btn_pr = my_btn_read();
    if(btn_pr == 0)
    {
        last_btn = btn_pr;
        printf("last btn: %d\r\n",btn_pr);
        data->state = LV_INDEV_STATE_PRESSED;
        printf("PRESS DETECTED\r\n");
    }
    data->btn_id = last_btn;
}
```

```
static void event_cb(lv_event_t * e)
{
    printf("%s\r\n",__func__);
    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t* label = lv_event_get_user_data(e);
    printf("code value is :%d\r\n",code);
    switch(code)
    {
    case LV_EVENT_LONG_PRESSED:
        lv_label_set_text(label, "LP");
        printf("The long button is pressed\r\n");
        break;
 
    case LV_EVENT_PRESSED:
        lv_label_set_text(label, "OP");
        printf("The press in event \r\n");
        break;
 
    case LV_EVENT_RELEASED:
        lv_label_set_text(label, "RP");
        printf("The release in event \r\n");
        break;

    case LV_EVENT_SHORT_CLICKED:
        lv_label_set_text(label, "SC");
        printf("The short clicked in event\r\n");
        break;
 
    default:
        break;
    }
}
```
Initally the label is "HI", after the button press it is
```
btn = lv_btn_create(screen);
     lv_obj_t* label_pin = lv_label_create(btn);
     lv_obj_set_pos(btn, 15, 100);
     lv_label_set_text(label_pin, "HI");
     lv_obj_add_event_cb(btn, event_cb, LV_EVENT_ALL, label_pin);
```
The logs when the button is pressed for only once
```
last btn: 0
PRESS DETECTED : 1
event_cb
code value is :19
event_cb
code value is :1
**The press in event**
event_cb
code value is :34
event_cb
code value is :44
event_cb
code value is :44
event_cb
code value is :41
event_cb
code value is :44
event_cb
code value is :44
event_cb
code value is :19
event_cb
code value is :14
event_cb
code value is :2
event_cb
code value is :20
event_cb
code value is :26
event_cb
code value is :27
event_cb
code value is :21
event_cb
code value is :22
event_cb
code value is :23
event_cb
code value is :24
event_cb
code value is :25
event_cb
code value is :19
event_cb
code value is :8
**The release in event**
event_cb
code value is :4
**The short clicked in event**
event_cb
```

liamHowatt, 2024-04-01:
Hey keerthansys22,

Thanks for the many details and congrats on the partial success.

Here are the event code names from your log below. As you can see there are many different kinds of events and not all of them have to do with presses/releases. Don't worry about the event_cb being called many times. It's okay. If we look at just the press/release related ones, it looks like the button just hasn't been pressed for long enough. There's a "pressed", "pressing", "released", and then a "short click" event is also generated.

The issue may be your `button_read` callback. Please follow the advice I gave earlier.
> data->btn_id should always be set to 0 since there's only one indev button. This tells LVGL which button an event occurred for. You will want to always set it to 0 in the callback.

<br>

```
last btn: 0
PRESS DETECTED : 1
event_cb
code value is LV_EVENT_HIT_TEST
event_cb
code value is LV_EVENT_PRESSED
**The press in event**
event_cb
code value is LV_EVENT_CANCEL
event_cb
code value is LV_EVENT_SIZE_CHANGED
event_cb
code value is LV_EVENT_SIZE_CHANGED
event_cb
code value is LV_EVENT_SCREEN_LOAD_START
event_cb
code value is LV_EVENT_SIZE_CHANGED
event_cb
code value is LV_EVENT_SIZE_CHANGED
event_cb
code value is LV_EVENT_HIT_TEST
event_cb
code value is LV_EVENT_KEY
event_cb
code value is LV_EVENT_PRESSING
event_cb
code value is LV_EVENT_INDEV_RESET
event_cb
code value is LV_EVENT_DRAW_POST_BEGIN
event_cb
code value is LV_EVENT_DRAW_POST
event_cb
code value is LV_EVENT_COVER_CHECK
event_cb
code value is LV_EVENT_REFR_EXT_DRAW_SIZE
event_cb
code value is LV_EVENT_DRAW_MAIN_BEGIN
event_cb
code value is LV_EVENT_DRAW_MAIN
event_cb
code value is LV_EVENT_DRAW_MAIN_END
event_cb
code value is LV_EVENT_HIT_TEST
event_cb
code value is LV_EVENT_RELEASED
**The release in event**
event_cb
code value is LV_EVENT_SHORT_CLICKED
**The short clicked in event**
event_cb
```

keerthansys22, 2024-04-24:
Hey @liamHowatt, Yes its working and I'm sorry for replying late.

Thanks for your advice and support :)

closed 2024-04-24 by keerthansys22