lvgl/lvgl #5946: [v9.1.0] canvas example 6 & 8 shows incorrect results?

BlynkGO, 2024-03-24:
### LVGL version

v9.1.0

### What happened?

[image]

I have tried canvas examples 6 and 8, 
but I cannot display images and vectors.

How to solve this problem?
Thank you.

liamHowatt, 2024-03-24:
Thanks @BlynkGO,

- The star image needs to be fixed. It will be fixed by this PR #5800
- `lv_canvas_finish_layer(canvas, &layer);` just needs to be added at the end. https://github.com/liamHowatt/lvgl/commit/1c80f9e6575132620c039a7c5fde9e2c9d539e2a I've been waiting to create a PR for it after the other one gets merged.

You can add the missing line at the end and the image will appear but it will be distorted. You can use a different image for now, or grab the fix from the PR.

BlynkGO, 2024-03-25:
Thank you very much @liamHowatt 

`canvas example 6`  the canvas can display image now.
[image]

and `canvas example 8` (canvas draw vector) still can't display anything

[image]

How to solve the example 8 issue?

liamHowatt, 2024-03-25:
Hey @BlynkGO, that example uses vector graphics. Make sure you have `LV_USE_VECTOR_GRAPHIC` set to `1` and have either `LV_USE_THORVG_INTERNAL` or `LV_USE_THORVG_EXTERNAL` set to `1` in your lv_conf.h.

For internal, make sure you compile and link the files in src/libs/thorvg. If you're using cmake, add `lvgl_thorvg` to your `target_link_libraries`

BlynkGO, 2024-03-26:
Hi @liamHowatt 

Thanks for the reply.

Now I'm using on Platformio for simulation.

When I config  platformio.ini by the following addition.

```
build_flags = 
  -D LV_USE_VECTOR_GRAPHIC=1
  -D LV_USE_THORVG_INTERNAL=1
```

It occurs some compile error.

[image]

I don't know how to link the files in src/libs/thorvg 
How to solve the issue?

BlynkGO, 2024-03-27:
@liamHowatt 

Thank you very much. It's worked!!!

closed 2024-03-30 by BlynkGO