lvgl/lvgl #5952: Containing Custom Buffers in Page/Screen Function

fatihozen3, 2024-03-25:
### Introduce the problem

Hi, I want to use some buffers in my screen's page. But I've encountered a problem while changing screens. How should I contain a buffer in my screen function ?

Let's say I have a function and objects as below:

```
static lv_obj_t *text = NULL;
lv_obj_t *main;

void setup_scr_main(void)
{
	main = lv_obj_create(NULL);
	lv_obj_set_size(main, 800, 480);
	lv_obj_set_scrollbar_mode(main, LV_SCROLLBAR_MODE_OFF);
	lv_obj_set_style_bg_opa(main, 0, LV_PART_MAIN|LV_STATE_DEFAULT);
	
	text = lv_label_create(main);
	lv_label_set_text(text, buffer);
	lv_label_set_long_mode(text, LV_LABEL_LONG_WRAP);
	lv_obj_set_pos(text, 100, 13);
	lv_obj_set_size(text, 130, 20);
	lv_obj_set_style_text_font(text, font, LV_PART_MAIN|LV_STATE_DEFAULT);
	lv_obj_set_style_text_color(text, lv_color_hex(0x000000), LV_PART_MAIN|LV_STATE_DEFAULT);
	
	lv_obj_update_layout(main);
```

I will fill buffer with below function 

```
void unicode_to_uft16(uint8_t * buf, uint16_t size, uint16_t * received_buf)
{
	for(int i = 0;i<size;i++)
	int i=0;
	{
		static int buf_index = 0;

		if(received_buf[0]<0x80)
				{buf[buf_index++] = (uint8_t) received_buf[0];}
		else
		{
				buf[buf_index++] = (uint8_t)(0xc0 | (received_buf[0] >> 6));
				buf[buf_index++] = (uint8_t)(0x80 | (received_buf[0] & 0x3f));
		}
	}
}
```

Buffers and how I call function:

```
uint8_t buffer[2] = {0};
uint16_t received_buffer[2] = {0x1ff,'\0'};
unicode_to_uft16(&buffer,size, &received_buffer);
```

While changing screens as below I encounter a problem as MemManage_Handler after 3rd or 4th screen change. How should I write buffer to text even if I change screen without a problem? When I don't change screen I will see it on the screen as expected.

```
unicode_to_uft16(&buffer,size, &received_buffer);
setup_scr_main();
lv_screen_load_anim(main, LV_SCR_LOAD_ANIM_NONE, 0, 0, true);
```

fatihozen3, 2024-03-26:
The problem is static int buf_index = 0; I made it non-static, it works now. Someone may use this  function to convert ascii characters to utf8 format.

kisvegabor, 2024-03-27:
Hi,

As you have found a solution, please close this issue.

closed 2024-03-27 by kisvegabor