lvgl/lvgl #5954: LVGL 9 Group missing function `lv_group_set_click_focus`

xamix, 2024-03-25:
### LVGL version

v9.1.0

### What happened?

Hi,

I use encoder and touch inputs on my project, and I want to always see which button is actually focused to use with encoder.

It seem in the previous version there was a function called `lv_group_set_click_focus` as said in this [issue](https://github.com/lvgl/lvgl/issues/533#issuecomment-448117546) to let buttons focused on click with touch.

Is there a solution to force the currently focused object to be shown with outline at each time in V9.1.0?
This seem to be related on the type of input device which show focus only on ENCODER and KEYPAD input in [lv_obj.c](https://github.com/lvgl/lvgl/blob/9dec22dfaa01fe65904c65ece711e350624182a4/src/core/lv_obj.c#L652).

Regards.

liamHowatt, 2024-03-28:
Hey @xamix,

I came to the same conclusion about that line in lv_obj.c.

`LV_STATE_FOCUS_KEY` is the state that themes and styles are are supposed to add an outline for when it's set. POINTERs focus objects the same way ENCODERS and KEYPADS do, except that ENCODERS and KEYPADS set the `LV_STATE_FOCUS_KEY` state on the object.

This seems to work for getting the outline with a POINTER indev.

```c
static void pointer_click_event_cb(lv_event_t * e)
{
    lv_obj_t * obj = lv_event_get_param(e);
    lv_obj_add_state(obj, LV_STATE_FOCUS_KEY);
}

lv_indev_add_event_cb(pointer_indev, pointer_click_event_cb, LV_EVENT_CLICKED, NULL);
```

xamix, 2024-03-28:
Hi @liamHowatt,

Thank you for your workaround.

However it is not as simple as it since we also need to remove this `LV_STATE_FOCUS_KEY` on the previous clicked object, otherwise multiple object will have the outline enabled.
Also at start when nothing was already clicked, then nothing is outlined even if something is currenty focused.

Regards.

kisvegabor, 2024-03-28:
Hi,

`LV_STATE_FOCUS_KEY` is added only when a keypad or encoder focused the widget, however `LV_STATE_FOCUSED` is always added and removed for any kind of focusing. So - if I understand the issue correctly - all you need to do, is  using `LV_STATE_FOCUSED` instead of `LV_STATE_FOCUS_KEY`.

liamHowatt, 2024-03-28:
xamix
> we also need to remove this `LV_STATE_FOCUS_KEY` on the previous clicked object, otherwise multiple object will have the outline enabled.

I am seeing only one set of things outlined at once when I run that snippet. The object(s) being unfocused automatically have that state removed so they aren't outlined anymore.

<br>

xamix
> Also at start when nothing was already clicked, then nothing is outlined even if something is currenty focused.

Are you basically asking for the outline to be set when anything is focused regardless of the indev?

This does that.
```diff
diff --git a/src/core/lv_obj.c b/src/core/lv_obj.c
index 682706ed3..0b1f0453b 100644
--- a/src/core/lv_obj.c
+++ b/src/core/lv_obj.c
@@ -648,8 +648,7 @@ static void lv_obj_event(const lv_obj_class_t * class_p, lv_event_t * e)
         lv_indev_t * indev = lv_indev_active();
         if(indev == NULL) indev = lv_event_get_indev(e);
 
-        lv_indev_type_t indev_type = lv_indev_get_type(indev);
-        if(indev_type == LV_INDEV_TYPE_KEYPAD || indev_type == LV_INDEV_TYPE_ENCODER) state |= LV_STATE_FOCUS_KEY;
+        state |= LV_STATE_FOCUS_KEY;
         if(editing) {
             state |= LV_STATE_EDITED;
             lv_obj_add_state(obj, state);
```

Maybe we could make this a configurable thing.

<br>

kisvegabor
> So - if I understand the issue correctly - all you need to do, is using `LV_STATE_FOCUSED` instead of `LV_STATE_FOCUS_KEY`.

Setting `LV_STATE_FOCUSED` in the callback didn't create the outline effect for me.

kisvegabor, 2024-04-02:
> Setting LV_STATE_FOCUSED in the callb

In `lv_theme_default.c` I've replaced all `LV_STATE_FOCUS_KEY`s with `LV_STATE_FOCUSED`, not created a default group, and I saw the outlines as the widgets ere clicked.

[image]

closed 2024-05-06 by lvgl-bot