lvgl/lvgl #5955: LVGL 9 lv_group_add_obj unecessary steps?

xamix, 2024-03-25:
### LVGL version

V9.1.1

### What happened?

Hi,

I need some clarification on [lv_group_add_obj](https://github.com/lvgl/lvgl/blob/9dec22dfaa01fe65904c65ece711e350624182a4/src/core/lv_group.c#L116) function.

At line [123](https://github.com/lvgl/lvgl/blob/9dec22dfaa01fe65904c65ece711e350624182a4/src/core/lv_group.c#L123) there is a a call to be sure the object is removed from its current group.
However at line [126](https://github.com/lvgl/lvgl/blob/9dec22dfaa01fe65904c65ece711e350624182a4/src/core/lv_group.c#L126) there is a check to do not add the object twice, this is unecessary since it was removed previously?
Also at line [135](https://github.com/lvgl/lvgl/blob/9dec22dfaa01fe65904c65ece711e350624182a4/src/core/lv_group.c#L135) it check if the object is already in a group and focused then it refocus it, but it will never occur if the object was previously removed from it's group?

Regards

kisvegabor, 2024-04-03:
True! In light of that [this whole section](https://github.com/lvgl/lvgl/blob/9dec22dfaa01fe65904c65ece711e350624182a4/src/core/lv_group.c#L125-L142) can be removed. Could you send a Pull request?

kisvegabor, 2024-04-25:
I've opened #6124 to fix it.