lvgl/lvgl #5972: analogclock is also missing!

xinsuinizhuan, 2024-03-27:
### LVGL version

9.0

### What happened?

In 9.3 has the analogclock and the meter, but now all of them is missing. But they are also useful. We hope to optimize function and add new function, but not missing.

liamHowatt, 2024-03-27:
> In 8.3 has the analogclock and the meter, but now all of them is missing.

Not for much longer 😉️

https://github.com/lvgl/lvgl/pull/5937#issuecomment-2021718961

There is an analog clock example using the `lv_scale` in this PR. If you have any feedback for the PR before it gets merged please comment.

You can do an analog clock with one hand/needle right now, before this PR, if you want. Most of the example applies to the current `master`.

xinsuinizhuan, 2024-03-28:
where is the example code?

liamHowatt, 2024-03-28:
New example from the PR:
https://github.com/lvgl/lvgl/blob/3df3e6878b06275f6ea57595e660d148dfb18bf2/examples/widgets/scale/lv_example_scale_6.c

It's here but it uses features that aren't merged into `master` yet. You can pull that branch locally and use it in your application or you can use this version of it that I modified to work on the current `master`. You should be able to use this with your current version of LVGL.

New example from the PR **modified to work on the current master**:
https://gist.github.com/liamHowatt/7bf7c0fd7e593ee01930531c3a8180b0

kisvegabor, 2024-04-12:
An analog clock example is added: https://docs.lvgl.io/master/widgets/scale.html#example

Closing this issue.

closed 2024-04-12 by kisvegabor