lvgl/lvgl #5986: Some stange screentearing in a stm32f769 discovery

OscarMonedero, 2024-03-28:
### LVGL version

v9.1.1

### What happened?

Hi,

I've detect this strange bug like a screen tearing, it happens when i try to avoid the screen tearing by using the TFT_NO_TEARING with ZONES = 4.

Any idea for how i can solve this?
Thanks

https://github.com/lvgl/lvgl/assets/80752001/3149770c-d349-47ec-b90b-5531a0e87057

[image]

https://github.com/lvgl/lvgl/assets/80752001/1009d05a-4cd0-453f-959c-c0b4033000cc

kisvegabor, 2024-04-03:
I've updated LVGL in the `lv_port_stm32f769` repo but I can't see the tearing. Have modified anything else in the project (except enabling `TFT_NO_TEARING`)?

OscarMonedero, 2024-04-04:
i didn't, it could be a hardware problem? or maybe a bad configuration with the ide?

OscarMonedero, 2024-04-04:
im trying to enable 8 zones for sthe screen to try to minimize the problem, but i don't know which balue should have scanlines in this case:
#if ZONES == 2
	uint16_t scanline = 200;
#elif ZONES == 4
	uint16_t scanline = 283;
#elif ZONES == 8
	uint16_t scanline = 360;

kisvegabor, 2024-04-05:
I don't think it's a hardware issue, but don't have any ideas either :slightly_frowning_face: 

Just an absurd one: Eclipse already tricked me a few times and haven't compiled the projects automatically when I changed something. So are you sure that the project is rebuilt with the changed  `TFT_NO_TEARING` setting?

kisvegabor, 2024-04-15:
I've created the driver based on an ST example, and just used the values from there. You also have a 800x480 display, right?

OscarMonedero, 2024-04-16:
yes, i have the screen that comes with the kit

kisvegabor, 2024-04-19:
Phew, I'm sorry but I have no idea :cry:

OscarMonedero, 2024-05-06:
don't worry, thanks!!
I'm also trying now to use LVGL in a stm32u5g9j-dk1. Is there any port for it or plan to port it? 
If hte answer is no, could you help me on what i need to use LVGL on it?

kisvegabor, 2024-06-20:
We have just published [this project](https://github.com/lvgl/lv_port_riverdi_stm32u5) for a Riverdi STM32U5 board. 
I hope it helps.

closed 2024-08-20 by lvgl-bot