lvgl/lvgl #5987: Cannot set rounded corner to image, but can set it to a container

pvginkel, 2024-03-28:
### LVGL version

v9.1.0

### What happened?

I'm attempting to add rounded corners to an image. The documentation on image says this is possible. If I set rounded corners to the image, they're not applied. If I however put the image into a container, the rounded corners are applied.

I'm using the Windows simulator as the basis for testing this.

### How to reproduce?

This works:

```c
auto image_cont = lv_obj_create(cont);
lv_obj_remove_style_all(image_cont);
lv_obj_set_style_radius(image_cont, lv_dpx(8), LV_PART_MAIN);
lv_obj_set_style_clip_corner(image_cont, true, LV_PART_MAIN);
lv_obj_set_size(image_cont, LV_PCT(100), LV_SIZE_CONTENT);

auto image = lv_image_create(image_cont);
lv_obj_set_width(image, LV_PCT(100));
```

This does not:

```c
auto image = lv_image_create(cont);
lv_obj_set_style_radius(image , lv_dpx(8), LV_PART_MAIN);
lv_obj_set_style_clip_corner(image, true, LV_PART_MAIN);
lv_obj_set_width(image, LV_PCT(100));
```

kisvegabor, 2024-04-03:
Hi,

Could you show where did you find it in the docs? [Here](https://docs.lvgl.io/master/widgets/image.html) I can't see anything related.

"Clip corner" clips the children of the widget, but the image is not the children of itself.

pvginkel, 2024-04-03:
Apologies, I used the wrong version of the documentation to try this. The latest version these remarks show in is the 8.4 version of the documentation:

https://docs.lvgl.io/8.4/widgets/core/img.html

> Rounded image
> 
> You can use lv_obj_set_style_radius to set radius to an image, and enable lv_obj_set_style_clip_corner to clip the content to rounded rectangle or circular shape. Please note this will have some negative performance impact to CPU based renderers.

This text does not appear in version 9 of the documentation.

kisvegabor, 2024-04-05:
At least the docs is correct :slightly_smiling_face: 
I've just tested it in v8.4 and it worked well for me there. 

Do consider the use of a wrapper widget an acceptable workaround?

pvginkel, 2024-04-06:
I do. Would it make sense to have a replacement for the 8.4 documentation section with the images documentation with a sample? I could have a stab at that.

kisvegabor, 2024-04-06:
> I could have a stab at that.

Having an example and a few sentences about it in the v9.1 docs would be really useful. Please also mention that for `clip_corner` LVGL allocates two buffer for the upper and lower part of the image with ARGB8888 (4 byte) color depth. It means for a 100x100 image with 15 px radius 2 pieces of 100 px x 15 px  x 4 bytes = 6kB each temporal buffer will be allocated.
If there is no memory to allocate the second buffer LVGL wait until the previous is ready, and try to allocate again.

closed 2024-05-02 by lvgl-bot