lvgl/lvgl #5991: Encoder does not fire LV_EVENT_LONG_PRESSED

aeroniemi, 2024-03-29:
### LVGL version

v9.1.0

### What happened?

I have an encoder indev. It successfully rotates and handles short clicks, but long clicks does not issue LV_EVENT_LONG_PRESSED, although they do issue LV_EVENT_LONG_PRESSED_REPEATED regularly

### How to reproduce?

```c

```

kisvegabor, 2024-04-03:
Hi,

I could reporduce the issue. @liamHowatt could you take a deeper look at it?

liamHowatt, 2024-04-03:
It has to do with the editability of the object being pressed. This logic is exclusive to the encoder. It's in `indev_encoder_proc` of lv_indev.c. If the object is _not_ editable or scrollable, everything works as expected. However events are inconsistent when it's editable/scrollable. Editing is entered+exited correctly but it seems the emitted events are a bit illogical.

Pressing encoder ENTER on for 1000 ms and off for 1000s repeating. Here is two cycles of that.

```
(press)
PRESS (emitted this time because the object is "editing")
(no LONG_PRESS emitted because the object is editable or scrollable. the object becomes "not editing")
REPEAT
REPEAT
REPEAT
REPEAT
REPEAT
(release 1000 ms after the press)
(no RELEASE emitted because the object is editable or scrollable)
(1000 ms passes)
(press)
(no PRESS emitted this time because the object is "not editing")
(no LONG_PRESS emitted because the object is editable or scrollable. the object becomes "editing")
REPEAT
REPEAT
REPEAT
REPEAT
REPEAT
(release 1000 ms after the press)
(no RELEASE emitted because the object is editable or scrollable)
(repeat from top indefinitely)
```

It's clear to me that this is intentional logic in `indev_encoder_proc` and not just a bug. How come we choose not to emit events if the target is editable/scrollable? Is this something we want to change now?

kisvegabor, 2024-04-05:
Phew, it's really messy. :sweat_smile: 

So the simplified logic is this:
- On click go to edit mode, but do not sent the click, release, etc events, as the click was used to go to edit mode and not for the user. (i.e. it's not a button click)
- On clicking again: send it normally to the being edited widget
- On long press: leave edit mode, but do not send long press, repeat and release events to widget as they was used to change mode.
 
In the code I think it's correct that for editable widget we don't send long press event:
1. In navigation mode, it's ignored, as the click was switching to edit mode
2. In edit mode long press is used go back to navigation mode.

For non editable widget the long press should be sent normally. @aeroniemi What kind of widget have you used to test the long press?

aeroniemi, 2024-04-05:
> For non editable widget the long press should be sent normally. @aeroniemi What kind of widget have you used to test the long press?

In my case I've assigned an event callback directly to the encoder indev that I expected would just fire every time the indev experienced the event (irrespective of what was/was not in focus or edit mode). I haven't seen it fire the long_press on any of the screens I've used so far (which contain combinations of: only labels, labels+buttons, arcs + labels + imagebuttons...).

Effectively I've aimed for this UX:
- Anytime you long press you return to the main menu, irrespective of what else is currently occuring (with a LV_INDEV_DEF_LONG_PRESS_TIME of 2000).

I'm guessing that wasn't how it was originally intended, but it seems to make the most sense for my use case (a M5Dial, so rotary encoder plus round touchscreen)

kisvegabor, 2024-04-06:
> lv_indev_add_event_cb(indev_encoder, long_press_cb, LV_EVENT_LONG_PRESSED, NULL); // never fires

Here you are really attaching the event to input device and we could send the events to input device even if we don't send them to the widget. Do you have time an interest to look into it an send a Pull request with the fix?

aeroniemi, 2024-04-06:
> Do you have time an interest to look into it an send a Pull request with the fix?

Happy to give a go to creating a PR for this, but I'm not wholly clear on what the expected behavior is. As I understand it, when a callback is attatched directly to an indev (``lv_indev_add_event_cb``) it is supposed to fire every time the event occurs, whilst those attatched to widgets (``lv_obj_add_event_cb``) are supposed to fire only when not used for changing the navigation/edit mode.

Therefore the fix would be to make the ``LV_EVENT_LONG_PRESS`` event fire if the callback is added through ``lv_indev_add_event_cb`` but not if added through ``lv_obj_add_event_cb``
 Is that correct?

kisvegabor, 2024-04-08:
Great, thank you! 

>  As I understand it, when a callback is attatched directly to an indev [...]

Correct!

`Therefore the fix would be to make the LV_EVENT_LONG_PRESS event fire if the callback is added through lv_indev_add_event_cb but not if added through lv_obj_add_event_cb
Is that correct?`

Yes, that's the case. `lv_obj_send_event` and `lv_indev_send_event` can be used as needed.

kisvegabor, 2024-04-15:
Thank you for the PR. Let's continue there!

PR #6064 (2024-04-17): fix(encoder): always fire LV_EVENT_LONG_PRESSED to indev callback
### Description of the feature or fix
Fixes #5991 

Currently LV_EVENT_LONG_PRESSED isn't fired by encoder buttons if the current obj is editable or scrollable - this fixes that by ensuring that it runs every time irrespective of the obj state.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new opti [... cut]
closed 2024-04-15 by kisvegabor