lvgl/lvgl #5994: “intrin.h”: No such file or directory

GianMeng, 2024-03-29:
### LVGL version

v9.1.0

### What happened?

When I built LVGL code in EDK2 env unther windows 10, got the **error**: `src\stdlib\builtin\lv_tlsf.c(110): fatal error C1083: can not open include file: “intrin.h”: No such file or directory`. 

It seems <intrin.h> is necessary since LVGL v9. However, our env does not support to include “intrin.h” file, though we can build code with vs2019. 

Do I have any way to evade using “intrin.h” file for update LVGL v9 from LVGL v8 ?

### How to reproduce?

Here is the .inf file used to build lvgl library, including:
``` C 
[Defines]
  INF_VERSION                    = 0x00010005
  BASE_NAME                      = lvgl
  FILE_GUID                      = 1afb3f8d-da69-39f8-2058-6857309dfb10
  MODULE_TYPE                    = UEFI_APPLICATION
  VERSION_STRING                 = 1.0
  LIBRARY_CLASS                  = lvglLib

[Sources]
    lvgl.h
    lv_conf.h

    src/lvgl.h
    src/lvgl_private.h
    src/lv_api_map_v8.h
    src/lv_api_map_v9_0.h
    src/lv_conf_internal.h
    src/lv_conf_kconfig.h
    src/lv_init.c
    src/lv_init.h
    
    src/core/lv_global.h
    src/core/lv_group.c
    src/core/lv_group.h
    src/core/lv_obj.c
    src/core/lv_obj.h
    src/core/lv_obj_class.c
    src/core/lv_obj_class.h
    src/core/lv_obj_draw.c
    src/core/lv_obj_draw.h
    src/core/lv_obj_event.c
    src/core/lv_obj_event.h
    src/core/lv_obj_id_builtin.c
    src/core/lv_obj_pos.c
    src/core/lv_obj_pos.h
    src/core/lv_obj_property.c
    src/core/lv_obj_property.h
    src/core/lv_obj_scroll.c
    src/core/lv_obj_scroll.h
    src/core/lv_obj_style.c
    src/core/lv_obj_style.h
    src/core/lv_obj_style_gen.c
    src/core/lv_obj_style_gen.h
    src/core/lv_obj_tree.c
    src/core/lv_obj_tree.h
    src/core/lv_refr.c
    src/core/lv_refr.h

    src/display/lv_display.c
    src/display/lv_display.h
    src/display/lv_display_private.h

    src/draw/lv_draw.c
    src/draw/lv_draw.h
[... 368 lines cut ...]
    src/widgets/textarea/lv_textarea.c
    src/widgets/textarea/lv_textarea.h
    src/widgets/tileview/lv_tileview.c
    src/widgets/tileview/lv_tileview.h
    src/widgets/win/lv_win.c
    src/widgets/win/lv_win.h

[Packages]
  Build/GUIPkg.dec  
  MdePkg/MdePkg.dec
  ShellPkg/ShellPkg.dec

[LibraryClasses]
  UefiApplicationEntryPoint
  UefiBootServicesTableLib
  UefiRuntimeServicesTableLib
  DebugLib
  BaseLib
  BaseMemoryLib
  MemoryAllocationLib
  
[BuildOptions]
  MSFT:*_*_*_CC_FLAGS = /w /Od
  MSFT:DEBUG_*_IA32_CC_FLAGS = /Od
```

kisvegabor, 2024-04-02:
Two ideas from [here](https://stackoverflow.com/questions/2520683/how-to-cope-with-intrin-h-no-such-file-or-directory):
1. Use `#include <x86intrin.h>` add flag to `gcc -msse2`.
2. Just remove this include.

GianMeng, 2024-04-08:
> Two ideas from [here](https://stackoverflow.com/questions/2520683/how-to-cope-with-intrin-h-no-such-file-or-directory):
> 
> 1. Use `#include <x86intrin.h>` add flag to `gcc -msse2`.
> 2. Just remove this include.

Thanks a lot! It works in ways 2 cause my env does not allow the use of `gcc`.

kisvegabor, 2024-04-08:
Interestingly this issue didn't came for others yet. Let's dive deeper if it's reported again. 

Closing this issue until that.

closed 2024-04-08 by kisvegabor