lvgl/lvgl #5998: [v9.1.0]   [v9.1.0]  The roller example3 cannot be compiled

BlynkGO, 2024-03-29:
### LVGL version

v9.1.0

### What happened?

 The roller example3   on v8.4.0 can be compiled.
 However on v9.1.0  The roller example3 cannot be compiled.
 
 The error as the following
 
 [image]

How to solve the issue?
Thank you.

kisvegabor, 2024-04-02:
It was removed from LVGL v9 as the masking was updated. In v9, to make the masks easier to use with GPUs, they can be  only bitmap images now. See [here](https://docs.lvgl.io/master/overview/style-props.html#bitmap-mask-src).

BlynkGO, 2024-04-03:
@kisvegabor 
Thanks for the answer.

I have another issue. 
The issue I have is how to compile `rlottie.` 
I've set `LV_USE_RLOTTIE` to 1 successfully, 
but encountered the following error:

[image]

How to solve this issue?
Thank you.

kisvegabor, 2024-04-04:
It can be really complicated to compile rlottie, however I'm planning to add a new Lottie engine which is built in to LVGL, so it can be compiled as any other LVGL files.

See here https://github.com/lvgl/lvgl/issues/6021

kisvegabor, 2024-04-19:
See the PR for Lottie here: https://github.com/lvgl/lvgl/pull/6081

closed 2024-04-19 by kisvegabor

MichaelDu9226, 2024-04-23:
There was no problem with the lv example lottie approve test, but it crashed when I replaced it with another lottie. json.

Link to another lottie.json:
https://lottiefiles.com/animations/toaster-cImjAFoBok

But when I change the toaster lottie buffer to a size of 40x40, it's ok.

The attachment contains the source code and JSON:
[toast_lottie_test.zip](https://github.com/lvgl/lvgl/files/15076423/toast_lottie_test.zip)

kisvegabor, 2024-04-23:
Thanks for testing it! 
It works for with a larger size too. Do you know where it's crashing?

MichaelDu9226, 2024-04-24:
> Thanks for testing it! It works for with a larger size too. Do you know where it's crashing?
The hardware platform I am using is ESP32-S3-LCD-EV-Board, and the software is based on ESP-BSP https://github.com/espressif2022/esp-bsp/tree/feature/add_RGB_support
The error reported is either main task or timer task overflow
After increasing the stack size, the toaster lottie now supports 220x220.
But the canvas seems to have been cropped.
There are two questions here
1. Why is the stack size so high? Can it be optimized?
2. The problem of the canvasbeing cropped
[image]

kisvegabor, 2024-04-24:
The cropping issue should be solved. Please update LVGL from the PR.

What is the current stack size?