lvgl/lvgl #6000: lvgl/src/../src/lv_conf_internal.h:59:18: fatal error: ../../lv_conf.h: No such file or directory

rtek1000, 2024-03-30:
### LVGL version

The latest version available on GitHub

### What happened?

I bought a display and I'm trying to compile the demo project, but an error occurs that a file is missing. Does anyone know how to generate this file?

https://en.aliexpress.com/item/1005005865107357.html

lvgl/src/../src/lv_conf_internal.h:59:18: fatal error: ../../lv_conf.h: No such file or directory

### How to reproduce?

https://github.com/NoosaHydro/2.4inch_ESP32-2432S024/blob/main/1-Demo/Demo_Arduino/1_2_Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch.ino

C47D, 2024-04-02:
Hi,

Did you followed the instructions available in the docs? They mention where to place the `lv_conf.h` file the project is missing.

C47D, 2024-04-03:
Is this one https://docs.lvgl.io/master/integration/framework/arduino.html

rtek1000, 2024-04-09:
> Is this one https://docs.lvgl.io/master/integration/framework/arduino.html

Thanks, I'll test it as soon as possible.

rtek1000, 2024-04-09:
Hi, I tried following the instructions in the document, but I didn't find the folder path.

Then I tried to put the file in the /lvgl/src folder, then it compiled.

But now it is asking for the file 'demos/lv_demos.h'

-----

Is there no way for the user to generate these files automatically? Or is the example in question customized beyond what the lvgl library proposes?

kisvegabor, 2024-04-24:
I'm sorry, I missed your reply :slightly_frowning_face: 

> Hi, I tried following the instructions in the document, but I didn't find the folder path.

Which is the folder that you haven't found?

rtek1000, 2024-04-25:
> Which is the folder that you haven't found?

----

> the instructions in the document:

> "into the Arduino Libraries directory next to the lvgl library folder"

> Then I tried to put the file in the /lvgl/src folder, then it compiled.

kisvegabor, 2024-04-25:
It should look like this:
[image]

After that the demos shall work too.

rtek1000, 2024-04-28:
> It should look like this: [...]

Thank you for your attention, in the case of the /library folder, as it houses all shared libraries, wouldn't it be confusing for all libraries to make this type of use of the libraries' root folder?

rtek1000, 2024-05-02:
> It's not ideal for sure, but I have no better idea. 🙁 In the Arduino build system we don't have too much flexibility.

I don't really understand the reason for using the root folder, since the lvgl folder has a file called lv_conf_template.h, wouldn't it be easier for the user to just create a copy of that lv_conf_template.h file in this same lvgl folder and rename it to lv_conf.h, keeping both files inside the lvgl folder?

kisvegabor, 2024-05-07:
Hm, interesting idea. Normally our intention is to keep `lvgl` untouched to make it easier to update in the future. However in case of Arduino it might be really simpler this way. I just don't know how to make try including `lv_conf.h` from 2 locations in [`lv_conf_internal.h`](https://github.com/lvgl/lvgl/blob/master/src/lv_conf_internal.h#L59). 

We can consider using 
```c
#if defined(__ARDUINO__)
#include "../lv_conf.h" //instead of ../../lv_conf.h
#endif
```

but it will break existing projects.

rtek1000, 2024-05-07:
Indeed, now that it is in use in this way, projects should break. It may be interesting to use #warning to guide the user to use the alternative path. I don't know if there is a way to test the existence of the file using #if, perhaps adding some version definition for "watershed" control, from that version the file is located in that folder.

kisvegabor, 2024-05-13:
If I'm not mistaken your `lv_conf.h` will be overwritten when you change LVGL version. What do you think about it?

rtek1000, 2024-06-02:
It is interesting to consult the Arduino forum to find out what the alternatives are, I know that there are several possible installation locations for Arduino IDE libraries. For example, the folder inside the IDE installation will be overwritten when updating the IDE.

rtek1000, 2024-07-18:
Hello, I tried to compile again the [example](https://github.com/NoosaHydro/2.4inch_ESP32-2432S024/blob/main/1-Demo/Demo_Arduino/1_2_Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch.ino), and some errors occurred. Have the codes changed a lot since 9 months ago when the example was posted?

`error: 'lv_disp_draw_buf_t' does not name a type; did you mean 'lv_draw_buf_t'`

Complete log:

```
/home/user/ESP32/ESP32-DISPLAY-2.4/2.4inch_ESP32-2432S024/1-Demo/Demo_Arduino/1_2_Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch.ino:16:8: error: 'lv_disp_draw_buf_t' does not name a type; did you mean 'lv_draw_buf_t'?
 static lv_disp_draw_buf_t draw_buf;
        ^~~~~~~~~~~~~~~~~~
        lv_draw_buf_t
/home/user/ESP32/ESP32-DISPLAY-2.4/2.4inch_ESP32-2432S024/1-Demo/Demo_Arduino/1_2_Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch.ino:127:20: error: variable or field 'my_disp_flush' declared void
 void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
                    ^~~~~~~~~~~~~
/home/user/ESP32/ESP32-DISPLAY-2.4/2.4inch_ESP32-2432S024/1-Demo/Demo_Arduino/1_2_Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch.ino:127:20: error: 'lv_disp_drv_t' was not declared in this scope
/home/user/ESP32/ESP32-DISPLAY-2.4/2.4inch_ESP32-2432S024/1-Demo/Demo_Arduino/1_2_Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch.ino:127:20: note: suggested alternative: 'lv_fs_drv_t'
 void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
                    ^~~~~~~~~~~~~
                    lv_fs_drv_t
/home/user/ESP32/ESP32-DISPLAY-2.4/2.4inch_ESP32-2432S024/1-Demo/Demo_Arduino/1_2_Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch.ino:127:35: error: 'disp' was not declared in this scope
 void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
                                   ^~~~
/home/user/ESP32/ESP32-DISPLAY-2.4/2.4inch_ESP32-2432S024/1-Demo/Demo_Arduino/1_2_Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch.ino:127:35: note: suggested alternative: 'dup'
 void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
                                   ^~~~
                                   dup
/home/user/ESP32/ESP32-DISPLAY-2.4/2.4inch_ESP32-2432S024/1-Demo/Demo_Arduino/1_2_Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch.ino:127:41: error: expected primary-expression before 'const'
 void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
                                         ^~~~~
/home/user/ESP32/ESP32-DISPLAY-2.4/2.4inch_ESP32-2432S024/1-Demo/Demo_Arduino/1_2_Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch.ino:127:75: error: expected primary-expression before '*' token
 void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
                                                                           ^
/home/user/ESP32/ESP32-DISPLAY-2.4/2.4inch_ESP32-2432S024/1-Demo/Demo_Arduino/1_2_Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch.ino:127:76: error: 'color_p' was not declared in this scope
 void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
                                                                            ^~~~~~~
/home/user/ESP32/ESP32-DISPLAY-2.4/2.4inch_ESP32-2432S024/1-Demo/Demo_Arduino/1_2_Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch.ino:127:76: note: suggested alternative: 'lv_color_t'
 void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
                                                                            ^~~~~~~
                                                                            lv_color_t
/home/user/ESP32/ESP32-DISPLAY-2.4/2.4inch_ESP32-2432S024/1-Demo/Demo_Arduino/1_2_Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch.ino:140:23: error: variable or field 'my_touchpad_read' declared void
 void my_touchpad_read(lv_indev_drv_t *indev_driver, lv_indev_data_t *data)
                       ^~~~~~~~~~~~~~
/home/user/ESP32/ESP32-DISPLAY-2.4/2.4inch_ESP32-2432S024/1-Demo/Demo_Arduino/1_2_Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch.ino:140:23: error: 'lv_indev_drv_t' was not declared in this scope
/home/user/ESP32/ESP32-DISPLAY-2.4/2.4inch_ESP32-2432S024/1-Demo/Demo_Arduino/1_2_Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch.ino:140:23: note: suggested alternative: 'lv_indev_data_t'
 void my_touchpad_read(lv_indev_drv_t *indev_driver, lv_indev_data_t *data)
                       ^~~~~~~~~~~~~~
                       lv_indev_data_t
/home/user/ESP32/ESP32-DISPLAY-2.4/2.4inch_ESP32-2432S024/1-Demo/Demo_Arduino/1_2_Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch.ino:140:39: error: 'indev_driver' was not declared in this scope
[... cut ...]
/home/user/ESP32/ESP32-DISPLAY-2.4/2.4inch_ESP32-2432S024/1-Demo/Demo_Arduino/1_2_Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch.ino:230:5: error: 'lv_demo_widgets' was not declared in this scope
     lv_demo_widgets(); // OK
     ^~~~~~~~~~~~~~~
/home/user/ESP32/ESP32-DISPLAY-2.4/2.4inch_ESP32-2432S024/1-Demo/Demo_Arduino/1_2_Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch.ino:230:5: note: suggested alternative: 'lv_demos_create'
     lv_demo_widgets(); // OK
     ^~~~~~~~~~~~~~~
     lv_demos_create
```

rtek1000, 2024-07-18:
Note, the 'demos' were placed in the sketch folder.

[image]

rtek1000, 2024-07-18:
The compilation using old libraries worked, thank you.

closed 2024-07-18 by rtek1000