lvgl/lvgl #6008: Gridnav interaction with spinboxes or rollers

di-jtrumpower, 2024-04-02:
### LVGL version

v9.1.0-1-gc5109ccce

### What happened?

I am trying to create a 3 digit spinbox/roller to spell out a number using gridnav (I am using a physical keypad with (up/down/left/right/enter). I have tried three different approaches and they all seem to not play nice with the way gridnav intercepts keypad events:

Use lv_roller widget, set up snapping, scroll one, and custom styling to achieve look. Issue is that keypad presses are being intercepted by gridnav to perform the scrolling, gridnav is hardcoded to perform a bounded scroll by 1/4 the height of the scroll window, this is making any sort of 1 key press move 1 index impossible without editing the lv_gridnav.c. My understanding is that scroll_one/snapping is all done in the touch related input and isn't supported by gridnav anyway, so I would have to implement it myself using key callbacks.

I also tried this by manually creating my own roller widget or using the lv_spinbox widget. Same issue with gridnav not sending keys to the respective widget, even if the widget is focused.

Am I missing something on how this should be done? Or is this just a current limitation of gridnav?

### How to reproduce?

```	
lv_gridnav_add(BaseScreen.body,LV_GRIDNAV_CTRL_SCROLL_FIRST);
	BaseScreen.gridnavGroup = lv_group_create();
    lv_group_add_obj(BaseScreen.gridnavGroup,BaseScreen.body);
	lv_indev_set_group(keypad_indev, BaseScreen.gridnavGroup);
             
    for(int i = 0; i < ARRAY_SIZE(roller); i++)
    {
roller[i] = lv_roller_create(BaseScreen.body);
        lv_obj_add_flag(roller[i],LV_OBJ_FLAG_SNAPPABLE);
        lv_obj_add_flag(roller[i],LV_OBJ_FLAG_SCROLLABLE );
        lv_obj_add_flag(roller[i],LV_OBJ_FLAG_SCROLL_WITH_ARROW);
        lv_obj_add_flag(roller[i],LV_OBJ_FLAG_SCROLL_ONE);
        lv_obj_set_scroll_snap_y(roller[i],LV_SCROLL_SNAP_CENTER);
        lv_roller_set_options(roller[i], "0\n1\n2\n3\n4\n5\n6\n7\n8\n9", LV_ROLLER_MODE_INFINITE);
		lv_obj_add_style(roller[i],BaseScreen.rollerStyle,LV_PART_MAIN);
		lv_obj_add_style(roller[i],BaseScreen.rollerStyleFocused,LV_PART_SELECTED);
		lv_obj_add_style(roller[i],BaseScreen.rollerStyleFocused,LV_STATE_FOCUSED);
        lv_roller_set_visible_row_count(roller[i],1);
}
```

kisvegabor, 2024-04-05:
Hey,

If see the problem. To make it more obvious you can add a dropdown to the gridnav too. It can be opened by pressing `ENTER`, but as the up/down arrows are not forward to the drop down no option can be selected.

In case of the roller imagine that there are 4 rollers in a 2 x 2 matrix. What would be the expected behavior when the first roller (top-left) is selected and the down arrow is pressed?
1. Adjust the roller?
2. Selected the roller below it?

Maybe we can add a flag to forward the up/down keys to the widgets but use the left/right keys for gridnav navigation.

What do you think?

liamHowatt, 2024-04-05:
Hey @di-jtrumpower,

I see the problem. I think the most universal solution is for us to modify `gridnav` to forward the arrow key events down to the focused object if no refocus/scroll occurred. I tinkered with this for a while but I'm not satisfied with the consistency of the behavior of what I came up with. It seems to work for your example but when I started to make a `lv_example_gridnav_5` it was really sensitive to the group settings.

try using my branch https://github.com/liamHowatt/lvgl/tree/fix/gridnav-key-forwarding or patch the changes in this commit manually https://github.com/lvgl/lvgl/commit/5f340983bfe2193dde655bc78a89f8c09ed7cbf1

Here is the version of your code I got working the way you wanted with the adjustment to lv_gridnav.c in my branch. Notice that I don't set any of the flags on the rollers to enable scrolling. I just let them handle the scrolling themselves when they receive the arrow keys.
```c
static void gridnav_issue(void)
{
    struct {
        lv_obj_t * body;
        lv_group_t * gridnavGroup;
    } BaseScreen = {
        .body = lv_screen_active(),
    };

    lv_gridnav_add(BaseScreen.body, LV_GRIDNAV_CTRL_SCROLL_FIRST);
    BaseScreen.gridnavGroup = lv_group_create();
    lv_group_add_obj(BaseScreen.gridnavGroup, BaseScreen.body);

    lv_indev_t * indev = NULL;
    for(;;) {
        indev = lv_indev_get_next(indev);
        if(!indev) {
            break;
        }
        if(lv_indev_get_type(indev) == LV_INDEV_TYPE_KEYPAD) {
            lv_indev_set_group(indev, BaseScreen.gridnavGroup);
        }
    }

    for(int j = 0; j < 2; j++) {
        for(int i = 0; i < 3; i++) {
            lv_roller_t * roller = lv_roller_create(BaseScreen.body);
            lv_roller_set_options(roller, "0\n1\n2\n3\n4\n5\n6\n7\n8\n9", LV_ROLLER_MODE_INFINITE);
            lv_roller_set_visible_row_count(roller,1);
            lv_obj_set_y(roller, j * 60);
            lv_obj_set_x(roller, i * 70);
        }
    }
}
```

liamHowatt, 2024-04-05:
@kisvegabor 

>     1. Adjust the roller?
>     2. Selected the roller below it?

In my branch it selects the roller below it if it can (2) and adjusts the roller if it can't (1).

> Maybe we can add a flag to forward the up/down keys to the widgets but use the left/right keys for gridnav navigation.

Aha, I think that makes sense.

@di-jtrumpower?

kisvegabor, 2024-04-05:
@liamHowatt Interesting idea! I'm just thinking about the edge cases.

What if we have 3 drop-downs below each other. 
1, The first one is selected. 
2. I press `ENTER`, so it opens. 
3. Now I press the down key. 
4. The drop-down is not scrollable, so it will try to find a new widget and it will find the middle drop-down.
5. So the middle drop down will be selected and the options of the first drop down  cannot be adjusted.

Am I missing something?

liamHowatt, 2024-04-05:
@kisvegabor I tried it, and what you described happened exactly. I could open all 3 drop-downs at the same time but not select any of the items. I think this is related to the problems I saw while trying to make an `lv_example_gridnav_5`.

> Maybe we can add a flag to forward the up/down keys to the widgets but use the left/right keys for gridnav navigation.

So this might be viable instead, maybe?

di-jtrumpower, 2024-04-05:
Hey guys thanks for looking at this so quickly.

I will check out the patch, it seems like it would work fine with my case, horizontally arranged scroll boxes.

As for the issue with vertical boxes the only other case I can think of is if its an infinite scroller. Maybe we use the opposite axis key or enter to leave the focus of that drop-down/scroller? i.e vertical scroller can be deselected by pressing RIGHT/LEFT, then UP/DOWN will focus the next box.

Might be overthinking it and a forward scroll flag would be cleanest solution.

kisvegabor, 2024-04-05:
> As for the issue with vertical boxes the only other case I can think of is if its an infinite scroller. Maybe we use the opposite axis key or enter to leave the focus of that drop-down/scroller? i.e vertical scroller can be deselected by pressing RIGHT/LEFT, then UP/DOWN will focus the next box.

It's very similar to the concept that we have with the edit/navigate states with encoder input devices.  See [here](https://docs.lvgl.io/master/overview/indev.html#edit-and-navigate-mode).

So we could put the selected widget into editing state on ENTER, and leave it somehow. The only issue is that the editing state is stored in the `group` however the children widgets are not added to the group, but only the parent with the gridnav.

I think the lowest hanging fruit is adding `LV_GRIDNAV_CTRL_NO_VERTICAL_MOVE` and `LV_GRIDNAV_CTRL_NO_HORIZONTAL_MOVE`. Better names are welcome, but the default behaviour should be to allow moving on both axes, so the flags should disable an axis.

di-jtrumpower, 2024-04-05:
> I think the lowest hanging fruit is adding `LV_GRIDNAV_CTRL_NO_VERTICAL_MOVE` and `LV_GRIDNAV_CTRL_NO_HORIZONTAL_MOVE`. Better names are welcome, but the default behaviour should be to allow moving on both axes, so the flags should disable an axis.

I agree, I think this is sufficient for most use cases of gridnav and children widgets. 

Maybe gridnav with editable/selection states is a better long term plan, but I'll leave that to the experts to decide.

kisvegabor, 2024-04-05:
> Maybe gridnav with editable/selection states is a better long term plan, but I'll leave that to the experts to decide.

Later it can be a new flag, but we needed to update all the widgets to read their editable state from somewhere else, instead of their group. Not too complicated, but the single axis navigation is simpler and already solves a lot of issues.

liamHowatt, 2024-04-08:
After this PR I'd like to keep this issue open and open another PR to make `roller` send the `LV_EVENT_VALUE_CHANGED` event when the value is selected with arrow keys. It already sends the event for pointer. I would update the example to sync the sliders and rollers using that event (see the new example in the PR).

kisvegabor, 2024-04-12:
I think if the keys are forwarded to the roller/slider the `LV_EVENT_VALUE_CHANGED` event is fired too.

liamHowatt, 2024-04-15:
Unfortunately I do not see that event when I attach a callback to the roller and move it with arrow keys. It seems the event is only sent in the `release_handler` in lv_roller.c which is called for release events.

I just added it in the PR for `LV_EVENT_KEY` and also for `LV_EVENT_ROTARY`. Is this okay?

closed 2024-05-07 by lvgl-bot

kisvegabor, 2024-05-07:
The issue of `LV_EVENT_VALUE_CHANGED` was discussed in https://github.com/lvgl/lvgl/pull/6044