lvgl/lvgl #6011: API retirement

kisvegabor, 2024-04-03:
### Problem to solve

Continue #5096

To provide a better developer experience we need to have standards to retire APIs.

### Success criteria

- All the non essential functions are removed from the public API (`lvgl.h`) 
- The retirement policy is documented
- The core API is tested

### Solution outline

Rework the API:
1. The core API should mean the widgets, events, animation, styles, 3rd party libraries, drivers and the essential types (e.g. `lv_area_t`). Review, restructure and clean up `lvgl.h` to not include anything else than these. It can be achieved by adding several `*_private.h` files and moving typedefs to `lv_types.h`. `_lv_*` functions shall be moved to their own `private.h` file. 
2. APIs that shall not be part of the public API: image decoding, text processing, draw functions (except the top level ones, e.g. `lv_draw_rect`), layout APIs, linked list, lru, cache, etc.
3. Not sure about: LVGL's STDIO API

Retirement policy:
1. Keep the API for at least one more major version (e.g. APIs retired in v9 shall be available in v10, and can be removed in v11)
2. Provide backward compatibility in `lv_api_map*.h` files
3. Add `LV_LOG_WARN` in old APIs (we need to have `static inline` functions in `lv_api_map*.h` to add a warning)
4. Update the docs
 

### Rabbit holes

1. Extract the API from [JSON](https://github.com/lvgl/lvgl/pull/5677)
2. Create a `.h` file from the prototypes and compare it with a reference `*.h` file. 

### Testing

Testing:
1. Extract the API from [JSON](https://github.com/lvgl/lvgl/pull/5677)
2. Create a `.h` file from the prototypes and compare it with a reference `*.h` file. 

### Teaching

1. Add warnings in the deprecated functions
2. On each major release communicate the deprecated APIs too

kisvegabor, 2024-04-05:
@liamHowatt As you have already gained experience with `_lv_text` / `lv_text` update in this field would you mind taking this update?

liamHowatt, 2024-04-05:
Okay!

kisvegabor, 2024-05-03:
Not stale.

kisvegabor, 2024-08-12:
Let's wrap up this issue. We can agree on keeping all deprecated APIs for at least one extra major release.   

Now we have `lv_api_map_v*.h`. This way we can clearly track when the API changes happened. 

We can decide how to inform users about specific API retirements case by case. 

So I close this issue now. Let's reopen it if needed.

closed 2024-08-12 by kisvegabor