lvgl/lvgl #6014: Support mouse wheel scroll and other gestures

kisvegabor, 2024-04-03:
### Problem to solve

The pointer input devices can do more then just press/release. The mouse has a scroll wheel and multiple buttons, touchpads can detect multiple gestures, like rotate, pitch, or two finger scroll. 
 

### Success criteria

The main gestures/actions can be detected and applied:
- scroll with wheel
- handle multiple mouse buttons
- forward complex touch gestures

### Solution outline

In `lv_indev_data_t` we can have a `gesture` field which can contain the detected gesture:
```c
- LV_INDEV_GESTURE_SCROLL, //mouse wheel or 2 fingers
- LV_INDEV_GESTURE_ROTATE, //will set `style_transform_rotate`
- LV_INDEV_GESTURE_SCALE, //will set `style_transform_scale`
- LV_INDEV_GESTURE_BUTTON_1, //indicates clicking with button 1
- LV_INDEV_GESTURE_BUTTON_2, //indicates clicking with button 2
- ...
```
This way the input device can report only a single point (x, y) but it can tell what to do there: rotate, scroll, etc.

As some touch controllers can detect these gestures but other can't we can have a function like:
```c
gesture = lv_indev_detect_gesture_from_points(points, point_cnt);
```
It can be called from the `read_cb` on demand, so LVGL itself needs to process only `LV_INDEV_GESTURE`s.

### Rabbit holes

- We need more info about the capabilities of he touch controllers.
- Detecting gestures can be very complicated. We can consider using a 3rd party library.

### Testing

For processing `LV_INDEV_GESTURE`s unit test work.

`lv_indev_detect_gesture_from_points` can be also tested with a lot of sample data.

### Teaching

Update the docs, example and the existing driver.

kisvegabor, 2024-04-03:
cc @XuNeo @bjsylvia

kisvegabor, 2024-04-18:
Not stale

XuNeo, 2024-04-18:
> This way the input device can report only a single point (x, y) but it can tell what to do there: rotate, scroll, etc.

I like this idea of simplicity. The upper application can use the event directly and ignore multiple touch input points.

It may worth to check how android deals with it, to see if more special actions needed.

> We need more info about the capabilities of he touch controllers.

We can use a configure option. It's less likely we need to change the algorithm between software and hardware in run time.

kisvegabor, 2024-04-19:
> We can use a configure option. It's less likely we need to change the algorithm between software and hardware in run time.

As LVGL supports multiple displays and dynamically created displays (and indev-s) we should prefer run time config.

kisvegabor, 2024-09-23:
@etag4048 will work on it soon

kisvegabor, 2025-03-22:
@etag4048 and @Daxidz added multi touch support. See [here](https://docs.lvgl.io/master/details/main-modules/indev.html#multi-touch-gestures)

At least conceptually it'd be great to add mouse wheel scroll support too. I imagine it as a new scroll gesture the we can process obj event handler by default. For this we don't need a built in gesture detector.