lvgl/lvgl #6017: Implement an OpenGL ES based renderer

kisvegabor, 2024-04-03:
### Problem to solve

Similar to the [SDL based render](https://github.com/lvgl/lvgl/issues/6016) we can have  OpenGL ES based renderer too. The concept is the same as in the case of SDL.

### Success criteria

Have a fully functional OpenGL ES based renderer.

### Solution outline

To be determined.

### Rabbit holes

To be determined.

### Testing

It would be great to add unit tests to verify the output, but it's not clear if we can do it in GitHub CI, and if we can get back the result from the OoenGL Textures.

### Teaching

Docs needs to be added.

kisvegabor, 2024-04-18:
Not stale

danyyliu, 2024-04-20:
I will work on this issue, and finish it step by step.

Some basic functions we need to develop firstly, before we start the render development:
1. OpenGL ES display (window) driver
2. Basic OpenGL render demo

kisvegabor, 2024-04-20:
Awesome! Thank you, Dany! ❤️

danyyliu, 2024-04-21:
I will try to make a pull request about the display driver around next weekend (April 20), and let you know if there is any risk.

kisvegabor, 2024-04-24:
Hi @danyyliu,
I just would like to check if there are any updates on the OpenGL driver.

danyyliu, 2024-04-27:
Still working on it.

The driver is ok now to show an array with GL_RGB565 data, but not the buffer created by lvgl. I guess it's because there is a different arrangement between the data of lvgl and opengl.

You can see the update of code here: https://github.com/danyyliu/lvgl/pull/1.

I will start an official PR after the driver works with example in lv_port_pc_vscode.

kisvegabor, 2024-04-28:
Thank you fir the update! 

It's not important at this stage, but please prefix the opengl related functions with `opengles`  For example: `lv_index_buffer_unbind` :arrow_right: `lv_opengles_index_buffer_unbind`

danyyliu, 2024-05-03:
[PR](https://github.com/danyyliu/lvgl/pull/1) updated. Currently, I executed some examples in lv_port_pc_vscode, and it's shown correctly.
[image]

For the driver, there are still some works to do:
1. Make the code more official, e.g. replace printf function with LVGL log function, etc.
2. Update the build scripts(e.g. in lv_port_pc_vscode) for opengles.
3. Make sure the PR pass CI check.
4. Do some more tests with more examples and different opengl version.

I will try to finish it ASAP.

kisvegabor, 2024-05-07:
Great, thank you for the update! Please also add a docs page [here](https://docs.lvgl.io/master/integration/driver/index.html).

kisvegabor, 2024-05-16:
@danyyliu Could you give some update about the status of this development?

danyyliu, 2024-05-17:
Updates in one sentence:
With current code in [PR](https://github.com/danyyliu/lvgl/pull/1), example in [doc](https://github.com/danyyliu/lvgl/blob/opengl_dev/docs/integration/driver/opengles.rst), and enable LV_USE_OPENGLES, we can create/exit GLFW window with examples show correctly and mouse/ESC_KEY work correctly.

Updates:
1. Added configuration for opengl in Kconfig. The PR can pass CI check when opengl config is disabled.
2. Fixed bugs and memory leaks when close the glfw window. But there is still a memory leak when using glfw, which is caused by a bug in libX11 library, details: [_XimOpenIM leak 408 bytes](https://core.tcl-lang.org/tk/tktview?name=e42eef33ee). This bug is not fixed in ubuntu 20.04(libx11 version 1.6.9), and fixed in ubuntu 22.04(libx11 version 1.7.5).
3. A [doc](https://github.com/danyyliu/lvgl/blob/opengl_dev/docs/integration/driver/opengles.rst) page added.

Works still need to do:
1. Enable opengl in Kconfig, and make sure everything works fine with CI check.
2. Figure out how to do the test with the driver.
3. Add support for at least another version of OpenGL.

I am looking forward to open an official PR next week, please let me know if there are other flaws need to be fixed.

kisvegabor, 2024-05-20:
Fantastic! Thanks for the update! Now the goal is to have a working driver as soon as possible. So adding support for a new version in not needed in the first pass. As this feature is "only" for opening a window and showing software rendered content we don't need to add anything to the `draw` folder. We can split the code when working on the OpenGL renderer.

KioZhang, 2024-06-03:
Our project may use gles for drawing instead of sw. Does the official currently have plans to use a specific version, 2.0 3.0 3.1...？

kisvegabor, 2024-06-05:
cc @danyyliu can you comment on that?

FYI, this is the roadmap for OpenGL ES support:
1. Add GLFW driver: done, will be merged soon: #6254
2. Allow rendering to any texture that you can manage
3. Render with OpenGL ES

danyyliu, 2024-06-06:
> Our project may use gles for drawing instead of sw. Does the official currently have plans to use a specific version, 2.0 3.0 3.1...？

We will work on it to support render with gles, but there is still no deadline settle down for it.
Since OpenGL ES 3.0 is backwards compatible with OpenGL ES 2.0, and there are still many devices which support version 2.0, so we may take version 2.0 as a start, but it will not take a long time before the support of other versions.

Please let me know if you have other questions. @KioZhang

kisvegabor, 2024-09-10:
@liamHowatt took over this task and will open a PR soon.

kisvegabor, 2024-10-07:
Not stale

PR #6861 (2024-10-13): feat(opengl): texture caching
Fixes #6017

Similar to LVGL's SDL texture caching. Reuse draw buffers by creating OpenGL textures for them and caching them. In theory, it should be faster because they do not have to be SW rendered again and they should blend faster than they would by software.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_t [... cut]
closed 2024-10-13 by kisvegabor