lvgl/lvgl #6018: Add blurring support

kisvegabor, 2024-04-03:
### Problem to solve

On modern smartphone UIs blurring is quite popular. It would be great if we could add it to LVGL as well.

### Success criteria

1. Have a [CSS filter](https://css-tricks.com/almanac/properties/f/filter/) kind of mode: blure the widget and all its children.
2. Have a [CSS backdrop filter](https://css-tricks.com/almanac/properties/b/backdrop-filter/) kind of behaviour: blur the widget's bounding box (The widget needs to be transparent) 

### Solution outline

**filter**
It's hard to blur a rounded rectangle, or a label, or a rotated image once they are rendered (i.e. blended). Therefore we shall create a layer and blur it when it's blended as an image. It means the image rendering (e.g. `lv_draw_sw_image`) needs to be boosted by a generic `filter` support, in which `blur` and `recolor` are just two special cases. We need keep in mind that to blur a smaller part of a layer, a larger area is required to see all the pixels which contribute to the blurring.

**backdrop**
In CSS when a new stacking context is created the backdrop filter cannot be applied. It means the cases when `opacity`, `transformations` or `blend modes` are used. (Basically when we would create a new layer). It seems doable to blur the area of the widget before blending it, but handling the rounded corners could be tricky. For this we don't need a new layer, but just need to write a function which blurs an area in a buffer (by considering a radius too). 

 
Note that in CSS blur is just one type of the filters, so we can approach it in a more modular way where the filter are just callbacks.

### Rabbit holes

Backdrop: 
  - Blur rounded corners
  - In PARTIAL or DIRECT modes when only the changed areas are invalidated we need to invalidate a larger area to have all the extra pixels for blurring. For this we need to have a `extra_invalidation_area` property (might be calculated on the fly) which is used to increase the invalidated area when a backdrop-blurred widget is invalidated. It can be very complicated as unrelated widgets can invalidate the a blurred widget. (e.g. a mouse cursor). Maybe we can draw the backdrop on a layer to be sure that it's always available.  

### Testing

Normal unit testing shall work

### Teaching

Add docs, examples and demos

kisvegabor, 2024-04-18:
Not stale

GorgonMeducer, 2024-04-22:
Just for your reference, arm-2d introduces a set of IIR (Gaussian) Blur APIs. It is faster than the normal Gaussian Blur. 
These APIs do not need to `invalidate a larger area to have all the extra pixels for blurring` as normal Gaussian Blur or Box Blur do.

kisvegabor, 2024-04-23:
Very interesting! Could you link a description of IIR gaussion blur? I have sound only some short definitions. 

Just to confirm: if you see only a small rectangle from a larger image, and blur that small rectangle, it will look the same as if you have blurred the whole rectangle. Is that correct?

kisvegabor, 2024-04-23:
ChatGPT just helped me out :slightly_smiling_face: 
It seems we do need a larger area else we won't know the previous value (start value) for the filer.

GorgonMeducer, 2024-04-24:
> if you see only a small rectangle from a larger image, and blur that small rectangle, it will look the same as if you have blurred the whole rectangle. Is that correct?

In arm-2d, IIR blur supports PFB (Partial Framebuffer). So it will look the same as if you have blurred the whole rectangle.  But this requires the display adapter service of arm-2d, which cannot be fit into LVGL.

> It seems we do need a larger area else we won't know the previous value (start value) for the filter.

You can save the status (the previous value). That's how Arm-2D supports PFB for IIR Blur.

GorgonMeducer, 2024-04-25:
@kisvegabor No, only the valid region is available. The Target region is the area you want to apply blur.

kisvegabor, 2024-04-25:
Thank your, but I don't get it. Theoretically, how can you get the accumulator value for an arbitrary pixel without knowing the previous pixels?

GorgonMeducer, 2024-04-26:
@kisvegabor I get the previous value from `ptStatusV` and `ptStatusH`

kisvegabor, 2024-04-29:
I definitetly need more time to understand how the magic happens :slightly_smiling_face:

GorgonMeducer, 2024-04-29:
You have plenty of time.

kisvegabor, 2024-09-26:
Not stale, it should be added in v9.3

kisvegabor, 2024-10-25:
Related issue: https://github.com/lvgl/lvgl/issues/7051#issuecomment-2412549291

AndreCostaaa, 2026-02-06:
Blurring support is now added see #9223 and #9331

closed 2026-02-06 by AndreCostaaa