lvgl/lvgl #6019: Improve the span widget

kisvegabor, 2024-04-03:
### Problem to solve

Inline text recoloring (e.g. `"it's #ff0000 red#"`) was removed in LVGL v9 because it wasn't not powerful enough. Instead the span widget shall be improved. 

### Success criteria

1. The span widget is easy to use and doesn't require too much boiler plate
2. Inline text commands can be processed and converted to spans.
3. The span is not significantly slower than the label

### Solution outline

**Update the API of the span**
It can look something like this:
```c
lv_obj_t * spans = lv_spangroup_create(lv_screen_active());
lv_obj_set_style_text_align(spans, LV_TEXT_ALIGN_CENTER, 0);
lv_obj_set_width(spans, 100); /*LV_SPAN_MODE_BREAK, LV_SIZE_CONTENT would mean LV_SPAN_MODE_EXPAND*/
lv_obj_center(spans);

lv_spangroup_add_span(spans, "Hello, I'm a");

lv_spangroup_add_span(spans, "big");
lv_spengroup_set_text_font(spans, NULL, &lv_font_montserrat_40); /*NULL means the last span*/
lv_spengroup_set_text_color(spans, NULL, lv_color_hex3(0xf00));

lv_spangroup_add_span(spans, " text.");
```

**Inline text processing**
It's a helper function which creates spans as parsing the text. The exact format is yet to be defined but something like this could work:
```c
lv_style_t styles[] = {...}; //Each style has a name, e.g. "style1", "style2"
lv_span_process_rich_text(parent,  "it's <style1 red> and this is <style2 big>", styles, style_cnt);
```

**Performance improvement**
Now the letters are drawn one by one which is no effective. @XuNeo [proposed](https://github.com/lvgl/lvgl/issues/5352#issuecomment-1914960761) to use the label widget as spans, but IMO it's overkill. Instead we can just buffer the letter from the same span in each row and draw them at once. In case of a span with 30 lines it means 30 draw operations. (It were 1 if we were used a label, but we have like 1000 now for each letter)

### Rabbit holes

To be defined

### Testing

Normal unit testing, and benchmarking on embedded hardware

### Teaching

Update the docs and the examples

kisvegabor, 2024-04-03:
cc @XuNeo @bjsylvia

kisvegabor, 2024-04-18:
Not stale

XuNeo, 2024-04-18:
Hi Gabor, 
This is an important feature, but is not currently under investigation from my side. Let's see if someone can jump in to take it.

kisvegabor, 2024-04-19:
I've discussed with @liamHowatt that he will work on it after the API retirement update.

andyrea76, 2024-05-10:
definitely not stale with the removal of recolor, see #6210

kisvegabor, 2024-05-13:
From https://github.com/lvgl/lvgl/issues/6210:

>  but you have a lot more objects to create potentially dynamically as elements to highlight change

Note that the `spans` are not `lv_obj`s. Just [small `struct`s](https://github.com/lvgl/lvgl/blob/master/src/widgets/span/lv_span.h#L56-L61) with a little info in them. When text is parsed to create the spans we can allocate a multiple `lv_span`s at once, so it's only 1 extra malloc.  I think in the end I believe the memory usage wont's be significantly larger especially for simple spans. 

> surely the old scheme of recolouring by the text parsing A: was already there and working for many people and B: could still exist in parallel to also not break something else in the not entirely trivial migration from 8.x up to 9

My intention was to have a single good way to add inline formatting. I felt like the recoloring-only adds a lot of complexity in exchange for a single not extensible feature.

I do agree that simple recoloring is quite common and there should be a boiler plate free way to handle it (i.e. without creating a new style for each color).

andyrea76, 2024-05-16:
is there no way to hide elements of the span? looking through lv_span.c it doesn't appear to be the case

kisvegabor, 2024-05-20:
Converting a mark up-ed string to spans shouldn't take so long to really affect the battery. Especially if we consider that it's running rarely. (I.e. not every frame). 

To make a compromise let's see how it works with the span and if it's really not good at the end we can consider adding back the old mechanism.

> is there no way to hide elements of the span?

Really there is no such a feature now.

andyrea76, 2024-05-21:
the span code module having stepped / looked through it is fairly vast by comparison to what was removed to do the recolouring? the system i have inherited that was making use of that functionality works on the principle of creating all the screens and popups when it starts up and the showing the relevant screen , updating / hiding parts of screens if / as appropriate. meaning it only has the need for a generic popup alert screen that has its text updated before being displayed which is a fairly small part of the label code . with the span approach, i would have to destroy any previous span group objects, create the spans as they are almost certainly different to previous, add all the styles and text sections, then have the span module create all the labels and set the styles etc on all those. rather than do some simple string parsing and colouring when painting. 

with the lack of visibility control i had to abandon the span entirely for the status icons and just revert to a flex row of labels in a container - basically doing it all by hand but at least i have control and can 'quickly' change the colour of each icon as necessary

kisvegabor, 2024-05-26:
I really understand why it's a problem. I also lacked this features sometimes... However I still believe we should add custom solution very carefully as they are risky. Later they could cause any kind of issues with flexibility, extensibility, or integrability.

azureussolutions, 2024-05-26:
Is it really that hard to re-instate the old the old recolouring functionality as, it was I assume, working and reliable for some time in version 8.x before being removed?

kisvegabor, 2024-05-30:
v9 was a good point to remove it and do something better instead. If we added back we would have a feature which that we might not want to maintain.

liamHowatt, 2024-06-25:
Would the inline text system support `"normal, <style1 blue, <style2 blue and big,> just blue again>."`? Or is nesting spans out of scope? I'm not actually sure how overlaying styles would work, exactly. But it that example would create 4 spans that aren't actually "nested" in the implementation. the span array would be {span with default style, span with style 1, span with style 1+2, span with style 1}.

Another question: should the inline text system support `%s`?
```c
lv_obj_t * spangroup = lv_span_process_rich_text(lv_screen_active(), styles, style_cnt, "<style1 %s>\n<style2 %s>", title_str, body_str);
```

kisvegabor, 2024-06-26:
IMO nesting is not needed, especially at the first pass.

> Another question: should the inline text system support %s?

Great question! We can support it in `lv_span_process_rich_text_fmt` to save the extra temporal memory usage for the user's buffer (we allocated the the final array size internally in one pass)

liamHowatt, 2024-06-26:
Okay! Would the user need to supply a buffer in any case? I'm failing to see the buffer that needs to be provided.

kisvegabor, 2024-06-27:
E.g. if we had **no** `lv_label_set_text_fmt` user needed to do this:
```c
char buf[256];
lv_snprintf(buf, 256, "x = %d", 1);
lv_label_set_text(label, buf);
```

Here `buf` is allocated on stack, and `lv_label_set_text` will malloc `strlen(buf) + 1 = 6 bytes`, and save `"x = 1"` there. 

In `lv_label_set_text_fmt` the final size calculated in a ["dry-run"](https://github.com/lvgl/lvgl/blob/master/src/misc/lv_text.c#L412) and only the 6 bytes are allocated. 

We can do the same with `span` to save the memory allocation on stack.

liamHowatt, 2024-06-27:
Oh of course. And it would support all the format specifiers because it uses `lv_snprintf` inside!

Could a user do this?
```c
const char * str_from_outside = "><style2 not the style you expected";
lv_span_process_rich_text_fmt([...], "My name is: <style3 %s!>", str_from_outside);
```
The intuitive implementation would allow it (i.e. 1. snprintf, 2. generate spans). Any reason we'd want to prevent it?

kisvegabor, 2024-06-27:
> Any reason we'd want to prevent it?

I think we should allow it. However, it seems an escaping mechanism needs to be invented. E.g.
```
<style1    -> beginning of a new span
\<style1   -> a string  like "<style1"
\\<style1  -> resolved to "\" + beginning of a new span
\\\<style1 ->  a string  like "\<style1"
```
It's just my 2 minute idea so don't take it too seriously 🙂 

Besides the `"<style1 text>"` approach is not written into stone either. Feel free to suggest a better alternative if you have any ideas.

liamHowatt, 2024-06-28:
My only thought about using `\` as an escape character is that C string literals interpret `\` too. To input a `\` into a string literal, you have to escape it like `\\`.

So your example would become:
```
<style1
\\<style1
\\\\<style1
\\\\\\<style1
```

In the MicroPython binding, there are raw strings `r""` which ignore escapes. And interestingly, GCC has `R"()"` which does the same, but it's not standard.

Python f-strings `f"my name is {str}"` use `{}`, which is escaped by `{{}}` doubling up the curly braces. I'm not sure if it's the best idea, but maybe we could do `<<style1 foo>>` to escape `<`. I'm not sure. To reduce parsing, and since there's no nesting, maybe we just need to escape the opening `<`, so `<<style1 foo>`.

kisvegabor, 2024-07-01:
>  To input a \ into a string literal, you have to escape it like \\.

Ah, true! 

> To reduce parsing, and since there's no nesting, maybe we just need to escape the opening <, so <<style1 foo>.

Good idea! I'm okay with that.

azureussolutions, 2024-07-02:
is this not a bit like the old recolour option now , just with a huge amount more code and CPU time behind it - albeit with more functionality

there were a good number of significant changes with going to 9.x from 8.x which were barriers for people.

I had to just give up on the spans in the end as the recolouring cases for me were actually sufficiently simple that I could just do it manually creating some objects without the behemoth of code that the spans brought

FWIW, the project I inherited creates all the screens with all their objects once at startup and then just brings whatever to the front with the rest hidden and updates the relevant widgets. being battery powered and limited on CPU grunt, this makes for the cleanest and lowest power option when running . I would really have liked to have seen the legacy recolour system maintained given it was there and worked as that just requires a bit of CPU time to parse and just flip the colour on the fly as a text string is processed - essentially it looks like you are doing that same degree of parsing to pull out the style formatters but then have to create n labels and all manner of other objects .

egonbeermat, 2024-09-16:
Please bring back `lv_label_set_recolor`. I use it in multiple places in my code, and the text in some of the labels varies between 2, 3 or 4 different data elements, with differing color. It is a very simple matter for me to set simply:

`lv_label_set_text_fmt(tl_lblInfo, "%s L#%ldM %s Hp#%dM", warningStatusColor, lvglMaxMemUsed / 1000, warningStatusColor, freeram() / 1000);`

or 

`lv_label_set_text_fmt(tl_lblInfo, "%s %dMHz# %dbpp", userCPUSpeed <=600 ? normalStatusColor : warningStatusColor, userCPUSpeed, currentBitDepth);`

etc. and use slightly different variations of this. I know you said the functionality is limited, but for myself (and it appears others), it's fine as it is, and to try to write some generic functions to create multiple spans and color them on the fly for the dozens of places I use this, is just pointless and tedious. I have never known an API just drop functionality like that (apart from prior LVGL upgrades :) ); deprecate it at the very least, please, whilst introducing and refining alternatives. And at the VERY least, add `lv_span_set_text_fmt()`!!

For now, this is the straw that breaks the camels back for me. I've done past migrations from v6 to v7, v7 to v8 and due to my experiences during those (significant, undocumented / poorly documented breaking changes) I delayed the migration to v9. I'm already wary that the increased size of lv_color_t to 3 bytes from 2 for LV_COLOR_DEPTH 16 is going to break my memory limits, which I was pushing to the max on a Teensy 4.1 with EXTRAM (I had to use #define LV_USE_LARGE_COORD 0 in V8.x.x and other tricks to fit code and buffers into available memory), and I've already discovered that lv_colorwheel, lv_meter, lv_label_set_recolor, LV_EVENT_DRAW_PART_BEGIN, LV_EVENT_DRAW_PART_END are just GONE, and suspect that LV_COLOR_CHROMA_KEY functionality to not draw pixels has gone), and I still have many 'unknowns' to resolve, so I am currently faced with significant re-engineering to keep some functionality as-is, no clue how to replace the lost lv_colorwheel and lv_meter, unsure how to migrate LV_EVENT_DRAW_PART_BEGIN/END and unsure what else is lurking out there and what other re-engineering I need to do, so I'm about done with LVGL completely, because even if I somehow make it through this migration, I know it will happen again and again of v10, v11.....so, I'm rolling back the hours of work I've done trying to upgrade to v9, and will stay with LVGL 8.x.x for the life of my product, unless something significant changes, or I find a passable alternative.

I really love LVGL. I really hate how you go about upgrading it.

[This is an older but mostly accurate video](https://www.youtube.com/watch?v=kmOkUK644m8) of my app showing off all the LVGL features I use. It's a Lightning Trigger / Camera Assistant, with a couple of 'kitchen sink' sub-apps in there (video player, photo viewer, etc), but you can see labels using recolor (top line, text and icons), my use of lv_meter, and use of LV_EVENT_DRAW_PART.... on the sliders for example, the use of LV_COLOR_CHROMA_KEY for the images on the menu icons, etc.

kisvegabor, 2024-09-26:
Okay, I stop fighting. Recoloring was really asked by too many people. I've opened an issue for it: https://github.com/lvgl/lvgl/issues/6944

I can allocte some resources for it only in 3-4 weeks. If you are interested in working on it, please comment in https://github.com/lvgl/lvgl/issues/6944

azureussolutions, 2024-10-11:
fundamentally the 2 schemes of recolouring text are not compatible with each other and are probably stopping a lot of people migrating up from 8.x . the spans i can see have use in certain applications but are massively overkill for most of the original uses of recolouring in 8.x . i couldn't make them work for me so just manually created different labels for the few instances I needed.

also i re-iterate a point from my original posts, CPU usage is horrific compared to the recolouring with spans and given this library is used I imagine in quite a lot of battery powered products like my applications are, wasting limited CPU horsepower in ARM processors on excess code is not helpful

kisvegabor, 2024-10-11:
https://github.com/lvgl/lvgl/pull/6966 is ready and adds back recoloring. It just waits for the final approval. Please try it out.

kisvegabor, 2024-10-29:
Not stale, just having no time. :smile:

kisvegabor, 2024-12-05:
Let close it for now in lack of time and interest.

closed 2024-12-05 by kisvegabor