lvgl/lvgl #6020: Double and triple click support

kisvegabor, 2024-04-03:
### Problem to solve

Continue #5351.
The users get used to the double and triple click as they are commonly used on PC. 

### Success criteria

1. Double and triple click are are recognized by LVGL internally. 
2. It works for all input device type

### Solution outline

In `lv_indev.c` a generic method can be implemented to handle press, click, release, short click, long press, long press repeat, double click and triple click. This method can be applied to all input devices. 

### Rabbit holes

Nothing for now.

### Testing

Once it's implemented deep testing shall be added for all detected "gestures" and input device types.

### Teaching

Update the docs.

### Considerations

I proposed to support complex cases like the second press after a click, or the long press after a double click, but thinking about it more I think they are not common, so we shouldn't focus on them.

AllanOricil, 2024-06-03:
I'm trying to implement this myself in lvgl 8. When are you guys thinking about releasing it? I need double touch for turning on and off my CYD.  Will this be backported to v8?

kisvegabor, 2024-06-05:
cc @niklasf do have an estimate about when you can finish the Pull Request?

niklasf, 2024-06-05:
Probably in the next few weeks. In the meantime, it's completely fine if someone else wants to take over or start a new patch.

AllanOricil, 2024-06-05:
@niklasf cool! So I will wait you to finish it! 
Im going to use it in this project 

https://github.com/AllanOricil/esp32-mfa-authenticator

Screen will turn off with double touching, and on with single touch.

AllanOricil, 2024-06-09:
I ended up implementing my own double touch solution using the input callback. Is working pretty well. 

https://github.com/AllanOricil/esp32-mfa-authenticator/pull/18/files

closed 2024-09-12 by kisvegabor

AllanOricil, 2025-01-20:
I have simplified a bit the double and single touch solution I created before

https://github.com/AllanOricil/esp32-mfa-authenticator/blob/d1ca27a9d234f6551dea1f0d9edb259e4e260357/src/touch.cpp#L179-L227

kisvegabor, 2025-01-24:
Cool! Could you send a PR to see the changes?

commit 651f69fd (2024-09-12): feat(indev): detect double and triple click (closes #6020) (#6187)