lvgl/lvgl #6021: Support Lottie animation

kisvegabor, 2024-04-03:
### Problem to solve

Add support for parsing and rendering Lottie animations.

LVGL already has support for [rlottie](https://docs.lvgl.io/master/libs/rlottie.html), but it's difficult to compile and it's not actively maintained. ([no commits](https://github.com/Samsung/rlottie/) for 8 month). Samsung also recommended to migrate to ThorVg. 

### Success criteria

1.Lottie animations can be played
2. Managed build systems are supported (i.e. no special build system is required)
3. The animations has a reasonable speed

### Solution outline

We already have ThorVG implemented in LVGL which supports Lottie. It works with managed build systems too, so it's easy to integrate in to any projects. 

As `lv_image` doesn't support animated images now, the Lottie animations can be rendered on an `lv_canvas` (similar to `lv_gif`). It also means that it has it's own API. E.g.
```c
lv_lottie_create(parent);
lv_lottie_set_buffer(lottie, buf, width, height, stride); //Will be the size of the canvas too
lv_lottie_set_draw_buf(lottie, draw_buf); 
lv_lottie_set_src_file(lottie, "A:path/to/lottie.json");
lv_lottie_set_src_data(lottie, data, size);
```  

### Rabbit holes

ThorVG is in C++, but probably it's not an issue.
 

### Testing

Normal unit testing.

### Teaching

Add examples and update the docs

### Considerations

Theoretically we could use vector graphics GPUs to render the Lottie animations (instead of ThorVG) but it would be way more complicated. Using ThorVG's Lottie engine is a low hanging fruit and should be very useful for many users. 

Later we can improve it without changing the API.

kisvegabor, 2024-04-03:
I've already implemented it and tested on Renesas, ESP, and ST. On an ESP-S3 I couldn't compile it due to a compiler bug, but other than that it was working well and with a good performance.

cc @XuNeo

MichaelDu9226, 2024-04-15:
Hi, @kisvegabor 
Any update on Lottie or ThorVG libs?
I am testing LVGL9 and ThorVG Lottie animations on s3_lcd_ev-board & https://github.com/espressif2022/esp-bsp.
I also meet compilation error similar to yours.
Looking forward to your update, thanks

kisvegabor, 2024-04-17:
These is an update :tada: Finally :smile: 

See https://github.com/lvgl/lvgl/pull/6081

It was tested only by me (and the CI) so far, so feedback would be very welcome! 

I close this issue, let's continue in the PR.

PR #6081 (2024-05-23): feat(lottie): add lottie widget
### Description of the feature or fix

closes #6021

**Overview**
- ThorVG is updated as there were Lottie related fixes in the newer versions
- I've added the Lottie support as widget which is odd a little bit as it relies on a 3rd party lib

**TODO**:
- [x] support loading Lottie from file (ThorVG needs to updated to use LVGL's file system API)
- [ ] consider replacing `malloc`s in ThorVG with `lv_malloc`
- [ ] consider using `lv_image_dsc_t` or `path` in `lv_lottile_set_src` instead of havong 2 functions for data and path
- [x] try compiling ThorVG with Emscr [... cut]
closed 2024-04-17 by kisvegabor