lvgl/lvgl #6022: Add SVG support

kisvegabor, 2024-04-03:
### Problem to solve

Using SVG images is a great way to have responsive UIs. We shall find a way to support them.

### Success criteria

1. SVG images can be opened just like PNG, JPG, etc
2. Use vector graphics GPUs to render SVGs  

### Solution outline

ThorVG has built-in SVG support, so it should be simple to implement it. We just need to add a new image decoder. 

The resolution always should be the same as the image's size.  

> Have an interface to easily pre-render SVG (or other images) on startup

It can be just a simple function like `lv_image_preload(img_src, buf, ...);`
 

### Rabbit holes

Can not see any now

### Testing

Normal unit testing should be enough.

### Teaching

Add docs and examples.

### Considerations

Use vector graphics GPUs to render SVGs -> It seems complicated and we can consider it as a next step later

kisvegabor, 2024-04-18:
Not stale, I'll work on it when the [Lottie update](https://github.com/lvgl/lvgl/pull/6081) is merge.

XuNeo, 2024-04-18:
> It can be just a simple function like `lv_image_preload(img_src, buf, ...);`

I imagined this function will return a const struct lv_vector_dsc_t, and this is why https://github.com/lvgl/lvgl/pull/5970 is opened. 

There are other cases may need consideration. For example how to add a normal image rotation to **`const `** lv_vector_dsc_t. It's great you can work on it. 👍

kisvegabor, 2024-04-18:
My idea was to return an `lv_image_dsc_t` or `lv_draw_buf_t`. It's basically just an image saved in a memory managed by you (and not the cache). So it applies to PNG, JPG, etc too.

I found that in many project it would be quite useful to just preload some images e.g. from an SD card, or convert all SVG on startup according to the display size.

XuNeo, 2024-04-19:
> My idea was to return an lv_image_dsc_t or lv_draw_buf_t.

Yes. I agree. I mean the `lv_image_dsc_t->data` field should have a type of lv_vector_dsc_t, we need a way to describe loaded vector image.

> I found that in many project it would be quite useful to just preload some images e.g. from an SD card, or convert all SVG on startup according to the display size.

We have done similar thing by call lv_image_decoder_open() early before apps starts, put them to cache. Maybe a cache attribute of `persist` would be even better.

kisvegabor, 2024-04-24:
In `lv_image_decoder_open(&decoder_dsc, src, &args);` there were a malloc temporarily. What if in `args` we optionally pass a  buffer?

kisvegabor, 2024-09-19:
No stale

kisvegabor, 2024-10-23:
Basic SVG support is added, but an image decoder is still needed to simplify the usage.

onecoolx, 2024-10-24:
> Basic SVG support is added, but an image decoder is still needed to simplify the usage.

feat(decoder): add svg image decoder for image widget
https://github.com/lvgl/lvgl/pull/7141

Michi202020, 2025-03-09:
Hello,
I have a display 4848S040 and i will that i can see my svg file on the display but nothing works. This is my test.svg:
`<svg xmlns="http://www.w3.org/2000/svg" viewBox="0 0 24 24">
  <g clip-path="url(#a)">
    <path d="M8.5 10.5a.5.5 0 1 1-1 0 .5.5 0 0 1 1 0Zm3.5.5a.5.5 0 1 0 0-1 .5.5 0 0 0 0 1Zm4.5-.5a.5.5 0 1 1-1 0 .5.5 0 0 1 1 0Z"/>
    <path fill-rule="evenodd" d="M12.354 1.146a.5.5 0 0 0-.707 0l-11.5 11.5A.5.5 0 0 0 .5 13.5H3v10a.5.5 0 0 0 .5.5h17a.5.5 0 0 0 .5-.5v-10h2.5a.5.5 0 0 0 .354-.854l-11.5-11.5ZM12 2.207 22.293 12.5H1.707L12 2.207Z" clip-rule="evenodd"/>
    <path d="M15.538 2.309A.5.5 0 0 1 16 2h3.5a.5.5 0 0 1 .5.5V6a.5.5 0 0 1-.854.354l-3.5-3.5a.5.5 0 0 1-.108-.545Z"/>
  </g>
  <defs>
    <clipPath id="a">
      <path d="M0 0h24v24H0z"/>
    </clipPath>
  </defs>
</svg>
`
and this is my main and in the first step i need only this svg on my display:
`#include <Arduino.h>
#include <lvgl.h>
#include <esp32_smartdisplay.h>
#include <FS.h>
#include <LITTLEFS.h>
#include <stdio.h>
#include <string.h>
#include "thorvg.h"

// Farb-Definitionen für SVG
#define SVG_WHITE 0xFFFF
#define SVG_BLACK 0x0000
#define SVG_RED 0xF800
#define SVG_GREEN 0x07E0
#define SVG_BLUE 0x001F
#define SVG_YELLOW 0xFFE0

// Forward-Deklaration
void testklaus();

// Canvas für SVG-Rendering
lv_obj_t* canvas = NULL;
lv_color_t* canvas_buf = NULL;
uint32_t canvas_w = 0;
uint32_t canvas_h = 0;

// Adapter für Display-Zugriff
class SmartDisplayAdapter {
public:
    SmartDisplayAdapter() {
        // Canvas erstellen, wenn noch nicht vorhanden
        if (canvas == NULL) {
            canvas_w = lv_display_get_horizontal_resolution(NULL);
            canvas_h = lv_display_get_vertical_resolution(NULL);
            
            size_t buf_size = canvas_w * canvas_h * sizeof(lv_color_t);
            canvas_buf = (lv_color_t*)heap_caps_malloc(buf_size, MALLOC_CAP_SPIRAM);
            
            if (canvas_buf == NULL) {
                Serial.println("FEHLER: Nicht genug Speicher für Canvas!");
                return;
            }
            
            canvas = lv_canvas_create(lv_scr_act());
            lv_canvas_set_buffer(canvas, canvas_buf, canvas_w, canvas_h, LV_COLOR_FORMAT_RGB565);
            lv_obj_set_pos(canvas, 0, 0);
            lv_obj_set_size(canvas, canvas_w, canvas_h);
            lv_canvas_fill_bg(canvas, lv_color_white(), LV_OPA_100);
            lv_obj_add_flag(canvas, LV_OBJ_FLAG_HIDDEN);
        }
    }
    
    void update() {
        // Erzwinge Display-Aktualisierung
[... 143 lines cut ...]
    
    // Dateisystem prüfen
    File root = LittleFS.open("/");
    File file = root.openNextFile();
    Serial.println("Dateien im LittleFS:");
    while (file) {
        Serial.print("  ");
        Serial.println(file.name());
        file = root.openNextFile();
    }
    
    // Verzögerung für LVGL-Initialisierung
    delay(500);
    
    // SVG testen
    testklaus();
}

void loop() {
    // LVGL-Task-Handler regelmäßig aufrufen
    lv_task_handler();
    delay(5);
}`

Many Thanks for Help

kisvegabor, 2025-03-23:
cc @onecoolx

onecoolx, 2025-03-26:
>  // LVGL-Task-Handler regelmäßig aufrufen
>     lv_task_handler();
>     delay(5);
> }
> Many Thanks for Help

You don't need to use thorvg directly to render SVG images now. You should use the LVGL native API to parse and render SVG, as shown in the following example:

https://docs.lvgl.io/master/details/libs/svg.html#example

Michi202020, 2025-03-31:
Hello, many thanks its works. Best regards Michael

kisvegabor, 2025-03-31:
Great! I close this issue then.

closed 2025-03-31 by kisvegabor