lvgl/lvgl #6024: Text & Image Shadow

shirazelec, 2024-04-04:
### Introduce the problem

To create a shadow, creating a label or a photo under the object slows down the display

### Proposal

This code can create shadows with better speed for photo and text

[user_misc.zip](https://github.com/lvgl/lvgl/files/14861648/user_misc.zip)

kisvegabor, 2024-04-04:
Could you share the slow version too to see exactly what we are trying to optimize?

shirazelec, 2024-04-04:
No version is slow
Only this method is more optimal for creating shadows
Please read the files

kisvegabor, 2024-04-05:
I mean please send a code snippet to test the slow version too. (E.g. `lv_obj_set_style_text_shadow_slow(...) {...}`) I would like to compare them and examine why the recommended version is faster.

shirazelec, 2024-04-05:
```C
  //old method
  lv_obj_t *lbl_shdow = lv_label_create(home_obj);
  lv_obj_set_style_text_color(lbl_shdow, lv_color_make(255, 255, 255), LV_PART_MAIN);
  lv_label_set_text(lbl_shdow, "Sample");
  lv_obj_align(lbl_shdow, LV_ALIGN_CENTER, 0, 0);

  lv_obj_t *lbl_main = lv_label_create(home_obj);
  lv_obj_set_style_text_color(lbl_main, lv_color_make(0, 0, 0), LV_PART_MAIN);
  lv_obj_align_to(lbl_main, lbl_shdow, LV_ALIGN_TOP_LEFT, -1, -1);
  lv_label_set_text(lbl_main, lv_label_get_text(lbl_shdow));

  //new method
  lv_obj_t *lbl1 = lv_label_create(home_obj);
  lv_obj_set_style_text_color(lbl1, lv_color_make(0, 0, 0), LV_PART_MAIN);
  lv_obj_set_style_text_shadow(lbl1, lv_color_make(255, 255, 255), 1, 1);
  lv_obj_center(lbl1);
  lv_label_set_text(lbl1, "Sample");
```

kisvegabor, 2024-04-08:
Got it now, thanks you! How much faster was your method? And on what platform have you tested it?

shirazelec, 2024-04-08:
I also tested it on esp32 and F1C100S, it was very effective in the speed of running animations

closed 2024-05-01 by lvgl-bot