lvgl/lvgl #6025: Moving wallpaper test from benchmark wrong coordinates

anaGrad, 2024-04-04:
### LVGL version

v9.1.0

### What happened?

During testing we found that the area that is saved in the draw task for the moving wallpaper test is the area of just the image and not the area of the whole rectangle on which the image should be placed. 
[image]

nicusorcitu, 2024-04-04:
Given that, the VGLite will wrongly draw the image only to the specified area 0,0 99,99.
For this particular case with the tile image on the full screen, the t->coords should have been the full screen.
Can this be fixed upstream? Thanks.

nicusorcitu, 2024-04-04:
@cristian-stoica

kisvegabor, 2024-04-04:
Thank you reporting it. It also explains an anomaly that we have just discovered with multi core rendering. 

I've added a fix in #6029.

anaGrad, 2024-04-09:
Thank you for the fix, it works perfectly.

PR #6029 (2024-04-17): fix(image): set the draw_task area correctly for tiled image
fixes #6025

### Description of the feature or fix

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) [... cut]