lvgl/lvgl #6026: Compiler error withing lv_label when LV_USE_BIDI is enabled

senad-zaimovic, 2024-04-04:
### LVGL version

9.1.0

### What happened?

LVGL compilation fails with error "error: 'txt_len' may be used uninitialized [-Werror=maybe-uninitialized]" when LV_USE_BIDI option is enabled on ESP-IDF version 5.2.1.

Additional error stack:
lvgl/src/widgets/label/lv_label.c:455:27: error: 'txt_len' may be used uninitialized [-Werror=maybe-uninitialized]
  455 |             logical_pos = _lv_bidi_get_logical_pos(&txt[line_start], NULL,
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  456 |                                                    txt_len, lv_obj_get_style_base_dir(obj, LV_PART_MAIN), cid, &is_rtl);
      |                                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

kisvegabor, 2024-04-04:
Thanks for the report! I think it's false positive as it's set `if(bidi)` [here](https://github.com/lvgl/lvgl/blob/master/src/widgets/label/lv_label.c#L403) and read if `if(bidi)` [here](https://github.com/lvgl/lvgl/blob/master/src/widgets/label/lv_label.c#L447).

Anyway, I created [a PR](https://github.com/lvgl/lvgl/pull/6028) to silence it.

senad-zaimovic, 2024-04-15:
I see that the PR is merged now. Thanks for checking and fixing this issue.

closed 2024-04-15 by senad-zaimovic