lvgl/lvgl #6030: Separate font size from lv_font_t into individual style

zylalx1, 2024-04-05:
### Problem to solve

Currently, the font size is tightly coupled with the lv_font_t structure, which represents a specific font at a specific size. This coupling has several disadvantages.

1. The coupling of font type and font size is not logically consistent. Font type and font size are independent attributes that should be decoupled to provide more flexibility and clarity in font handling.

2. Presently, directly setting font size using numerical values or binding font size to variables brings inconvenience to users. Users need to manage lv_font_t pointers for all possible font sizes they might use, which increases complexity and maintenance overhead.

3. Most mainstream UI frameworks such as HTML/CSS, WPF, and Qt separate font family and font size as distinct attributes. In LVGL, achieving similar functionality requires additional effort.

### Success criteria

1. lv_font_t represents only the font family and its supported sizes.
2. User can use font size as a style. Default size will be used if the given size is not supported.

### Solution outline

Introducing a new property, LV_STYLE_TEXT_FONT_SIZE, and adding relevant APIs.

The lv_font_t structure would be modified to represent only the font family, with the addition of a supported_size[] field to accommodate different font sizes. The first entry in the supported_size[] array would denote the default font size. For backward compatibility, when LV_STYLE_TEXT_FONT_SIZE is set to its default value or falls outside the range specified in supported_size[], the default font size would be used. For instance, {16, 14, 16, 18} indicates support for three font sizes, with a default size of 16.

For font engines capable of rendering fonts of arbitrary sizes, a special flag (e.g., 0x1000, 0x2000) could be used in conjunction with supported_size[] to denote the range of supported sizes. For example, {16, 0x100A, 0x20C8} denotes support for font sizes ranging from 10 to 200, with a default size of 16.

The get_glyph_dsc and get_glyph_bitmap callback functions would be modified to include a font size parameter. Font position-related fields such as line_height and base_line would be replaced with functions that accept font size as a parameter.

To support getting bitmaps of different sizes for the same bitmap font, the dsc field in lv_font_t could be turned into a pointer array, or the glyph_bitmap and glyph_dsc fields in lv_font_fmt_txt_dsc_t could be turned into pointer arrays. (I’m not sure which one is better.) The first entry would correspond to the default font size, for backward compatibility.

When drawing characters, the font size would be passed as a parameter. The supported_size[] array would be consulted to determine the appropriate font size (using the default size if not found or if outside the specified range). For bitmap fonts, the corresponding bitmap for the specified size would be retrieved from the pointer array based on the index in supported_size[]. For engine-rendered fonts, the corresponding font size value would be passed to the font engine.

### Rabbit holes

I’m not sure to what extent these modifications can maintain backward compatibility.

### Testing

Normal unit testing should be enough.

### Teaching

Add docs and examples.

kisvegabor, 2024-04-17:
Sorry for the late reply. 

I agree in general. To motivation behind coupling the font family and size was that earlier only bitmap fonts were used and I couldn't even imagine that once we can render TTF fonts in runtime. However, MCUs got more powerful and we are here now.

I think there are 3 use cases:
1. a static bitmap font is used with singe sizes (e.g. `lv_font_montserrat_14`)
    - `font_size` is ignored
    - this is what we have now 
2. a list of bitmap fonts is used with various sizes (e.g. `lv_font_montserrat_14/16/18`)
    - `font_size` is a hint, and we need to select the best matching option
    - it applies to bitmap fonts only
    - similarly what you have proposed, we can create a special font which is just a wrapper for all the possible fixed size fonts (e.g. `lv_font_montserrat`). It select the best matching font and calls its `get_glyph_bitmap` and `get_glyph_dsc` callbacks internally. 
3. a vector font is used with any size
    - `font_size` is passed to the font engine

I'm also concerned abut backward compatibility. `font->base_line` and `font->line_height` should be moved to `lv_font_fmt_txt_dsc_t` and (as you mentioned) both callbacks shall be extended with a size parameter. 

We can change the font converter like that and add some `#if`s to check the LVGL version inside the generated C files, but all users need to update their fonts for a newer LVGL version.

Maybe we can keep `font->base_line` and `font->line_height` for the time being and if they are set, use them. If not, we will know that it's a new font. 

What do you think?

cc @W-Mai, could you tell your opinion considering how it would work with FreeType?

zylalx1, 2024-04-30:
Thank you for agreeing with the concept. However, as a beginner on LVGL, I'm not very familiar with the entire codebase. Therefore, I would prefer if someone else could handle the implementation of this feature.

kisvegabor, 2024-05-02:
No problem. Let's keep this issue open and see if someone picks it up.

zylalx1, 2024-05-22:
Maybe we should keep this open?

kisvegabor, 2024-09-05:
Related to https://github.com/lvgl/lvgl/pull/6482
It will be extended to be more generic. Let's close it in favor of it.

closed 2024-09-05 by kisvegabor