lvgl/lvgl #6034: slow dynamic image refresh

cyberumut, 2024-04-06:
### Introduce the problem

when I go from page1 to page2 the background image loads very slowly my processor is esp32s3 I have a 7" screen.

I am loading the background image as .bin in rgb565 format 

Do you have any suggestions

liamHowatt, 2024-04-09:
Hi @cyberumut,

Are you able to create a simple benchmark that compares the speed of LVGL loading the background image vs you sending the image from static buffer over the display interface without LVGL?

closed 2024-05-01 by lvgl-bot

Gabboide, 2024-05-29:
Hi, I have the same problem using Arduino Giga with the display shield. Everythings seems slowed down when a bin image is loaded.

liamHowatt, 2024-06-01:
Are you loading it from a file? Or a C array?

Gabboide, 2024-06-02:
Thanks for your answer Liam. I'm loading it from a bin file saved in the USB host connected to the Arduino GIGA. After the sketch is loaded, everything seems to work fine, but after a little time (usually around 1 minute) it won't show the image any more, instead there is a white background with "No data". This happens if I switch the background between an image and a simple colored background, but also if I use only the image as the background and switch only the graphical objects above it. I don't know if it's a memory issue or if it's linked to the USB management by Arduino. I'm linking you the code I used.

`//Libraries
#include <string.h>
#include "Arduino_USBHostMbed5.h"
#include "DigitalOut.h"
#include "FATFileSystem.h"
#include "Arduino_H7_Video.h"
#include "Arduino_GigaDisplayTouch.h"
#include "lvgl.h"

//Libraries initialization
Arduino_H7_Video          Display(800, 480, GigaDisplayShield);
Arduino_GigaDisplayTouch  TouchDetector;
USBHostMSD msd;
mbed::FATFileSystem usb("usb");

//Constants
const int DISPLAY_WIDTH = Display.width();
const int DISPLAY_HEIGHT = Display.height();
const long refreshIntervalMain = 20;

enum NAV {
  DASHBOARD = 1,  //1
  MENU            //2
  };

//Variables
bool isSerialAvailable = false;

unsigned long previousMillis = 0;

NAV currentNavIndex;
NAV newNavIndex;

static lv_obj_t * currentView = NULL;
static lv_obj_t * oldView = NULL;
static lv_obj_t * bootView = NULL;
static lv_obj_t * bootInfoLabel = NULL;
static lv_obj_t * dashboardBg = NULL;
static lv_obj_t * menuBg = NULL;

//Functions
//--------------------------------------------------------Arduino-------------------------------------------------------
void(* resetFunc) (void) = 0;//declare reset function at address 0
void rebootArduino() {
  resetFunc();
}
//----------------------------------------------------------------------------------------------------------------------

//--------------------------------------------------------Serial--------------------------------------------------------
template<typename T, typename U>
void printSerialMsg( T message, U modifier, bool newLine = false ){
  if(isSerialAvailable) {
    if (newLine) {
      Serial.println(message, modifier);
    } else {
      Serial.print(message, modifier);
    }
  }
}

template<typename T>
void printSerialMsg( T message, bool newLine = false ){
  if(isSerialAvailable) {
    if (newLine) {
      Serial.println(message);
    } else {
      Serial.print(message);
    }
[... 266 lines cut ...]

  initDashboardBackground();
  initMenuBackground();  

  currentView = createViewDashboard();
  
  showObject(dashboardBg);
  showObject(currentView);
  currentNavIndex = DASHBOARD;
  newNavIndex = DASHBOARD;

  lv_obj_del(bootView);
  bootView = NULL;
}

void loop() {
  unsigned long currentMillis = millis();

  if (abs(currentMillis - previousMillis) >= refreshIntervalMain) {
    previousMillis = currentMillis;
    navigation_handler();
    lv_timer_handler();
  }
}
`

liamHowatt, 2024-06-05:
Thanks for all the details! I changed your code the minimum amount to be able to run in the simulator. I don't observe the image disappearing after 1 minute. Do you have `LV_USE_STDLIB_MALLOC` set to `LV_STDLIB_BUILTIN` in your lv_conf.h? Change it to `LV_STDLIB_CLIB` and use mbed_statistics again to watch for a memory leak. If it's already set to `LV_STDLIB_CLIB` and you don't see the memory increasing out of control, then I can suggest some other ideas.