lvgl/lvgl #6035: Crash in _lv_event_mark_deleted() when call lv_obj_del() or lv_obj_del_async

t-walker-wei, 2024-04-07:
### LVGL version

v8.3.10

### What happened?

I work with Platformio in VS Code
**Board**: ESP32-DOIT-DEVKIT-V1 
**Platform**: espressif32
**Framework**: arduino

I'm creating a GUI with creat new widgets and delete old widgets to save memory in ESP32-DOIT-DEVKIT-V1 with 320KB ram.
When I called lv_obj_del() or lv_obj_del_async() in button clicked event callback, I got my device crash frequently with panic ifno:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400ee8a6  PS      : 0x00060530  A0      : 0x800f1d90  A1      : 0x3ffd0600  
A2      : 0x3ffbc464  A3      : 0x00040018  A4      : 0x000001c5  A5      : 0x3f40f874
A6      : 0x3f40f818  A7      : 0x00000012  A8      : 0x800ee8a6  A9      : 0x3ffd05a0  
A10     : 0x00004f69  A11     : 0x3f40f530  A12     : 0x0000009d  A13     : 0x3f40f64c
A14     : 0x3f40f550  A15     : 0x00000012  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0004001c  LBEG    : 0x4008b301  LEND    : 0x4008b311  LCOUNT  : 0xfffffffd  

Backtrace: 0x400ee8a3:0x3ffd0600 0x400f1d8d:0x3ffd0620 0x400f2065:0x3ffd0640 0x400f2072:0x3ffd0660 0x400f6fc7:0x3ffd0680 0x400f6f53:0x3ffd06a0 0x400f7098:0x3ffd06c0 0x400dd49d:0x3ffd06e0 0x400ee77d:0x3ffd0700 0x400ee83e:0x3ffd0720 0x400ef5be:0x3ffd0760 0x400f00c2:0x3ffd0780 0x401019a9:0x3ffd07c0 0x40101a51:0x3ffd07e0 0x400d4a68:0x3ffd0800 0x40179b94:0x3ffd0820

  #0  0x400ee8a3:0x3ffd0600 in _lv_event_mark_deleted at lib/lvgl/src/core/lv_event.c:157
  #1  0x400f1d8d:0x3ffd0620 in lv_obj_destructor at lib/lvgl/src/core/lv_obj.c:454
  #2  0x400f2065:0x3ffd0640 in _lv_obj_destruct at lib/lvgl/src/core/lv_obj_class.c:137 (discriminator 1)
  #3  0x400f2072:0x3ffd0660 in _lv_obj_destruct at lib/lvgl/src/core/lv_obj_class.c:144
  #4  0x400f6fc7:0x3ffd0680 in obj_del_core at lib/lvgl/src/core/lv_obj_tree.c:397
  #5  0x400f6f53:0x3ffd06a0 in obj_del_core at lib/lvgl/src/core/lv_obj_tree.c:374
  #6  0x400f7098:0x3ffd06c0 in lv_obj_del at lib/lvgl/src/core/lv_obj_tree.c:68
  #7  0x400dd49d:0x3ffd06e0 in k99_cancel_trans_btn_clicked_event_cb(_lv_event_t*) at src/ui/ui.cpp:2225
  #8  0x400ee77d:0x3ffd0700 in event_send_core at lib/lvgl/src/core/lv_event.c:469
  #9  0x400ee83e:0x3ffd0720 in lv_event_send at lib/lvgl/src/core/lv_event.c:75
  #10 0x400ef5be:0x3ffd0760 in indev_keypad_proc at lib/lvgl/src/core/lv_indev.c:532
  #11 0x400f00c2:0x3ffd0780 in lv_indev_read_timer_cb at lib/lvgl/src/core/lv_indev.c:104
  #12 0x401019a9:0x3ffd07c0 in lv_timer_exec at lib/lvgl/src/misc/lv_timer.c:313 (discriminator 2)
  #13 0x40101a51:0x3ffd07e0 in lv_timer_handler at lib/lvgl/src/misc/lv_timer.c:109
  #14 0x400d4a68:0x3ffd0800 in loop() at lib/lvgl/src/lv_api_map.h:37
      (inlined by) loop() at src/main.cpp:2020
  #15 0x40179b94:0x3ffd0820 in loopTask(void*) at C:/Users/offic/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
```
So I googled about  **_lv_event_mark_deleted** and **lv_obj_del()**, but I didn't resolved my issue.  I have tried few ways:
1. Try to use lv_obj_del_async()
2. Remove the timers for changing  the subwidgets in this widget
3. I saw somebody said use mutex, but I call this lv_obj_del() function in event callback not task, so I cannot use mutex
4. Change to create two pages with parent null, use lv_scr_load_anim() to load new page and delete old page by set auto_del parameter to true

After a bit search in the LVGL source code, I try to log what happen by modify _lv_event_mark_deleted() to log the event and event code and some where call by lv_obj_del() ( What I can do is only like this because I think maybe have some memory issue in my code, but I cannot debug with debug mode because I don't have JTAG/cmsis-dap/jlink )
```
void _lv_event_mark_deleted(lv_obj_t * obj)
{
    LV_LOG_WARN(  "WALKER: _lv_event_mark_deleted() obj : %p", obj  );
    lv_event_t * e = event_head;
    while(e) {
        LV_LOG_WARN(  "WALKER: _lv_event_mark_deleted() event -> code : %p -> %d", e, e->code );        
        if(e->current_target == obj || e->target == obj) e->deleted = 1;
        e = e->prev;
    }
}
```
What I got when crash:
```
[Warn]  (136.487, +10)   _lv_event_mark_deleted: WALKER: _lv_event_mark_deleted() obj : 0x3ffbc6c8      (in lv_event.c line #153)
[Warn]  (136.498, +11)   _lv_event_mark_deleted: WALKER: _lv_event_mark_deleted() event : 0x3ffd0724, 7         (in lv_event.c line #157)
[Warn]  (136.509, +11)   _lv_event_mark_deleted: WALKER: _lv_event_mark_deleted() event : 0x3ffde8e4, -16777216         (in lv_event.c line #157)
[Warn]  (136.521, +12)   _lv_event_mark_deleted: WALKER: _lv_event_mark_deleted() event : 0x3ffdea0c, 133821    (in lv_event.c line #157)
[Warn]  (136.533, +12)   _lv_event_mark_deleted: WALKER: _lv_event_mark_deleted() event : 0x3ffdea0c, 133821    (in lv_event.c line #157)
[Warn]  (136.545, +12)   _lv_event_mark_deleted: WALKER: _lv_event_mark_deleted() event : 0x3ffdea0c, 133821    (in lv_event.c line #157)
[Warn]  (136.556, +11)   _lv_event_mark_deleted: WALKER: _lv_event_mark_deleted() event : 0x3ffdea0c, 133821    (in lv_event.c line #157)
[Warn]  (136.568, +12)   _lv_event_mark_deleted: WALKER: _lv_event_mark_deleted() event : 0x3ffdea0c, 133821    (in lv_event.c line #157)
[Warn]  (136.580, +12)   _lv_event_mark_deleted: WALKER: _lv_event_mark_deleted() event : 0x3ffdea0c, 133821    (in lv_event.c line #157)
[Warn]  (136.591, +11)   _lv_event_mark_deleted: WALKER: _lv_event_mark_deleted() event : 0x3ffdea0c, 133821    (in lv_event.c line #157)
[Warn]  (136.603, +12)   _lv_event_mark_deleted: WALKER: _lv_event_mark_deleted() event : 0x3ffdea0c, 133821    (in lv_event.c line #157)
[Warn]  (136.615, +12)   _lv_event_mark_deleted: WALKER: _lv_event_mark_deleted() event : 0x3ffdea0c, 133821    (in lv_event.c line #157)
[Warn]  (136.627, +12)   _lv_event_mark_deleted: WALKER: _lv_event_mark_deleted() event : 0x3ffdea0c, 133821    (in lv_event.c line #157)
```
The obj **0x3ffbc6c8** is first child (lv_label) of the widget I want to delete.
I check  the event codes in lv_event_code_t enum, the event code should be 0 to 45 and 0x80(int: 128), but what I got event code is -16777216 and 133821.  And it seems like go inside deadloop by e->prev point to e itself. Is it somewhere overflow in  the memory when try to walk through the linked list of events?
I am a newbie in C/C++, I don't know how to debug this issue, but I make a trick in the code, I make a condition in while(), it works for now, but I don't if it would cause another bug. I change the _lv_event_mark_deleted() to:
```
void _lv_event_mark_deleted(lv_obj_t * obj)
{
    lv_event_t * e = event_head;
    while(e) {
        if ( e -> code >= LV_EVENT_ALL && e -> code <= LV_EVENT_PREPROCESS ){
            if(e->current_target == obj || e->target == obj) e->deleted = 1;
            e = e->prev;
        } else {
            e = NULL; //Exit the while loop when got the event code not in lv_event_code_t enum
        }
    }
}
```
If someone can give me some solution for this issue? Thank you very much in advance.

### How to reproduce?

```
static lv_obj_t *main_page;
static lv_obj_t *status_page;

static void exit_status_btn_clicked_event_cb( lv_event_t *e ){
    if( lv_event_get_code(e) != LV_EVENT_CLICKED ){
        return;
    }
    lv_obj_clear_flag( main_page, LV_OBJ_FLAG_HIDDEN ); // Show the main_page
    if( status_page != NULL && lv_obj_is_valid( status_page ) ){
            lv_obj_del(status_page); // Crash here
            status_page = NULL;
    }
}

static void load_status_btn_clicked_event_cb( lv_event_t *e ){
    if( lv_event_get_code(e) != LV_EVENT_CLICKED ){
        return;
    }
    lv_obj_add_flag( main_page, LV_OBJ_FLAG_HIDDEN ); // Not delete the main_page
    create_status_page();
}

void create_status_page(){
    status_page = lv_obj_create(lv_scr_act());
    /**Codes for create other widgets*/
    lv_obj_t *exit_btn =  lv_btn_create( status_page );
    lv_obj_add_event_cb(exit_btn, exit_status_btn_clicked_event_cb, LV_EVENT_CLICKED, NULL );
}

void create_main_page(){
    main_page = lv_obj_create(lv_scr_act());
    /**Codes for create other widgets*/
    lv_obj_t *status_btn =  lv_btn_create( main_page );
    lv_obj_add_event_cb(status_btn, load_status_btn_clicked_event_cb, LV_EVENT_CLICKED, NULL );
}
```

kisvegabor, 2024-04-08:
I've tested your code and it works well for me. A few ideas:
- `lv_obj_is_valid` is not safe to use this way. It can happen that a widget was freed and on it's exact memory address an other widget was allocated.  `lv_obj_is_valid` will say that it's valid, but it's still not the widget that you wanted. 
- Try increasing the stack size of the LVGL task.

t-walker-wei, 2024-04-23:
@kisvegabor Thank you very much for reply. 
Last week I found where is this crashing problem from.
I have registered custom event code by lv_event_register_id(), and send to my new create object by lv_event_send(), this cause the poiter problem. After not use the cusom event code, lv_obj_del() works well.

closed 2024-04-23 by t-walker-wei

strijar, 2024-09-01:
I encountered exactly the same problem in the same version but under Linux

kisvegabor, 2024-09-02:
@strijar 
Just to be sure, please send a minimal code the reproduce the issue.

strijar, 2024-09-03:
I don't understand yet how this can be done. This is a fairly large project that runs on ARM. I also have custom events, but I tried disabling them and it didn't help. However, the problem occurs sometimes. Sometimes the object is deleted normally, and sometimes it leads to a crash.

kisvegabor, 2024-09-03:
Please enable `LV_USE_ASSERT_OBJ` and `LV_USE_ASSERT_MEM_INTEGRITY`. Hopefully it helps to spot the issue.

strijar, 2024-09-05:
Enabling LV_USE_ASSERT_OBJ and LV_USE_ASSERT_MEM_INTEGRITY didn't help to figure it out. I even installed and configured gdbserver. But I still can't figure out what's causing the error! The crash occurs on e = e->prev;

kisvegabor, 2024-09-05:
Can you use address sanitizer and set `LV_USE_STDLIB_MALLOC LV_STDLIB_CLIB`?

strijar, 2024-09-05:
I turned on sanitizer, it reacts to SIGV and does not see anything else criminal. Yes, I already have LV_MEM_CUSTOM enabled (#define LV_MEM_CUSTOM_ALLOC malloc)

kisvegabor, 2024-09-05:
I'm out of ideas. :slightly_frowning_face: 
Please try to isolate the problem by disabling parts of the application until only a small code snippet remains.

strijar, 2024-09-05:
I have threads in my application, maybe the problem is somewhere in this. I made a message queue for work between threads a long time ago - everything worked. But suddenly something broke. Of course I will figure it out!

t-walker-wei, 2024-09-06:
Somehow it is the pointer issue, the node of widget event link list point to invalid address.
I hope you figure it out soon.

strijar, 2024-09-06:
e->prev is filled only in the lv_event_send() function. Perhaps I have this function called simultaneously from different threads. I will look in this direction

strijar, 2024-09-06:
I think I found something strange. In the _lv_event_mark_deleted() function, the loop is executed while .prev != NULL but I didn't find a place where it was cleared. In the lv_event_send() function, it points to event_head and only to it. Although maybe I misunderstood something?

strijar, 2024-09-09:
So far I have managed to find that the problem occurs when deleting object "X" if there is a request to update object "Y" (lv_obj_invalidate). In this case, the update request goes through my event queue (because it is in a different thread)

kisvegabor, 2024-09-10:
Do you call LVGL from multiple threads? If you pay attention to this: https://docs.lvgl.io/master/porting/os.html

strijar, 2024-09-10:
Yes, I read about it. And that's why I implemented my event queue between threads. LVGL functions are called only from the main thread when a event arrives from my queue.

kisvegabor, 2024-09-11:
Oh, I see.

> So far I have managed to find that the problem occurs when deleting object "X" if there is a request to update object "Y" (lv_obj_invalidate)

How do you trigger to the call of `lv_obj_invalidate`? Do you call it manually?

strijar, 2024-09-11:
Yes, I send a event from the thread to the my queue and when I see it from the LVGL loop, I call lv_obj_invalidate()

kisvegabor, 2024-09-12:
I don't understand every part of your code (in lack the whole context) but [this part is suspicious](https://github.com/strijar/brass_gui/blob/main/src/events.c#L72-L81) as it's not protected by a mutex. What if the queue get updated an `item` will point to a new something. Can you try locking the whole `event_obj_check` function?

strijar, 2024-09-12:
This part is executed in the main LVGL loop (https://github.com/strijar/brass_gui/blob/main/src/main.c#L111C1-L121C1). That's why it doesn't need to be protected. The mutex that is used there protects the queue itself. I have now rewritten this code using a linked event list (I haven't committed it yet). But it didn't solve the problem.

kisvegabor, 2024-09-18:
Have you found a solution?

strijar, 2024-09-18:
So it turned out to be much more complicated. I have it done so that you can close a window by pressing the "Esc button" or another one. So when a window is deleted by pressing the Esc button, it crashes. But when by pressing another button, everything is fine. As a solution, I have removed the ability to close by pressing Esc for now.

strijar, 2024-09-23:
I found a guaranteed working workaround - if you use lv_ob_del_delayed() instead of lv_obj_del() then everything works

strijar, 2024-09-24:
It seems the problem was due to the use of threads. As I wrote, I "isolated the use" of only lv_obj_invalidate() and lv_event_send() and it worked for some time. Yesterday I removed my event queue and started implementing blocking via mutexes as the documentation suggests. And the problem (I hope so!) went away.

kisvegabor, 2024-09-25:
Ah, these are big mysteries of the programming universe. I'm sorry that I couldn't help more. :slightly_frowning_face: