lvgl/lvgl #6038: Flush wait callback clarifications

faxe1008, 2024-04-07:
### LVGL version

v9.1.0

### What happened?

I am currently working on migrating the zephyr gluecode to the new v9.1.0 release. In the past (v8.4) we had support for a seperate flush thread which would speed up rendering by quite a bit. It seems like some underlying assumption that LVGL makes in regards to flushing and waiting for the flush to finish has changed in the new release. This is the basic setup that we are using on v9.1.0:

```
struct lvgl_display_flush {
	lv_display_t *display;
	uint16_t x;
	uint16_t y;
	struct display_buffer_descriptor desc;
	void *buf;
};

K_SEM_DEFINE(flush_complete, 0, 1);
/* Message queue will only ever need to queue one message */
K_MSGQ_DEFINE(flush_queue, sizeof(struct lvgl_display_flush), 1, 1);

void lvgl_flush_thread_entry(void *arg1, void *arg2, void *arg3)
{
	struct lvgl_display_flush flush;
	struct lvgl_disp_data *data;

	while (1) {
		k_msgq_get(&flush_queue, &flush, K_FOREVER);
		data = (struct lvgl_disp_data *)lv_display_get_user_data(flush.display);

		display_write(data->display_dev, flush.x, flush.y, &flush.desc, flush.buf);

		lv_display_flush_ready(flush.display);
		k_sem_give(&flush_complete);
	}
}

K_THREAD_DEFINE(lvgl_flush_thread, CONFIG_LV_Z_FLUSH_THREAD_STACK_SIZE, lvgl_flush_thread_entry,
		NULL, NULL, NULL, K_PRIO_COOP(CONFIG_LV_Z_FLUSH_THREAD_PRIO), 0, 0);

void lvgl_wait_cb(lv_display_t *display)
{
	k_sem_take(&flush_complete, K_FOREVER);
}

void lvgl_flush_display(struct lvgl_display_flush *request)
{
	/*
	 * LVGL will only start a flush once the previous one is complete,
	 * so we can reset the flush state semaphore here.
	 */
	k_sem_reset(&flush_complete);
	k_msgq_put(&flush_queue, request, K_FOREVER);
	/* Explicitly yield, in case the calling thread is a cooperative one */
	k_yield();
}

void lvgl_flush_cb_24bit(lv_display_t *display, const lv_area_t *area, uint8_t *px_map)
{
	uint16_t w = area->x2 - area->x1 + 1;
	uint16_t h = area->y2 - area->y1 + 1;
	struct lvgl_display_flush flush;

	flush.display = display;
	flush.x = area->x1;
	flush.y = area->y1;
	flush.desc.buf_size = w * 3U * h;
	flush.desc.width = w;
	flush.desc.pitch = w;
	flush.desc.height = h;
	flush.buf = (void *)px_map;
	lvgl_flush_display(&flush);
}

void set_lvgl_rendering_cb(lv_display_t *display)
{
	lv_display_set_flush_wait_cb(display, lvgl_wait_cb);
	lv_display_set_flush_cb(display, lvgl_flush_cb_24bit);
}
```

So essentially we have `lvgl_flush_thread` which is waiting for any new flush task in the `flush_queue`. After processing the item the thread gives the `flush_complete` semaphore, which is awaited by the wait callback.

Setting a breakpoint on `lvgl_wait_cb` I can see it is called before the `lvgl_flush_display` is called even once - leading to eternal waiting on the semaphore. What is the intended behaviour for the flush callback? Does it need to perform checks if flushing is actually needed? (the queue not being empty?). Is calling `lv_display_flush_ready` still needed?

kisvegabor, 2024-04-08:
Hey,

Your approach is almost correct.

>  Does it need to perform checks if flushing is actually needed? (the queue not being empty?). 

My approach was that  LVGL can call the `flush_wait_cb` at any time for synchronization, even if there is no flushing is progress. However, as I'm looking into it again, it seems we can handle optimize it internally and call the `flush_wait_cb` only if flushing is in progress. Could you try adding this patch:
```diff
diff --git a/src/core/lv_refr.c b/src/core/lv_refr.c
index 1dfbab79f..34234229d 100644
--- a/src/core/lv_refr.c
+++ b/src/core/lv_refr.c
@@ -1078,11 +1078,15 @@ static void wait_for_flushing(lv_display_t * disp)
     lv_display_send_event(disp, LV_EVENT_FLUSH_WAIT_START, NULL);
 
     if(disp->flush_wait_cb) {
-        disp->flush_wait_cb(disp);
+        if(disp->flushing) {
+            disp->flush_wait_cb(disp);
+        }
+        disp->flushing = 0;
     }
     else {
         while(disp->flushing);
``` 

> Is calling lv_display_flush_ready still needed?

No, you can use either the `lv_display_flush_ready` or the `flush_wait_cb`.

faxe1008, 2024-04-09:
@kisvegabor I applied your suggested patch and I now see at least one frame being rendered. For now I will add another synchronization primitive, so we do not have to wait for 9.2.0 but I will be happy to remove it  once this makes it in :^).

I can obeserve the semaphore being given and taken at regular intervals, so I guess this remaining issue is situated in our gluecode. Regarding the flush callback 

I have one question remaining: in v9 the `set_px_cb` was removed. The docs mention that this now needs to be done in the `flush_cb` itsself. Is my assumption correct that the `px_map` parameter in the example:
```
void my_flush_cb(lv_display_t * display, const lv_area_t * area, void * px_map)
```
Is in the color format of the display? If thats the case is there any way to "add" a new color format to be handeled by the drawing engine? For example epds with odd color counts?

kisvegabor, 2024-04-15:
Let me know how it goes and if we need to change anything else on our end :slightly_smiling_face: 

Yes `px_map` is in the color format of the display and you can convert it to anything you like. If you convert it to a smaller bpp format, you can do the conversation in place in partial and full mode (in direct mode having 2 color formats in a screen sized buffer wouldn't work). 

A PR for [L8](https://github.com/lvgl/lvgl/pull/5800) (8 bit grayscale) is almost ready. It should be the preferred color format for monochrome-like displays.

Adding new target color formats is also possible, but we should add it in the upstream LVGL for everyone's happiness :slightly_smiling_face: However, it's a larger work and it's not worth the effort for a single special format. 

Could you tell what target color format do you miss, so that we can see if it can/shall be implemented?

kisvegabor, 2024-05-02:
@faxe1008 how it goes? :slightly_smiling_face: 
Have you find any other issues that we can fix on our side?

faxe1008, 2024-05-02:
Hey @kisvegabor sorry I missed your previous reply. Your initial response did help me to resolve the issue with the flushing thread 👍. 

For the color formats I was intrested in the I1 format as this is the format currently used by the monochrome displays in Zephyr. I followed the L8 pull request and the amount of work required to pull it off seems quite substantial, although I'd be intrested in giving it a try. Might need some guidance along the way though :^). 

As for this issue, I'd appreciate if your suggested patch makes it into v9.1.1. Thanks for you help on this. I might raise a PR tomorrow to add an explanation the docs so others benefit as well :^)

kisvegabor, 2024-05-03:
I've opened https://github.com/lvgl/lvgl/pull/6174 to add the fix.

I was also thinking about adding support for L1 target color format. Do you think it's enough to horizontally mapped the bits with MSB on the left arrangement? 

Just let me know if you have any question regarding L8 or anything else. :slightly_smiling_face:

closed 2024-05-10 by kisvegabor