lvgl/lvgl #6046: can't move to tile >10

FloHomi, 2024-04-09:
### LVGL version

9.0.1-dev

### What happened?

I have a tile_view object and add more than 10 tiles to the columns. When I want to jump to the tiles using the cb function it works just fine, until tile >10. There it does nothing. I noticed that the x coords that it wants to jump to become negative, but I am not scilled enouth to track it down further...
Maybe someone can help here.
Thanks!

### How to reproduce?

```
lv_obj_t* content_tile_view;
#define HOME_MENUE_ICON_COUNT12
lv_obj_t* sidebar_tiles[4]
lv_obj_t* home_menue_tiles[HOME_MENUE_ICON_COUNT]

static void select_home_menue_tile_cb(lv_event_t* e){
    lv_obj_t* target = lv_event_get_current_target(e);
    int page = lv_event_get_user_data(e);
    if (home_menue_icon_data[page].corresponding_tile != NULL && page < HOME_MENUE_ICON_COUNT) {
        lv_tileview_set_tile(content_tile_view, home_menue_tiles[page], LV_ANIM_OFF);
    }
}

static void create_tiled_content(lv_obj_t* parent)
{
    content_tile_view = lv_tileview_create(parent);

    //Tile1: Light Menue
    sidebar_tiles[0] = lv_tileview_add_tile(content_tile_view, 0, 0, LV_DIR_BOTTOM);

    //Tile2: Home Menue
    sidebar_tiles[1] = lv_tileview_add_tile(content_tile_view, 0, 1, LV_DIR_VER);

    for (int i = 0; i < HOME_MENUE_ICON_COUNT; i++) {
       home_menue_tiles[i] = lv_tileview_add_tile(content_tile_view, i + 1, 1, LV_DIR_NONE);
    }
}
```

liamHowatt, 2024-04-09:
Thanks @FloHomi, I see the issue. You're right, the x coordinate of the 11th tile becomes negative. From what I could gather, it's happening because the positions of the tiles are set using the special percent coordinates, which is fine unless the percent value is > 1000% at which point that's the representation of a negative percent value. It's greater than 1000% in this case because it's column number 11 * 100%.

Can we increase this limit? There doesn't seem to be a shortage of available bits. Percentages are `_LV_COORD_TYPE_SPEC`. The only other `_LV_COORD_TYPE_SPEC` coord is `LV_SIZE_CONTENT` which we can bump up from `2001` to `LV_COORD_MAX` and then the max stored percent value can be `LV_COORD_MAX - 1` and the max positive `(LV_COORD_MAX - 1) / 2`.

I am opening a PR.

FloHomi, 2024-04-10:
Great :) thank you very much @liamHowatt! 
is it stupid of me to use so many tiles to organise my GUI? I will have it on an embedded system later. Or is I better to always create and destroy an container object? 
I guess it is not a common practice to to this current behaviour.

liamHowatt, 2024-04-10:
> Or is I better to always create and destroy an container object?

I think using many tiles is totally fine unless memory is scarce.

I would like a second opinion on this but I believe that out-of-view or hidden objects have a very low CPU footprint. The memory they occupy is the main overhead. If you exhaust the available memory it may be time to consider creating and destroying. You would lose the exact state of your GUI screens, however. It's a tradeoff.

liamHowatt, 2024-04-10:
https://docs.lvgl.io/master/overview/display.html#screens
Tileview works but if you're looking for something simpler this may be what you're looking for.

```
lv_obj_t * old_screen = lv_screen_active();
lv_obj_t * new_screen = lv_obj_create(NULL);
...
lv_screen_load(old_screen);
```

Screens take up the whole display like the tiles. When they're created they become the active screen. `lv_screen_load` switches the active screen to an existing one.

FloHomi, 2024-04-10:
Thank you again for your great input 👍 
Best regards
Florian

PR #6051 (2024-04-21): fix(area): Increase coordinate percent range beyond +-1000
### Description of the feature or fix

Fixes #6046

Use all the available bits in a `int32_t` coordinate for storing percent values instead of limiting it to just +-1000%. See the issue for more details. The issue was a case where percent values >1000% were failing to be created in a `tileview` with more than 10 tiles.

The new coordinate percent range is +-268435455%

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if r [... cut]
closed 2024-04-21 by kisvegabor