lvgl/lvgl #6048: Using fallback font with keyboard widget causes crash

zjanosy, 2024-04-09:
### LVGL version

v9.0.1dev

### What happened?

Wanted to change the keyboard widget font to a TrueType font. This font, however, does not include the special symbols needed for the keyboard. Tried to set the "fallback" to the Montserrat 26 bitmap font, but this caused a crash in `lv_cache_entry_get_ref()`:

Exception thrown: read access violation.
entry was 0xFFFFFFFFFFFFFFF7.

### How to reproduce?

```
    /*Create TrueType font*/
    lv_font_t *my_font = lv_tiny_ttf_create_data(font_ABCMonumentGrotesk_Regular, sizeof(font_ABCMonumentGrotesk_Regular), 36);
    /*Add fallback font for special keyboard glyphs*/
    my_font->fallback = &lv_font_montserrat_26;
    /*Create keyboard*/
    lv_obj_t *p = lv_keyboard_create(lv_screen_active());
    lv_obj_set_style_text_font(p, my_font, LV_PART_ITEMS | LV_STATE_DEFAULT);
```

liamHowatt, 2024-04-09:
Fixed by #5985. It fixed a subtle bug.

https://github.com/lvgl/lvgl/blob/51b8d8e4ea7ed4cb7a4ad295234740b81fc46e6b/src/draw/lv_draw_label.c#L441-L443

should be

```c
if(g.resolved_font && g.resolved_font->release_glyph) { 
    g.resolved_font->release_glyph(font, &g); 
}
```

I am opening a PR that backports this fix to the v9.0 release branch. From what I can tell, the #5985 PR can be safely cherry picked.

kisvegabor, 2024-04-12:
On my end it works with with the latest `master`. @zjanosy please confirm it.

kisvegabor, 2024-04-28:
The PR is merged, closing this issue.

PR #6052 (2024-04-10): feat(font): add lv_font_glyph_release_draw_data api to release glyph data
### Description of the feature or fix

Fixes #6048

This PR backports @W-Mai's #5985 PR to the `release/v9.0` branch.

Does the version need to be updated anywhere? Such as in lvgl.h?

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen [... cut]
closed 2024-04-28 by kisvegabor