lvgl/lvgl #6049: Demo for the OSAL

faxe1008, 2024-04-09:
### Problem to solve

During development of LVGL integrations one task with the new release is the support of the OSAL. It would be very useful to have tests of some sort that integrators could refer to to verify their implementation. 

### Success criteria

LVGL integrators have a convenient way of verifying their OSAL implementation

### Solution outline

I would propose this to be a demo rather then a feature though, because there might be customer preferences as to what framework to use. A demo could be vendor neutral way to provide the results by using print statements and `LV_ASSERT` etc.

### Rabbit holes

None that I know of.

### Testing

Running integrations tests (at least for a linux pthread osal) within the pipeline.

### Teaching

Adding a note to the OS porting section and of course to the demos.

kisvegabor, 2024-04-15:
Hi,

Just to be sure I understand correctly: in a demo or example we should create a few mutexes and threads with printf-s to demonstrate that OS layer is working well. 

If so, I think it should be rather an example which could be also a good starting point for anyone using LVGL with an OS. (E.g. show that `lv_timer_handler` needs to be protected by a mutex). I can imagine something like this:
1. We have the main thread in which a button is create
2. A new thread is created, which creates a label, and waits for a signal
3. When the button is pressed, a signal is triggered, and the other thread increments a counter and updates the label.

Besides the example - as you have mentioned - there should be a test too at least for pthread, but I think it's simpler as we can write any complicated code.

kisvegabor, 2024-05-02:
Sorry for not responding. GitHub doesn't send notification about reactions and I didn't notice the :+1: 

We will work on it!

faxe1008, 2024-05-02:
Hey @kisvegabor 😃, no worries. Maybe I'll find some time to work on this :^)

PR #6182 (2024-06-01): feat(demos): add demo for the OSAL
Adds a demo demonstrating the use of the OSAL.

Resolves: #6049

### Description of the feature or fix

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/script [... cut]
closed 2024-06-01 by FASTSHIFT