lvgl/lvgl #6053: Layer Blending

kdschlosser, 2024-04-10:
### Introduce the problem

Currently there seems to be no way to control the blending of 2 layers. While this example doesn't deal with multiple layers it shows what the lack of blending support for layers assumes.

[image]

Look at the knob where it is sitting on top of the indicator. This is not supposed to happen but it does. This is because of how LVGL renders the indicator and then it renders the knob on top of that.

If there was the ability to set different blend modes at the layer level this kind of thing could be mitigated. A layer could be created for the knob and a blend mode could be set so if overlapping colors exist and they are identical including the alpha values then when the images are merged together the pixels on the top layer that have color and an alpha of > 0 would be the ruling color otherwise the layer on the bottom would be the ruling color. 

### Proposal

If you look at almost any image editing software that supports layers there is the ability to change the blend mode between layers and also the ability to set an additional opacity that is global for the layer that is specified.

I do not think this would be too hard to add to LVGL as it already has some blending functions built into it. The current supported blend modes are

  * normal
  * additive
  * subtractive
  * multiply

Here is a list of different blend modes that could be supported by LVGL in addition to the ones above

  * lighten: `max (A, B)`
  * darken: `min(A, B)`
  * difference: `abs(B - A)`
  * divide: `A / B`
  * color dodge: `B / (1.0 - A)`
  * linear dodge: `A + B`
  * color burn: `1.0 - (1.0 - B) / A`
  * linear burn: `A + B - 1.0`
  * linear light: `2.0 * A + B - 1.0`
  * hard mix:
  ```
  if (A <= (1 - B))
      0
  else
      1
  ```
  * pin light:
  ```
  if 2.0 * A - 1.0 > B
      2.0 * A - 1.0
  else if (A < 0.5 * B) 
      2.0 * A
  else
      d
  ```
  * soft light: 
  ```
  if A < 0.5
      B - (1.0 - 2.0 * A) * B * (1.0 - B)
  else if B < 0.25
      B + (2.0 * A - 1.0) * B * ((16.0 * B - 12.0) * B + 3.0)
  else
      B + (2.0 * A - 1.0) * (sqrt(B) - B)
  ```
  * hard light:
  ```
  if A < 0.5
      2.0 * A * B 
  else
      1.0 - 2.0 * (1.0 - A) * (1.0 - B)
  ```
  * vivid light:
  ```
  if A < 0.5
      1.0 - (1.0 - B) / (2.0 * A)
  else
      B / (2.0 * (1.0 - A))
  ```
  * overlay: 
  ```
  if B < 0.5 
      2.0 * A * B 
  else 
      1.0 - (2.0 * (1.0 - A) * (1.0 - B))
   ```
  * screen: `1.0 - (1.0 - A) * (1.0 - B)` or `A + B - A * B`
  * exclusion: `A + B - 2.0 * A * B`
  * xor: `A ^ D`
  * average: `(B + A) / 2.0`
  * ghosting: `gray(B) - gray(A) + B + A / 5.0`
  * luminosity: `gray(A) + B - gray(B)`

gray is defined as
```
gray(pixel)
    (pixel.red * 0.299 + pixel.green * 0.587 + pixel.blue * 0.114) / 255.0
```

A is a pixel in the active layer and B is the pixel in the next layer down that has the same coordinates. A pixel is a vector containing red, green blue and alpha. The equation gets performed on each of the channels in the pixel. all equations assume the channel value is normalized into the 0.0 to 1.0 range. 

I believe the above should be able to be implemented fairly easily for the purposes of adding additional blending modes. Adding support to the layers would be more involved. I would make a go at doing it but I am not able to fully wrap my head around how the whole layer system in LVGL works.

kisvegabor, 2024-04-15:
> A layer could be created for the knob and a blend mode could be set so if overlapping colors exist and they are identical including the alpha values then when the images are merged together the [...]

It could work, but it doesn't sound like a standard blending mode. 

---

Now layers are created automatically on widget level if there is [a style property which makes it required](https://github.com/lvgl/lvgl/blob/master/src/core/lv_obj_style.c#L954-L965). (E.g. transformations). We could add a function to create layers by the users in the draw events to transform/blend/mask/etc the parts too, but there is an issue. If random layers can be created in the draw events and they can be drawn anywhere (e.g. by rotating around an external pivot point), LVGL might not invalidate the final area of the part, as it doesn't know the settings from the custom draw event. 

We can either say
- the part can occupy the same or small area as normally
- we can add a style property to enlarge the invalidated area (not ideal as it needs to be handled everywhere)  

However I think it'd be mostly okay, as typical use cases involve 
- rotating the knob of the arc to the angle of the arc 
- changing the blend mode of a part
- add mask to a part
- anything else?

These can be drawn to the same size (if the arc knob is a circle)

kdschlosser, 2024-04-16:
You understand what I am getting at with it. when the parts are rendered they are rendered onto the same layer. if they were rendered onto different layers and then all brought together into a single layer that would solve the inability to have the knob be the exact same as the indicator when setting the opacity < 255.

kisvegabor, 2024-04-17:
Actually we don't need a new layer for that. When the knob blended we can examine the background pixel on the fly and do some magic. 

However I'm not sure that we need to solve the issue of the semi-transparent knob like this. What if the indicator is blue and the  knob is red. What would be expected behaviour?

kdschlosser, 2024-04-17:
> However I'm not sure that we need to solve the issue of the semi-transparent knob like this. What if the indicator is blue and the knob is red. What would be expected behaviour?

This is where layers would step in and the user being able to provide instruction as to what the blending mode should be for the layers. Get it now?

So when I user a paint program that supports layers I have the ability to change the blend modes for each layer. Now I am not saying that you should do that for each layer but maybe give the user the ability to change it. Right now it only uses additive, there is no reason why LVGL shouldn't support the layers I listed above and allow the user to specify what one to use when rendering a widget.

kisvegabor, 2024-04-19:
> This is where layers would step in and the user being able to provide instruction as to what the blending mode should be for the layers. Get it now?

Not yet :smile: Take this as an example:
[image]

Let's say the knob is a layer, but when we merge it, the indicator is already there, burnt into the buffer. We cannot remove it. So there is no way to hide the indicator under a semi-transparent knob.

kdschlosser, 2024-04-19:
I know. But if the user has the ability to change how the layers are blended together how the color are mixed up together can be altered. So say I want the top layer to override the layer below it. This would hold true for any pixel that is not completely transparent. So the issue you posted above becomes a non issue because the indicator would bump to the side of the knob and not the middle of it.

kisvegabor, 2024-04-28:
> So the issue you posted above becomes a non issue because the indicator would bump to the side of the knob and not the middle of it.

If you want that, you can just make the know 100% opa.

closed 2024-05-22 by lvgl-bot