lvgl/lvgl #6057: Montserrat 10 is missing in the LVGL default title font list in menuconfig

tnqzh123, 2024-04-10:
### LVGL version

9.1.0

### What happened?

I'm using LVGL on ESP32-S3. I've noticed that LVGL have a built-in Montserrat 10 font, but it isn't in the default title font list in menuconfig, while all other fonts are in the list.

[image]
[image]

I found `LV_FONT_DEFAULT_MONTSERRAT_10` is missing in the `LV_FONT_DEFAULT` choice in Kconfig file:

https://github.com/lvgl/lvgl/blob/4d0c02982758304fa39d9b4ddc4be0dc991ed00f/Kconfig#L688-L700

and `CONFIG_LV_FONT_DEFAULT_MONTSERRAT_10` is still in src/lvgl_conf_kconfig.h: 

https://github.com/lvgl/lvgl/blob/4d0c02982758304fa39d9b4ddc4be0dc991ed00f/src/lv_conf_kconfig.h#L146-L154

After adding `LV_FONT_DEFAULT_MONTSERRAT_10` to the Kconfig file, it appeared in the menuconfig, and working well.

[image]

liamHowatt, 2024-04-10:
Thanks @tnqzh123, do you feel like opening a PR for it?

tnqzh123, 2024-04-10:
> Thanks @tnqzh123, do you feel like opening a PR for it?

@liamHowatt Of course sure. #6058

PR #6058 (2024-04-12): fix(kconfig): add Montserrat 10 font to default title font list in Kconfig (#6057)
### Description of the feature or fix

Add Montserrat 10 font to default title font list in Kconfig.

Fix #6057 

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_ [... cut]
closed 2024-04-12 by kisvegabor