lvgl/lvgl #6059: close the frame, break!

xinsuinizhuan, 2024-04-11:
### LVGL version

v9.1.1

### What happened?

in windows，when close the pop window，break:
[image]
because the disp_refr_timer is NULL in lv_sysmon.c

xinsuinizhuan, 2024-04-11:
to add a judget:
[image]

liamHowatt, 2024-04-11:
Hi, your default display is not set for some reason. Did you create one? If so, did you use `lv_sdl_window_create` or `lv_windows_create_display`?

liamHowatt, 2024-04-11:
Also, since you appear to have a debugging environment, could you please check to see if you reach this line and whether the returned timer is NULL?

https://github.com/lvgl/lvgl/blob/eb62ddf4ea8e84ea76cc89d932ce20f0066b8618/src/display/lv_display.c#L95

xinsuinizhuan, 2024-04-15:
> Hi, your default display is not set for some reason. Did you create one? If so, did you use `lv_sdl_window_create` or `lv_windows_create_display`?

 i set #define LV_USE_OS   LV_OS_WINDOWS in lv_conf.h and i use lv_windows_create_display:
[image]

xinsuinizhuan, 2024-04-15:
> disp->refr_timer = lv_timer_create

it not NULL:
[image]

liamHowatt, 2024-04-15:
Strange. I am going to use Windows soon and will look at this. You seem comfortable with debugging. I welcome you to investigate the issue if you want.

liamHowatt, 2024-04-18:
Yes! I reproduced the problem.

Sysmon expects a display to exist as long as `lv_deinit` hasn't been called, so if all the displays have been deleted then sysmon will encounter the `NULL` access. So your suggested fix is a possible solution. Although, the Windows driver is the only one to not immediately call `lv_deinit()` or `exit(0)` on the last display delete compared to the SDL and X11 drivers, which do. So maybe it would be better to change the Windows driver to do that? It actually does, but in a 200ms timer callback that checks if the display was deleted. Also, the Windows driver calls `lv_display_delete` and `lv_timer_delete` in a separate thread which [should have a mutex](https://docs.lvgl.io/master/porting/os.html#tasks-and-threads), and also add a mutex in the Windows example code. If it can be done without threads and mutexes maybe that'd be even better?

cc @MouriNaruto 

https://github.com/lvgl/lvgl/blob/be4a9d1e735f191151aae5acf4a2018ca2e86f0b/src/drivers/windows/lv_windows_context.c#L626-L644

MouriNaruto, 2024-04-19:
@liamHowatt By design becuase Windows backend supports multiple LVGL display instances.

We need think more better workaround than other backend does.

Kenji Mouri

liamHowatt, 2024-04-19:
I appreciate your input. Do you have any thoughts about what kind of workaround would be good?

kisvegabor, 2024-04-19:
I think it should be fixed by https://github.com/lvgl/lvgl/pull/6073

I'm on Linux, could you try it out on Windows?

xinsuinizhuan, 2024-04-28:
i try the newest code, still break:
[image]

kisvegabor, 2024-04-29:
#6073 is not merged yet.

liamHowatt, 2024-04-29:
Tested just now on Windows. I can confirm the crash does not happen anymore with the sysmon changes for the trivial case.

However, when I create and delete 1M timers in a loop and close the window, the debugger throws a different memory exception each time. It's because LVGL is being reentered by the separate Windows thread when the windows gets closed. For a few different runs, I get a memory access exception [here](https://github.com/lvgl/lvgl/blob/d0a9e8b817132335db504543c16437adf0a244b5/src/misc/lv_ll.c#L233), [here](https://github.com/lvgl/lvgl/blob/d0a9e8b817132335db504543c16437adf0a244b5/src/misc/lv_ll.c#L333), [here](https://github.com/lvgl/lvgl/blob/d0a9e8b817132335db504543c16437adf0a244b5/src/stdlib/builtin/lv_tlsf.c#L591).

I'm using LVGL's allocator with 512 MB space. Here is the test code
```c
    while (1)
    {
        uint32_t time_till_next = lv_timer_handler();
        lv_delay_ms(time_till_next);

        static lv_timer_t * timers[1000000];
        for (int i=0; i<1000000; i++) {
            timers[i] = lv_timer_create(timer_cb, 100, NULL);
        }
        for (int i=0; i<1000000; i++) {
            lv_timer_delete(timers[i]);
        }
    }
```

The main thread is in `lv_timer_create` or `lv_timer_delete` when [this thread](https://github.com/lvgl/lvgl/blob/d0a9e8b817132335db504543c16437adf0a244b5/src/drivers/windows/lv_windows_context.c#L639) calls `lv_display_delete` and `lv_timer_delete`.

kisvegabor, 2024-05-02:
Great! Thanks for testing it! 

I'm thinking about having `lv_lock()` and `lv_unlock()` functions which are locking/unlocking an `lv_mutex`. These functions could be called in `lv_timer_handler` internally, and the user could use these functions too in the drivers or the application code. 

What do you think?

liamHowatt, 2024-05-07:
So it's an instance of `lv_mutex_t` that's shared by both the internals and the user? It would be use the mutex implementation from `LV_USE_OS`? It would live in the global, get initialized at init, and be accessed using `lv_(un)lock`. This is a good idea!

For a first PR, are you thinking:

- Add `lv_lock` and `lv_unlock`
- Add init code
- Update thread/mutex docs
- Use the lock in
  - `lv_timer_handler`
  - the Windows driver
- In `lv_windows_context.c` add `#if LV_OS_WINDOWS != 1 #error` because now it depends on the Windows OSAL for the mutex.

One thing though. In my example that causes the memory faults, that would not be fixed by this unless the user adds the `lv_lock` around that code. Is that the expected requirement from the user? Or else some more parts of LVGL would need to acquire the lock. i.e., the allocator, mainly anything in the global singleton.

kisvegabor, 2024-05-08:
> For a first PR, are you thinking: [...]

Perfect summary! :100: 

> In my example that causes the memory faults, that would not be fixed by this unless the user adds the lv_lock around that code. Is that the expected requirement from the user?

In a general case I assume that
- in the main thread after a quick init the user calls  `lv_timer_handler` (protected internally by `lv_(un)lock`)
- in windows context thread we will add `lv_(un)lock()`

So two weak point remained:
1. any code before `lv_timer_handler` (I think the timer creation/deletion in your code was added there too): If it's a concern the user  can use `lv_(un)lock()` here too.
2. other threads created by the user: we can expect the user to use `lv_(un)lock()`

> Or else some more parts of LVGL would need to acquire the lock. i.e., the allocator, mainly anything in the global singleton.

I would like to avoid that. I think it's a very deep rabbit hole as we need to protect all the widgets too. E.g. what if `lv_label_set_text()` wants to adjust the same label at the same time.

liamHowatt, 2024-05-13:
All sounds good! I will open a PR after the API retirement.

liamHowatt, 2024-06-12:
Not stale.

PR #6425 (2024-07-08): fix(Windows): use global lock
### Description of the feature or fix

Fixes #6059

Use the LVGL global lock in the Windows backend to control access to data and LVGL APIs shared with the Windows backend thread message callback.

Prevent memory faults, especially when the display is deleted.

CC @MouriNaruto

Fixes #6428

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you adde [... cut]
closed 2024-07-08 by FASTSHIFT