lvgl/lvgl #6060: when i create a chart by lv_chart_create, but the lv_chart_set_axis_tick and lv_chart_set_zoom_x function is discard, how to replace?

xinsuinizhuan, 2024-04-11:
### LVGL version

v9.1.1

### What happened?

in 8.3 version, lv_chart_create create the chart ,then use  lv_chart_set_axis_tick and  lv_chart_set_zoom_x to set, but these function discard in 9.0 version ,how to replace them?
[image]

liamHowatt, 2024-04-11:
Hey,

This example uses `lv_scale` as a chart axis. You should be able to achieve all of the same effects `lv_scale`. Let me know if you need any clarification or help achieving what you need.

https://docs.lvgl.io/master/examples.html#axis-ticks-and-labels-with-scrolling

xinsuinizhuan, 2024-04-15:
I think the  lv_chart_set_axis_tick and lv_chart_set_zoom_x is needed, if they could not achieve the effect, then i could use the scale, not all functions to custom.

xinsuinizhuan, 2024-04-15:
others, the scale need to create a obj, then to create the scale, but when i use the  lv_chart_set_axis_tick and lv_chart_set_zoom_x，it set to chart directly, not need to create the obj.

liamHowatt, 2024-04-15:
Perhaps we can bring those functions back. It must have been changed for a reason though. [This](https://github.com/lvgl/lvgl/commit/6163437afd4502a5ecf9940c805a06cc3f161f86) is the commit that removed those functions and [here](https://github.com/lvgl/lvgl/commits/master/?after=7e7590a0a3da66fc2ab665cc1bb8786c2e71ed1c+1030) are all the other changes that day, Oct 12, 2023. @kisvegabor, what's the history behind it?

xinsuinizhuan, 2024-04-26:
> We have a separate scale widget, because the scale-like behaviour is uses on a lot places. This way the scale can be used in a more flexible and modular way. For example you can make it scroll with the chart or make it fixed.
> 
> However I agree that some helper functions would be nice to make it easier to add the axis to the chart. (or at least some better example would be great) I can't immerse in now, so ideas are very welcome.

Yes，i agree to custom, but i thongt not all widgets that use the scale to custom，some helper functions would be nice to make it easier to add the axis to the chart。when to add them back?

closed 2024-05-18 by lvgl-bot