lvgl/lvgl #6061: height of lv_roller is abnormal

wachsh302, 2024-04-11:
### LVGL version

v8.3.11

### What happened?

use lv_roller with font size 20 of SourceHanSansCN_Normal: normal
but use lv_roller with font size 28 of SourceHanSansCN_Normal: the height is too small, text is truncated
[image]

liamHowatt, 2024-04-11:
Hey,

I tried using that font at size 20 and 28 on a roller in v8.3. I used default theme and mono theme.

If I understand correctly, is the problem the top edge of the "00" is getting cut off? If so, try adding `lv_roller_set_visible_row_count(roller, 3)` to the roller. This is something that happens for all larger fonts.

Before
[image]

After
[image]

Or is the problem that the gray selection field is shifted upwards for the "30" roller? I could not reproduce that.

kisvegabor, 2024-05-03:
As the issue seems to be resolved, I close it.

closed 2024-05-03 by kisvegabor