lvgl/lvgl #6066: When different font sizes are set for LV_PART_MAIN and LV_PART_SELECTED of lv_roller, the text display position of the LV_PART_SELECTED section will be slightly higher.

BigWhite0, 2024-04-12:
### LVGL version

v8.3.8

### What happened?

When different font sizes are set for LV_PART_MAIN and LV_PART_SELECTED of lv_roller, the text display position of the LV_PART_SELECTED section will be slightly higher.
[image]

When making the following modifications to lv_roller.c, it will restore normal functionality.
[image]
[image]

### How to reproduce?

void lv_example_roller_2(void)
{
    /*A style to make the selected option larger*/
    static lv_style_t style_sel;
    lv_style_init(&style_sel);
    lv_style_set_text_font(&style_sel, &lv_font_montserrat_22);

    const char * opts = "1\n2\n3\n4\n5\n6\n7\n8\n9\n10";
    lv_obj_t * roller;

    /*A roller on the right with right aligned text, and custom width*/
    roller = lv_roller_create(lv_scr_act());
    lv_obj_set_style_text_font(roller, &lv_font_montserrat_30, LV_PART_MAIN);
    lv_obj_set_style_text_font(roller, &lv_font_montserrat_48, LV_PART_SELECTED);
    lv_roller_set_options(roller, opts, LV_ROLLER_MODE_INFINITE);
    lv_obj_set_width(roller, 80);
    lv_obj_add_style(roller, &style_sel, LV_PART_SELECTED);
    lv_obj_set_style_text_align(roller, LV_TEXT_ALIGN_RIGHT, 0);
    lv_obj_align(roller, LV_ALIGN_RIGHT_MID, -10, 0);
    lv_roller_set_selected(roller, 8, LV_ANIM_ON);
    lv_obj_center(roller);
    lv_obj_set_style_bg_opa(roller, LV_OPA_100, LV_PART_MAIN);
    lv_obj_set_style_bg_color(roller, lv_color_hex(0xffff00), LV_PART_MAIN);
    lv_obj_set_style_text_line_space(roller, 0, LV_PART_MAIN);
    lv_roller_set_visible_row_count(roller, 3);
}

liamHowatt, 2024-04-16:
Thanks for the info!

I came to the same conclusion. I believe removing that logic is the best solution.

I was scratching my head trying to understand exactly what `corr` is correcting. I believe `corr` is a duplication of the correction logic done in `get_sel_area`. I found out that there was once a time when `get_sel_area` didn't exist so `corr` was needed. But then `get_sel_area` was added so `corr` became redundant. Now it can be removed.

#2340, c774c26 is when `get_sel_area` was added to the v8.0 release branch.

I'm opening a PR to remove `corr`.

#6070 is another issue that opened 3 days before this one, reporting the exact same thing but for master. Read my writeup in that issue as well. I'm opening an identical PR for master too.

kisvegabor, 2024-05-03:
Should be resolved by https://github.com/lvgl/lvgl/pull/6083 and https://github.com/lvgl/lvgl/pull/6082

kisvegabor, 2024-05-16:
#6083 and #6082 are merged so I'm closing it.

closed 2024-05-16 by kisvegabor