lvgl/lvgl #6069: How to prohibit a specified color from appearing inside the lvgl

chenyinen, 2024-04-14:
### Introduce the problem

My hardware image format is 565, in order to make the video layer display, I can only set the colorkey, such as FB fill 0xffff, it will display the underlying video. But normal white will also become transparent, so I want to actively set the 0xffff when the application layer needs it, and fill it with 0xfffe when the frame is usually painted white, or is there a better way?

liamHowatt, 2024-04-16:
Interesting problem! To reiterate: you're using hardware that has an independent video stream and you're using LVGL to draw overlay graphics? The hardware interprets a certain magic color value as a pixel that should passthrough the underlying video pixel? So you want to stop LVGL from using that magic color value?

Do you already have a partial solution working where you're scanning for 0xffff in the display flush cb and replacing them with 0xfffe? I think that's a good solution.

Could you make the color key bright green or another color you're not likely to render accidentally? 0x07e0 is pure green. You could use that as the color key and then just not use pure bright green in your graphics.

Another option is to use that green 0x07e0 color key and then in your display flush cb, do this to every uint16_t pixel:
```c
pix &= ~0b1111111111011111u;
```
It unsets the least significant bit of the green channel so the bright green color can never occur. It has a side effect of reducing the depth of the green channel from 6 bits to 5 bits, like the red and blue channels.

<hr>

One final option is something like this:

Where your color key is 0xffff (white) and you dim everything an unnoticeable amount.
```c
/* create a screen-sized rectangle on the sys layer so it's on top */
lv_obj_t * dimmer = lv_obj_create(lv_layer_sys());
lv_obj_remove_style_all(dimmer);
lv_obj_set_size(dimmer, lv_pct(100), lv_pct(100));

/* do either... */

/* this: */
/* black layer with very low opacity */
lv_obj_set_style_bg_color(dimmer, lv_color_black(), 0);
lv_obj_set_style_bg_opa(dimmer, 4, 0);

/* or this: */
/* subtract tiny amount of color from everything below this layer */
lv_obj_set_style_bg_color(dimmer, lv_color_make(4, 4, 4), 0);
lv_obj_set_style_bg_opa(dimmer, LV_OPA_COVER, 0);
lv_obj_set_style_blend_mode(dimmer, LV_BLEND_MODE_SUBTRACTIVE, 0);
```

In my testing I could dim everything an unnoticeable amount and cause all 0xffff to become a slightly lower value.

kisvegabor, 2024-04-16:
You can also check all the pixels of the rendered image in the `flush_cb` and change 0xffff to something else.

chenyinen, 2024-04-17:
> 您还可以检查渲染图像的所有像素，`flush_cb`并将 0xffff 更改为其他值。

But this will disable all colorkeys. I hope to set the colorkey through the specified interface.
Because my hardware environment can only set fb to colorkey to display video. If I don’t want to display video, then fb needs to never have colorkey pixels. When the video needs to be displayed, the specified area fb is filled with colorkey, but at this time Colorykey cannot appear in other areas, even in png images, JPEG images, etc. with pixels equal to colorkey. Is there a universal processing method? I don't want to modify the image through PS to eliminate the pixels of the specified colorkey

chenyinen, 2024-04-17:
> Interesting problem! To reiterate: you're using hardware that has an independent video stream and you're using LVGL to draw overlay graphics? The hardware interprets a certain magic color value as a pixel that should passthrough the underlying video pixel? So you want to stop LVGL from using that magic color value?
> 
> Do you already have a partial solution working where you're scanning for 0xffff in the display flush cb and replacing them with 0xfffe? I think that's a good solution.
> 
> Could you make the color key bright green or another color you're not likely to render accidentally? 0x07e0 is pure green. You could use that as the color key and then just not use pure bright green in your graphics.
> 
> Another option is to use that green 0x07e0 color key and then in your display flush cb, do this to every uint16_t pixel:
> 
> ```c
> pix &= 0b1111111111011111u;
> ```
> 
> It unsets the least significant bit of the green channel so the bright green color can never occur. It has a side effect of reducing the depth of the green channel from 6 bits to 5 bits, like the red and blue channels.
> 
> One final option is something like this:
> 
> Where your color key is 0xffff (white) and you dim everything an unnoticeable amount.
> 
> ```c
> /* create a screen-sized rectangle on the sys layer so it's on top */
> lv_obj_t * dimmer = lv_obj_create(lv_layer_sys());
> lv_obj_remove_style_all(dimmer);
> lv_obj_set_size(dimmer, lv_pct(100), lv_pct(100));
> 
> /* do either... */
> 
> /* this: */
> /* black layer with very low opacity */
> lv_obj_set_style_bg_color(dimmer, lv_color_black(), 0);
> lv_obj_set_style_bg_opa(dimmer, 4, 0);
> 
> /* or this: */
> /* subtract tiny amount of color from everything below this layer */
> lv_obj_set_style_bg_color(dimmer, lv_color_make(4, 4, 4), 0);
> lv_obj_set_style_bg_opa(dimmer, LV_OPA_COVER, 0);
> lv_obj_set_style_blend_mode(dimmer, LV_BLEND_MODE_SUBTRACTIVE, 0);
> ```
> 
> In my testing I could dim everything an unnoticeable amount and cause all 0xffff to become a slightly lower value.

Yes, this is exactly my usage scenario, and these are all good solutions, but there is a flaw. I only want to set the colorkey key when needed. For example, if I open the front camera, I need to set fb to Colorkey displays the video and also needs to display some png images. If these png images contain colorkey, their display will be abnormal.
If I disable colorkey in flash cb, then I cannot actively set colorkey through lvgl unless I directly operate fb, but this is obviously impossible.

liamHowatt, 2024-04-19:
Are you saying that the reason the listed solutions won't work is because sometimes you need to filter the color key from LVGL's output and other times you want to intentionally generate it with LVGL and keep it?

What will your UI look like? Will the video area be a rectangle surrounded by UI, or UI on the bottom edge only, or UI components floating on top of the video?

chenyinen, 2024-04-20:
My video area will not be displayed all the time, usually the UI wizard is displayed normally, and when entering the playback interface, I need to use the middle area to display the video

closed 2024-04-24 by chenyinen