lvgl/lvgl #6070: LV Roller text is not centered.

Kirbyrawr, 2024-04-14:
### LVGL version

9.1.0

### What happened?

[image]

It was fixed here:
https://github.com/lvgl/lvgl/commit/708272c37cacc190e327373f1972dadccbf137e9

However in the LVGL 9.1.0 version it points to LV_PART_MAIN instead of LV_PART_SELECTED as in the issue 3 years ago.

### How to reproduce?

https://github.com/lvgl/lvgl/blob/af642572aabbb7a882b7926d78e62c04b967df9c/examples/widgets/roller/lv_example_roller_2.c

Even in the examples is looking wrong.

liamHowatt, 2024-04-16:
Thanks for linking that v7 fix. That info helped me figure it out. There was a similar issue created this week but for v8.3 #6066. That fix you linked works for the v7 version of lv_roller.c but a fair bit has changed since v7 and there's a better fix for the current v9.1.

Here's a timeline.

v7.0 had the bug
v7 release branch had the fix you linked on May 11, 2021. It works, but I'm not convinced that was the best solution even for that version.
The master branch had 01d1c87 July 5, 2021 which refactored and corrected some of the logic in `draw_main`.
v8.0 release branch had c774c26 July 16, 2021 which was a backport of 01d1c87 for v8.

Now, I believe 01d1c87 / c774c26 almost fixed it but there's a leftover redundancy that forgot to be removed. I'm opening a PR for it now. See the similar issue #6066 for more info. I'm opening an identical PR for v8.4.

Kirbyrawr, 2024-04-17:
Hi @liamHowatt thank you so much, let me know if i can help or test it.

kisvegabor, 2024-05-03:
Should be fixed in #https://github.com/lvgl/lvgl/pull/6083 and https://github.com/lvgl/lvgl/pull/6082

PR #6083 (2024-05-16): fix(roller): remove redundant logic in draw selected
### Description of the feature or fix

Fixes #6070

See issues #6070 and #6066 for why this logic is redundant.

This is a master branch PR. I also opened a v8.4 PR  #6082

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl [... cut]
closed 2024-05-16 by kisvegabor