lvgl/lvgl #6078: Compilation error for FATFS

mgaman, 2024-04-16:
### LVGL version

v9.1.0

### What happened?

Enabled LV_USE_FS_FATFS in lv_conf,h and got the following error:
.pio/libdeps/esp32-s3-devkitc-1/lvgl/src/libs/fsdrv/lv_fs_fatfs.c:228:5: error: unknown type name 'DIR'; did you mean 'DDR'?

### How to reproduce?

#define LV_USE_FS_FATFS 1
#if LV_USE_FS_FATFS
    #define LV_FS_FATFS_LETTER 'F'     /*Set an upper cased letter on which the drive will accessible (e.g. 'A')*/
    #define LV_FS_FATFS_CACHE_SIZE 0    /*>0 to cache this number of bytes in lv_fs_read()*/
#endif

liamHowatt, 2024-04-16:
Please see https://github.com/lvgl/lvgl/issues/6076#issuecomment-2060066717. The same applies for FATFS.

brandon-kim, 2024-04-18:
ff.h uses FF_DIR instead of DIR, so My simple solution is to add define below the ff.h include as shown below.
#include "ff.h"
#define DIR FF_DIR
this may not be a proper solution, Hope this will be a reference to solve your problem.

liamHowatt, 2024-04-19:
Thanks @brandon-kim! I see that ff.h is an [Arduino header](https://github.com/stm32duino/FatFs/blob/main/src/ff.h). Are there many people like you and @mgaman who are using a different version of that header which defines `DIR` as `FF_DIR`?

anujdeshpande, 2024-04-19:
@liamHowatt if it helps - I too had to set the define (esp-idf 5.1,lvgl 8.4, fatfs from esp-idf)

liamHowatt, 2024-04-25:
I sampled some projects that modify the ff.h from the ChaN project. I see that ESP IDF changed `DIR` to `FF_DIR` and MbedOS changed `DIR` to `FATFS_DIR`. I assume because they ran into a conflict with the standard header dirent.h that defines `DIR` which is a type that is completely unrelated to the one in FATFS.

I'm going to open a PR that adds `#define FATFS_DIR DIR` and `#define FF_DIR DIR` _before_ the ff.h include so the ff.h, regardless of which flavor, esp, mbed, or unmodified will have `DIR` defined as `DIR` inside.

PR #6128 (2024-05-12): fix(fsdrv/fatfs): support FF_DIR and FATFS_DIR typedef in ff.h
### Description of the feature or fix

Fixes #6078

See the issue for more details.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://git [... cut]
closed 2024-05-12 by FASTSHIFT