lvgl/lvgl #6084: Unable to change new PNG picture on loop

Neothai, 2024-04-17:
### LVGL version

v9.1.0

### What happened?

Details of the hardware I use
MCU: ESP32-S3 Flash 16MB, PSRAM 8MB
Display: 3.5 inch TFT ILI9488 (16-bit parallel Communication type)

This project uses Arduino-IDE software developed.

Details of this project
Because my project needs to fetch a PNG image from the internet and display it on the display, I download the image and put it on LittleFS and then display the PNG image on the display using the PNG decoder within the library. Therefore, PNG images cannot be imported through LVGL's online image conversion tool.

I have to do this process every time. times when users request Therefore, it is necessary that I perform those processes on the loop of the program.

My main problem is I cannot change the new image even though the downloaded file data sent to LVGL is a new file data (with the same file name).
The nature of this problem is when I changed the image to a new one. But the same picture still appears on the display.

[ImageData.txt](https://github.com/lvgl/lvgl/files/15007045/ImageData.txt)

"data1" in the code is this image.
[image]

"data2" is this image.
[image]

"data3" is this image.
[image]

After powering on when executing this command (I'll call it "cmd1" for short.)
```cpp
if('a' == Serial.read()){
    if(lv_obj_is_valid(imgobj_qr1)){lv_obj_delete_async(imgobj_qr1); Serial.println("del old1------------------------");}
    imgobj_qr1 = lv_image_create(lv_screen_active()); Serial.println("create new1");
    Serial.println("change img");
    base64_decodeFunc(data2, "/decoded_image.png"); //Convert images in Base64 format to PNG. Then store it in LittleFS with the file name "decoded_image.png"
    lv_image_set_src(imgobj_qr1, "C:/decoded_image.png");
    lv_obj_align(imgobj_qr1, LV_ALIGN_CENTER, 0, 0);
  }
```
Result is
[image]

and execute this command next(I'll call it "cmd3" for short.)
```cpp
if('c' == Serial.read()){
    if(lv_obj_is_valid(imgobj_qr1)){lv_obj_delete_async(imgobj_qr1); Serial.println("del old3------------------------");}
    imgobj_qr1 = lv_image_create(lv_screen_active()); Serial.println("create new3");
    Serial.println("change img3");
    base64_decodeFunc(data3, "/decoded_image.png");
    lv_image_set_src(imgobj_qr1, "C:/decoded_image.png");
    lv_obj_align(imgobj_qr1, LV_ALIGN_CENTER, 0, 0);
  }
```
Result is
[image]
The results remain the same as the first time.

But after powering on again and executing the "cmd3" command first, I get the following results.
[image]
and execute "cmd1" command next, Result is
[image]

Is there any information you would like from me or what should I do next? Please tell me!😊

### How to reproduce?

```cpp
[... 174 lines cut ...]
  
  if('b' == Serial.read()){
    if(lv_obj_is_valid(imgobj_qr1)){lv_obj_delete_async(imgobj_qr1); Serial.println("del old2------------------------");}
    imgobj_qr1 = lv_image_create(lv_screen_active()); Serial.println("create new2");
    Serial.println("change img2");
    base64_decodeFunc(data1, "/decoded_image2.png");
    lv_image_set_src(imgobj_qr1, "C:/decoded_image2.png");
    lv_obj_align(imgobj_qr1, LV_ALIGN_CENTER, 0, 0);
  }
  
  if('c' == Serial.read()){
    if(lv_obj_is_valid(imgobj_qr1)){lv_obj_delete_async(imgobj_qr1); Serial.println("del old3------------------------");}
    imgobj_qr1 = lv_image_create(lv_screen_active()); Serial.println("create new3");
    Serial.println("change img3");
    base64_decodeFunc(data3, "/decoded_image.png");
    lv_image_set_src(imgobj_qr1, "C:/decoded_image.png");
    lv_obj_align(imgobj_qr1, LV_ALIGN_CENTER, 0, 0);
  }

  unsigned long tick_millis = millis() - lvgl_tick_millis;
  lvgl_tick_millis = millis();
  lv_tick_inc(tick_millis);
  delay(2);
}
```

kisvegabor, 2024-04-19:
Does the UI still work after changing the image?

Please try these:
- call `lv_obj_invalidate(imgobj_qr1)`
- call `lv_image_cache_drop("C:/decoded_image.png");`
- set `LV_CACHE_DEF_SIZE 0` in `lv_conf.h`
- try saving the image with a different name

Neothai, 2024-04-20:
Thank you for the good advice. I did the test as you suggested and got results like this.

> Does the UI still work after changing the image

Yes, the UI and other objects still work normally.
___
> call `lv_obj_invalidate(imgobj_qr1)`

I have tried doing this
``` 
...
lv_image_set_src(imgobj_qr1, "C:/decoded_image.png");
lv_obj_align(imgobj_qr1, LV_ALIGN_CENTER, 0, 0);

lv_obj_invalidate(imgobj_qr1); // <---
```
But still nothing has changed.
____

> call `lv_image_cache_drop("C:/decoded_image.png");`

**It works**, but needs to call `lv_image_cache_drop("C:/decoded_image.png");` before creating an image like this.
```
lv_image_cache_drop("C:/decoded_image.png"); //<--
if(lv_obj_is_valid(imgobj_qr1)){lv_obj_delete_async(imgobj_qr1); Serial.println("del old1------------------------");}
imgobj_qr1 = lv_image_create(lv_screen_active()); Serial.println("create new1");
...
lv_image_set_src(imgobj_qr1, "C:/decoded_image.png");
lv_obj_align(imgobj_qr1, LV_ALIGN_CENTER, 0, 0);
```
____
> set `LV_CACHE_DEF_SIZE` 0 in lv_conf.h`

I originally set this to
```
#define LV_MEM_SIZE                   (3072U * 1024U) // Allocate on PSRAM

#define LV_CACHE_DEF_SIZE             (1500U * 1024U)
#define LV_IMAGE_HEADER_CACHE_DEF_CNT (512U * 1024U)
```
When changed to
```
#define LV_MEM_SIZE                   (3072U * 1024U) // Allocate on PSRAM

#define LV_CACHE_DEF_SIZE             0
#define LV_IMAGE_HEADER_CACHE_DEF_CNT 0
```
It works too. But it takes a very long time to draw and the file is constantly being read (for large images).
____

> try saving the image with a different name

It works too (when `LV_CACHE_DEF_SIZE` and `LV_IMAGE_HEADER_CACHE_DEF_CNT` are not 0), but images in the old cache are not cleared when changing to new images. Then the cache will be full when changing the image to a new one within 3-4 times.
____
In conclusion, I think your suggestion **call `lv_image_cache_drop("C:/decoded_image.png");`** is the best solution.

And this would be great if in the documentation https://docs.lvgl.io/master/libs/lodepng.html it was suggested to add `lv_image_cache_drop(imageSrc);` before loading new images.

Thanks again for the great solution to this problem. 😊

closed 2024-04-20 by Neothai

kisvegabor, 2024-04-22:
Awesome! :tada: 

Actually it's related to images in general, because LVGL caches images by file name (or pointer in case of variables). So if you change the content of the image you need to tell LVGL to re-cache it.

@W-Mai I just saw that the docs of the image caching is outdated [here](https://docs.lvgl.io/overview/image.html#image-caching). Do you have time to update it?