lvgl/lvgl #6088: Core panic'ed lv_slider_set_value 8.4

Jay-esp, 2024-04-18:
### LVGL version

8.4

### What happened?

Core panic'ed when settings value of a slider, upgraded from 8.3 to 8.4, same, took the code to bare bones, regardless if animation is on or off, i used the same core many times and sliders too but this is the first time i set the value of a slider.
Exception decoder gives me this:
PC: 0x400ed0e3: lv_bar_set_value at E:\workdir\arduino\preamplifier\arduino-1.8.19\portable\sketchbook\libraries\lvgl\src\widgets\lv_bar.c line 95
EXCVADDR: 0x00000024

Decoding stack results
0x400ed0e0: lv_bar_set_value at E:\workdir\arduino\preamplifier\arduino-1.8.19\portable\sketchbook\libraries\lvgl\src\widgets\lv_bar.c line 91
0x400d1c6e: loop() at e:\workdir\arduino\preamplifier\arduino-1.8.19\portable\sketchbook\libraries\lvgl\src/widgets/lv_slider.h line 86
0x400f8a6d: loopTask(void*) at E:\workdir\arduino\preamplifier\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.11\cores\esp32\main.cpp line 50

### How to reproduce?

#include <SPI.h>
#include <lvgl.h>
#include <Ticker.h>
#include <TFT_eSPI.h>

#define LVGL_TICK_PERIOD 20

// spi chip selects
#define CS_LCD 15
#define CS_TOUCH 5

#define LV_USE_DEBUG = 0;

Ticker Tick; /* timer for interrupt handler */
TFT_eSPI TFT = TFT_eSPI(); /* TFT instance */

static lv_disp_draw_buf_t draw_buf;
static lv_color_t buf[480 * 320 / 10];

// timers
unsigned long LVGL_Timer;
unsigned long looptimer;

// pointer to update
static lv_obj_t *slider_volume;

/* Display flushing */
void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
  uint32_t w = (area->x2 - area->x1 + 1);
  uint32_t h = (area->y2 - area->y1 + 1);
  TFT.startWrite();
  TFT.setAddrWindow(area->x1, area->y1, w, h);
  TFT.pushColors(&color_p->full, w * h, true);
  TFT.endWrite();
  lv_disp_flush_ready(disp);
}

/* Interrupt driven periodic handler */
static void lv_tick_handler(void)
{
  lv_tick_inc(LVGL_TICK_PERIOD);
}

void setup() {
  Serial.begin(115200);

  pinMode(CS_LCD, OUTPUT);
  pinMode(CS_TOUCH, OUTPUT);
  digitalWrite(CS_LCD, HIGH);
  digitalWrite(CS_TOUCH, HIGH);

  lv_init();

  TFT.init();
  TFT.setRotation(3); /* Landscape orientation */

  lv_disp_draw_buf_init(&draw_buf, buf, NULL, (480 * 320 / 10));  /*Initialize the display buffer.*/

  /*Initialize the display*/
  static lv_disp_drv_t disp_drv;        /*Descriptor of a display driver*/
  lv_disp_drv_init(&disp_drv);          /*Basic initialization*/
  disp_drv.flush_cb = my_disp_flush;    /*Set your driver function*/
  disp_drv.draw_buf = &draw_buf;          /*Assign the buffer to the display*/
  disp_drv.hor_res = 480;   /*Set the horizontal resolution of the display*/
  disp_drv.ver_res = 320;   /*Set the verizontal resolution of the display*/
  lv_disp_drv_register(&disp_drv);      /*Finally register the driver*/

  /*Initialize the graphics library's tick*/
  Tick.attach_ms(LVGL_TICK_PERIOD, lv_tick_handler);

  lv_obj_t * slider_volume = lv_slider_create(lv_scr_act());
  lv_obj_align(slider_volume, LV_ALIGN_TOP_LEFT, 50, 150);

  LVGL_Timer = millis() - 200;
  looptimer = millis();

}

void loop() {
  if (millis() - looptimer > 5000) {
    lv_slider_set_value(slider_volume, 0, LV_ANIM_ON);
  }
  if (millis() >= LVGL_Timer + 5) {
    lv_task_handler();
    LVGL_Timer = millis();
  }
}

Jay-esp, 2024-04-18:
I upgraded to 9.1 but same problem

PC: 0x400e9f17: lv_bar_set_value at E:\workdir\arduino\preamplifier\arduino-1.8.19\portable\sketchbook\libraries\lvgl\src\widgets\bar\lv_bar.c line 96
EXCVADDR: 0x0000002c

Decoding stack results
0x400e9f14: lv_bar_set_value at E:\workdir\arduino\preamplifier\arduino-1.8.19\portable\sketchbook\libraries\lvgl\src\widgets\bar\lv_bar.c line 92
0x400d1a81: loop() at e:\workdir\arduino\preamplifier\arduino-1.8.19\portable\sketchbook\libraries\lvgl\src/widgets/slider/lv_slider.h line 79
0x400f9a79: loopTask(void*) at E:\workdir\arduino\preamplifier\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.11\cores\esp32\main.cpp line 50

kisvegabor, 2024-04-19:
Can you add this at the beginning of `lv_bar_set_value()`?
```c
printf("%p\n", obj);
```

Jay-esp, 2024-04-19:
i get 0x0
So no object?
Funny thing is...if i set the value somewhere else like while i setup the screens it does not crash and i get 0x3ffc50f4
In fact i get this number 0x3ffc50f4 probably when the slider is first drawn, then again when i set the value where i build the screen and thirdly when i set the value at the point it crashes.
I can use the slider, everything else seems to work, i just cant set the value. Bar set is not called when i move the slider.

Jay-esp, 2024-04-19:
Ok, the error is sitting in front of the keyboard...

static lv_obj_t *slider_volume;
lv_obj_t * slider_volume = lv_slider_create(lv_scr_act());

i declared the object twice

closed 2024-04-19 by Jay-esp