lvgl/lvgl #6092: Unable to load image from SD card - text file loads fine with `lv_fs_read`

anujdeshpande, 2024-04-19:
### LVGL version

v8.4.0

### What happened?

I have a Elecrow 7inch RGB screen with esp32s3+psram that has a SD card.
I have enabled fatfs as my preferred fs and that seems to work as expected. I tested this by adding a txt file to the SD card, and read it.

```
        lv_fs_file_t f;
        lv_fs_res_t res = lv_fs_open(&f, "A:readme.txt", LV_FS_MODE_RD);        /*From file*/
        ESP_LOGI(TAG,"%d",res);
        uint32_t read_num;
        uint8_t buf[TUX_READ_BUFF_SIZE];
        res = lv_fs_read(&f, buf, TUX_READ_BUFF_SIZE-1, &read_num);
        if(res != LV_FS_RES_OK) {
                ESP_LOGE(TAG, "Failed to read from ");
        }
        lv_fs_close(&f);

        // Display the read contents from the file
        ESP_LOGI(TAG, "Read  : %s", buf);
```

The output of the above section is as follows -

```
I (1292) LVGL: 0
I (1292) LVGL: Read  : Anuj's Demo!
```

So I added

```
        res = lv_fs_open(&f, "A:FirstScreenPerm800x480.bin", LV_FS_MODE_RD);
        ESP_LOGI(TAG,"%d",res);
        lv_fs_close(&f);
        lv_img_set_src(bg, "A:FirstScreenPerm800x480.bin");
```

The output of the above is

```
����?
I (1292) LVGL: 12
```

anujdeshpande, 2024-04-21:
Same result with the png library as well

anujdeshpande, 2024-04-21:
Turns out it was a problem with my display driver. This was completely unrelated to fatfs.

closed 2024-04-21 by anujdeshpande

ductuanhoang, 2025-02-22:
Hi @anujdeshpande I have the same issue, so Do you know what the problem is?
I can read the txt file , but nothing show when I load the image from SD card
my config:
`
/*PNG decoder library*/
#define LV_USE_PNG 1

/*BMP decoder library*/
#define LV_USE_BMP 1
/*API for FATFS (needs to be added separately). Uses f_open, f_read, etc*/
#define LV_USE_FS_FATFS  1
#if LV_USE_FS_FATFS
    #define LV_FS_FATFS_LETTER 'A'     /*Set an upper cased letter on which the drive will accessible (e.g. 'A')*/
    #define LV_FS_FATFS_CACHE_SIZE 1    /*>0 to cache this number of bytes in lv_fs_read()*/
#endif`
Thanks!!!!

anujdeshpande, 2025-02-24:
@ductuanhoang my issue was with the display driver. Nothing to do with fatfs

kisvegabor, 2025-02-24:
It can be an put of memory issue too. Please enable [logging](https://docs.lvgl.io/master/details/debugging/log.html) to see wht's happening.

ductuanhoang, 2025-02-25:
Hi @kisvegabor, yes, I also enabled logging and nothing shows any error (in the previous bard, I tried to work with the esp32 worm kit JC2435W328).

ductuanhoang, 2025-02-25:
I have tested with SD card read by function "lv_fs_open" and it also works this is a trace logging message I do not see any error message when trying to load the image from the SD card. Now all code I test on board "esp32-S3-Touch-LCD-7"
`[Info]	(9.320, +10)	 lv_img_create: begin 	(in lv_img.c line #60)
[Trace]	(9.326, +6)	 lv_obj_class_create_obj: Creating object with 0x3c0a33c0 class on 0x3fcee63c parent 	(in lv_obj_class.c line #45)
[Trace]	(9.338, +12)	 lv_mem_alloc: allocating 60 bytes 	(in lv_mem.c line #127)
[Trace]	(9.344, +6)	 lv_mem_alloc: allocated at 0x3fcee6b8 	(in lv_mem.c line #160)
[Trace]	(9.352, +8)	 lv_obj_class_create_obj: creating normal object 	(in lv_obj_class.c line #82)
[Trace]	(9.360, +8)	 lv_mem_alloc: allocating 28 bytes 	(in lv_mem.c line #127)
[Trace]	(9.368, +8)	 lv_mem_alloc: allocated at 0x3fcee704 	(in lv_mem.c line #160)
[Trace]	(9.376, +8)	 lv_mem_alloc: allocating 4 bytes 	(in lv_mem.c line #127)
[Trace]	(9.382, +6)	 lv_mem_alloc: allocated at 0x3fcee24c 	(in lv_mem.c line #160)
[Trace]	(9.390, +8)	 lv_obj_constructor: begin 	(in lv_obj.c line #403)
[Trace]	(9.396, +6)	 lv_obj_constructor: finished 	(in lv_obj.c line #428)
[Trace]	(9.402, +6)	 lv_img_constructor: begin 	(in lv_img.c line #391)
[Trace]	(9.408, +6)	 lv_img_constructor: finished 	(in lv_img.c line #412)
[Trace]	(9.416, +8)	 event_send_core: Sending event 42 to 0x3fcee6b8 with 0x0 param 	(in lv_event.c line #428)
[Trace]	(9.426, +10)	 event_send_core: Sending event 19 to 0x3fcee6b8 with 0x3fcebcae param 	(in lv_event.c line #428)
[Trace]	(9.436, +10)	 event_send_core: Sending event 19 to 0x3fcee6b8 with 0x3fcebdce param 	(in lv_event.c line #428)
[Trace]	(9.446, +10)	 event_send_core: Sending event 34 to 0x3fcee63c with 0x3fcee6b8 param 	(in lv_event.c line #428)
[Trace]	(9.456, +10)	 event_send_core: Sending event 35 to 0x3fcee63c with 0x3fcee6b8 param 	(in lv_event.c line #428)`

ductuanhoang, 2025-02-25:
Hi, I have an update now I can display to the TFT screen but only one time :)) I still find the reason between the custom driver and lvgl load image this is my driver the CS pin control by expander IC, I think some still wrong with this
```
void setCS(bool state)
{
    if (state == false)
        expander->digitalWrite(SD_CS, LOW);
    else
        expander->digitalWrite(SD_CS, HIGH);
}

static void *fs_open(lv_fs_drv_t *drv, const char *path, lv_fs_mode_t mode)
{
    lv_fs_file_t *file_p = new lv_fs_file_t(); // Allocate LVGL file struct
    int res = 0;
    setCS(false);
    res = lv_fs_open(file_p, path, mode);
    // setCS(true);
    if (res == LV_FS_RES_OK)
        return (void *)file_p; // Return file reference
    else
    {
        delete file_p;
        return NULL;
    }
}

static lv_fs_res_t fs_write(lv_fs_drv_t *drv, void *file_p, const void *buf, uint32_t btw, uint32_t *bw)
{
    lv_fs_res_t res = LV_FS_RES_UNKNOWN;
    setCS(false);
    res = lv_fs_write((lv_fs_file_t *)file_p, buf, btw, bw);
    setCS(true);
    return res;
}

static lv_fs_res_t fs_seek(lv_fs_drv_t *drv, void *file_p, uint32_t pos, lv_fs_whence_t whence)
{
    lv_fs_res_t res = LV_FS_RES_UNKNOWN;
    setCS(false);
    res = lv_fs_seek((lv_fs_file_t *)file_p, pos, whence);
    setCS(true);
    return res;
}

static lv_fs_res_t fs_tell(lv_fs_drv_t *drv, void *file_p, uint32_t *pos)
{
    lv_fs_res_t res = LV_FS_RES_UNKNOWN;
    setCS(false);
    res = lv_fs_tell((lv_fs_file_t *)file_p, pos);
    setCS(true);
    return res;
}

static lv_fs_res_t fs_close(lv_fs_drv_t *drv, void *file_p)
{
    lv_fs_res_t res = LV_FS_RES_UNKNOWN;
    setCS(false);
    res = lv_fs_close((lv_fs_file_t *)file_p);
    setCS(true);
    return res;
}

static lv_fs_res_t fs_read(lv_fs_drv_t *drv, void *file_p, void *buf, uint32_t btr, uint32_t *br)
{
    lv_fs_res_t res = LV_FS_RES_UNKNOWN;
    setCS(false);
    res = lv_fs_read((lv_fs_file_t *)file_p, buf, btr, br);
    setCS(true);
    return res;
}

static void *fs_dir_open(lv_fs_drv_t *drv, const char *path)
{
    lv_fs_dir_t *rddir_p = new lv_fs_dir_t(); // Allocate directory structure
    int res = 0;
    setCS(false);
    res = lv_fs_dir_open(rddir_p, path);
    // setCS(true);
    setCS(false);
    if (res == LV_FS_RES_OK)
        return (void *)rddir_p; // Return directory reference
    else
    {
        delete rddir_p;
        return NULL;
    }
}

static lv_fs_res_t fs_dir_close(lv_fs_drv_t *drv, void *rddir_p)
{
    lv_fs_res_t res = LV_FS_RES_UNKNOWN;
    setCS(false);
    res = lv_fs_dir_close((lv_fs_dir_t *)rddir_p);
    setCS(true);
    return res;
}

static lv_fs_res_t fs_dir_read(lv_fs_drv_t *drv, void *rddir_p, char *fn)
{
    lv_fs_res_t res = LV_FS_RES_UNKNOWN;
    setCS(false);
    res = lv_fs_dir_read((lv_fs_dir_t *)rddir_p, fn);
    setCS(true);
    return res;
}

static void lv_port_sd_fs_init(void)
{
    /*---------------------------------------------------
     * Register the file system interface in LVGL
     *--------------------------------------------------*/

    static lv_fs_drv_t fs_drv;
    lv_fs_drv_init(&fs_drv);

    /*Set up fields...*/
    fs_drv.letter = 'A';
    fs_drv.cache_size = 1;

    fs_drv.open_cb = fs_open;
    fs_drv.close_cb = fs_close;
    fs_drv.read_cb = fs_read;
    fs_drv.write_cb = fs_write;
    fs_drv.seek_cb = fs_seek;
    fs_drv.tell_cb = fs_tell;

    fs_drv.dir_close_cb = fs_dir_close;
    fs_drv.dir_open_cb = fs_dir_open;
    fs_drv.dir_read_cb = fs_dir_read;

    lv_fs_drv_register(&fs_drv);
}
```

ductuanhoang, 2025-02-25:
I have fixed, this issue is "CS" control pin so  need to will always hold this pin to low and remove all CS control code.

[image]

kisvegabor, 2025-02-25:
Awesome! Congratulations! 🎉