lvgl/lvgl #6099: Strange behavior ans crash with use of tabview

Herve14, 2024-04-21:
### LVGL version

V9.1.1 

### What happened?

Hi,
i am using lvgl with arduino Due (you can see a rc transmitterr posted the project part on the forum)  and i  have sommes issue when i want to switch my code to lvgl 9.1.
I  get crash w/ tabview  that i use o lot.
I try the widget and i  can fill only the first tab (Profile)... 
Some time i get this king of  message [Error] lv_chart_set_series_color: Asserted at expression: series != NULL (NULL pointer) lv_chart.c:363
I  can't show the wrong code because it's difficult to know exactly where the code crash even with  a JTAG probe...

With lvgl 8 i get no issue..   

Do you think Arduino Due can't support LVGL 9 version???

BEst Regards

kisvegabor, 2024-04-25:
Hi,

Can you increase `LV_MEM_SIZE` a little bit?

Herve14, 2024-05-02:
Hi,

       Sorry for this late response. Yes, I tried different values ​​of LV_MEM_SIZE up to (64*1024) without any noticeable effect.

A debug code for arduino Due i get this informations about use of memory  (in loop) :

> Dynamic ram used: 21896
> Program static ram used 62040
> Stack ram used 64
> 
> My guess at free mem: 14304

I think there's someting wrong in the lv_conf.h

Best regards

Herve14, 2024-05-03:
Hi, thank you for your return..

well i finally  spent some time to understand the "new" memory management to tune it perfectly..

Now LVGL v9 works fine w/arduino Due with C malloc or builtin lv_malloc. we can close this subject!!

Just a question : what is the recommanded malloc funtion? ( i have more free memory with "c" malloc..  )

Best regards

kisvegabor, 2024-05-07:
> Just a question : what is the recommanded malloc funtion? ( i have more free memory with "c" malloc.. )

How do you monitor the free memory with C malloc? C doesn't have any memory monitoring features. I think you see an `LV_MEM_SIZE` increase in RAM usage with LVGL's malloc (LVGL allocates here). With C malloc it is a little bit more sneaky as the memory used for malloc (called `heap`) is not explicitly shown in RAM usage.

Herve14, 2024-05-08:
Hi Gabor

Well, to monitor Arduino DUE free memory i use this lines of code

1 : In global section :

    extern char _end;
    extern "C" char *sbrk(int i);
    char *ramstart = (char *)0x20070000;  //for SAM3X8E
    char *ramend = (char *)0x20088000;

2 : In loop section

        char *heapend = sbrk(0);
        register char * stack_ptr asm ("sp");
        struct mallinfo mi = mallinfo();
        printf("\nDynamic ram used: %d\n", mi.uordblks);
        printf("Program static ram used %d\n", &_end - ramstart);
        printf("Stack ram used %d\n\n", ramend - stack_ptr);
        printf("My guess at free mem: %d\n", stack_ptr - heapend + mi.fordblks);

Best regards

kisvegabor, 2024-05-13:
Oh, very smart! :+1: 

How large the difference is with LVGL malloc vs stdlib malloc? If it's not too significant then it's probably just because our allocator is just a good enough solution. Platform specific solutions could be more optimal or they might trade speed and memory overhead differently.

closed 2024-06-11 by lvgl-bot